evcc-io/evcc #7780: unable getting the SoC of my Telsa

abdulson, 2023-05-02:
### Describe the bug

at the gui the loadpoint is visible and the connected Tesla, but log states that no infosof SoC available:[main  ] ERROR 2023/05/02 16:14:32 vehicle status: 408 Request Timeout
[lp-4  ] INFO 2023/05/02 16:14:46 car connected
[main  ] ERROR 2023/05/02 16:14:46 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:14:53 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:02 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:12 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:22 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:33 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:40 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:42 vehicle status: 408 Request Timeout
[lp-1  ] INFO 2023/05/02 16:15:44 vehicle updated: unknown -> MYLR
[lp-1  ] ERROR 2023/05/02 16:16:12 vehicle odometer: 408 Request Timeout
[lp-1  ] ERROR 2023/05/02 16:16:13 vehicle soc: 408 Request Timeout

It worked until i upgraded the firmware of my Tesla today.

### Steps to reproduce

1. upgrade of Telsa firmware
2.
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: tesla
  title: MYLR
  accessToken: ....
  refreshToken: ...
  vin: LRW....
  capacity: 82
  phases: 3 
  mode: off
  minSoC: 30 
  targetSoC: 75 
  minCurrent: 6 
  maxCurrent: 16 
  cloud: false 
  name: TMYLR
```

### Log details

```text
[main  ] ERROR 2023/05/02 16:14:32 vehicle status: 408 Request Timeout
[lp-4  ] INFO 2023/05/02 16:14:46 car connected
[main  ] ERROR 2023/05/02 16:14:46 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:14:53 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:02 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:12 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:22 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:33 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:40 vehicle status: 408 Request Timeout
[main  ] ERROR 2023/05/02 16:15:42 vehicle status: 408 Request Timeout
[lp-1  ] INFO 2023/05/02 16:15:44 vehicle updated: unknown -> MYLR
[lp-1  ] ERROR 2023/05/02 16:16:12 vehicle odometer: 408 Request Timeout
[lp-1  ] ERROR 2023/05/02 16:16:13 vehicle soc: 408 Request Timeout
```

### What type of operating system are you running?

Docker container

### Version

0.116.7

thiemog, 2023-05-02:
I experience the same behaviour.

evcc version: 0.116.7
Tesla Model Y Version: 2023.12.1.1

cschlipf, 2023-05-02:
Wake up your car. Access it using the Tesla App and try again. 408 says your car is currently sleeping.

andig, 2023-05-03:
At this time there's not much we can do since we don't wake up vehicles for retrieving soc. It will work when you've just connected your Tesla. Alterantively, you could use Teslamate and fetch soc from there using the `custom` integration. 

/cc @premultiply regarding explicit wakeup in certain situations.

closed 2023-05-03 by andig

cschlipf, 2023-05-03:
Do we really need an explicit wakeup? Starting the charging wakes the car up anyway. The only issue may be during the set up, but maybe you could add just an output like "If the car does not respond, please wake it up and try again."

Just my 2ct.

Explicit wakeups raise the danger of unnecessary wakeups and more battery drain.