evcc-io/evcc #7791: Ladeplanung wird nicht fortgesetzt und zeigt falschen SoC

Balujuhu, 2023-05-03:
### Describe the bug

Moin,

gefühlt tritt das Problem seit der Umstellung der Hyundai api auf.

Der alle Modi laufen, soweit ich das beurteilen kann, wie sie sollen.

Das einzige was bei beiden unserer Hyundai BEV nicht mehr klappt ist die Ladeplanung. Irgendwann stimmt der SoC den EVCC anzeigt aber nicht und die Ladung wird gestoppt weil das Ziel angeblich erreicht ist. Schaut man dann in der Bluelink App stimmt das aber nicht.

Wie bekomme ich das log denn für den Zeitpunkt wenn der Fehler auftritt? Stoppe ich den Service, wird das log erst ab diesem Zeitpunkt angefertigt. Auch stimmt mit stoppen des EVCC Dienstes und fertigen des log der SoC wieder in EVCC.

Habe auch schon getestet ob es an "ausstecken simulieren geht", aber das ändert nichts am Fehlerbild. 

SoC des Fahrzeugs:

[image]

Hier wurde vor stoppen des EVCC Dienstes 92% angezeigt, nach stoppen stimmte der SoC plötzlich und Start der Ladung macht ja auch keinen Sinn mehr:

[image]

### Steps to reproduce

1. Ladeplanung starten
2. irgendwann (es scheint nicht mehr weiter zu gehen, wenn EVCC eine "Pause" macht um günstigere Preise zu nutzen) stimmen die werte des SoC nicht mehr und die Ladung wird nicht fortgesetzt

### Configuration details

```yaml
[main  ] INFO 2023/05/03 05:16:21 evcc 0.116.7 (6773536d)
[main  ] INFO 2023/05/03 05:16:21 using config file: /etc/evcc.yaml
[db    ] INFO 2023/05/03 05:16:22 using sqlite database: /home/pi/.evcc/evcc.db

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 192.168.1.223  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: solaredge-inverter 
  id: 1  
  host: 192.168.1.223  
[... 172 lines cut ...]
[site  ] DEBUG 2023/05/03 05:12:45 pv power: 0W
[site  ] DEBUG 2023/05/03 05:12:45 grid power: 160W
[site  ] DEBUG 2023/05/03 05:12:45 site power: 160W
[lp-1  ] DEBUG 2023/05/03 05:12:45 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/03 05:12:45 charge meter total import: 2689.614kWh
[lp-1  ] DEBUG 2023/05/03 05:12:45 charger status: B
[lp-1  ] DEBUG 2023/05/03 05:12:45 pv charge current: 0A = 0A + -0.696A (160W @ 1p)
[site  ] DEBUG 2023/05/03 05:12:55 ----
[lp-1  ] DEBUG 2023/05/03 05:12:55 charge power: 0W
[site  ] DEBUG 2023/05/03 05:12:55 pv power: 0W
[site  ] DEBUG 2023/05/03 05:12:55 grid power: 162W
[site  ] DEBUG 2023/05/03 05:12:55 site power: 162W
[lp-1  ] DEBUG 2023/05/03 05:12:55 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/03 05:12:55 charge meter total import: 2689.614kWh
[lp-1  ] DEBUG 2023/05/03 05:12:55 charger status: B
[lp-1  ] DEBUG 2023/05/03 05:12:55 pv charge current: 0A = 0A + -0.704A (162W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.7 (6773536d)

Hofyyy, 2023-05-03:
Moin,

hast du die chance das noch einmal mit dem loglevel trace für dem Lodpoint zu ziehen?

Dann gucke ich mal im detail rein.

Balujuhu, 2023-05-03:
Wenn du mir jetzt sagst was du genau meinst, könnt ich dir das beantworten 😂

ich hangel mich nur mit copy/paste befehlen da durch :D

Balujuhu, 2023-05-03:
Also ich ersetze 

log: info
levels:
  cache: error 

durch 

log: debug
levels:
lp-1: trace
core: debug

Oder füge ich das komplett hinzu?

Hofyyy, 2023-05-03:
Unter levels fügst du eigentlich nur die Zeile mit lp-1: trace ein "zusätzlich"

Balujuhu, 2023-05-04:
Die Nacht wieder passiert. Die Ladung fängt an wenn die Planung es vorsieht und mitten in der Nacht ist dann Ende und EVCC zeigt einen falschen SoC an. 

Log sieht so aus (habe die obige Änderung drin):

[main  ] INFO 2023/05/04 05:29:20 evcc 0.116.7 (6773536d)
[main  ] INFO 2023/05/04 05:29:20 using config file: /etc/evcc.yaml
[main  ] INFO 2023/05/04 05:29:20 starting ui and api at :7070
[db    ] INFO 2023/05/04 05:29:21 using sqlite database: /home/pi/.evcc/evcc.db
[site  ] INFO 2023/05/04 05:29:25 site config:
[site  ] INFO 2023/05/04 05:29:25   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/05/04 05:29:25     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/04 05:29:25     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/04 05:29:25   vehicles:
[site  ] INFO 2023/05/04 05:29:25     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] INFO 2023/05/04 05:29:25     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/05/04 05:29:25 loadpoint 1:
[lp-1  ] INFO 2023/05/04 05:29:25   mode:        pv
[lp-1  ] INFO 2023/05/04 05:29:25   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/05/04 05:29:25   meters:      charge ✓
[lp-1  ] INFO 2023/05/04 05:29:25     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/05/04 05:29:25 phase timer inactive
[lp-1  ] DEBUG 2023/05/04 05:29:25 pv timer inactive
[lp-1  ] DEBUG 2023/05/04 05:29:25 guard timer inactive
[site  ] DEBUG 2023/05/04 05:29:25 ----
[lp-1  ] DEBUG 2023/05/04 05:29:25 charge power: 0W
[site  ] DEBUG 2023/05/04 05:29:25 pv power: 0W
[site  ] DEBUG 2023/05/04 05:29:25 grid power: 187W
[site  ] DEBUG 2023/05/04 05:29:25 site power: 187W
[lp-1  ] DEBUG 2023/05/04 05:29:25 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/04 05:29:25 charge meter total import: 2717.925kWh
[lp-1  ] DEBUG 2023/05/04 05:29:25 charger status: B
[lp-1  ] INFO 2023/05/04 05:29:25 car connected
[lp-1  ] DEBUG 2023/05/04 05:29:25 vehicle api refresh
[lp-1  ] DEBUG 2023/05/04 05:29:25 pv timer elapse
[lp-1  ] DEBUG 2023/05/04 05:29:25 pv timer inactive
[lp-1  ] DEBUG 2023/05/04 05:29:25 charger: guard elapse
[lp-1  ] DEBUG 2023/05/04 05:29:25 guard timer inactive
[lp-1  ] DEBUG 2023/05/04 05:29:25 charge meter total import: 2717.925kWh
[main  ] DEBUG 2023/05/04 05:29:43 vehicle status: B (Ioniq)
[main  ] DEBUG 2023/05/04 05:30:07 vehicle status: A (Kona)
[lp-1  ] INFO 2023/05/04 05:30:07 vehicle updated: unknown -> Ioniq
[lp-1  ] DEBUG 2023/05/04 05:30:07 set charge mode: pv
[lp-1  ] DEBUG 2023/05/04 05:30:07 set min current: 6
[lp-1  ] DEBUG 2023/05/04 05:30:07 set max current: 10
[lp-1  ] DEBUG 2023/05/04 05:30:07 set min soc: 40
[lp-1  ] DEBUG 2023/05/04 05:30:07 set target soc: 80
[lp-1  ] DEBUG 2023/05/04 05:30:07 vehicle status: B
[lp-1  ] DEBUG 2023/05/04 05:30:07 vehicle soc: 66%
[lp-1  ] DEBUG 2023/05/04 05:30:07 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/05/04 05:30:07 vehicle range: 179km
[lp-1  ] DEBUG 2023/05/04 05:30:07 pv charge current: 0A = 0A + -0.813A (187W @ 1p)
[lp-1  ] DEBUG 2023/05/04 05:30:07 pv enable timer reset
[lp-1  ] DEBUG 2023/05/04 05:30:07 pv timer inactive
[site  ] DEBUG 2023/05/04 05:30:07 ----
[lp-1  ] DEBUG 2023/05/04 05:30:08 charge power: 0W
[site  ] DEBUG 2023/05/04 05:30:08 pv power: 0W
[site  ] DEBUG 2023/05/04 05:30:08 grid power: 189W
[site  ] DEBUG 2023/05/04 05:30:08 site power: 189W
[lp-1  ] DEBUG 2023/05/04 05:30:08 vehicle odometer: 20055km
[lp-1  ] DEBUG 2023/05/04 05:30:08 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/04 05:30:08 charge meter total import: 2717.925kWh
[lp-1  ] DEBUG 2023/05/04 05:30:08 charger status: B
[lp-1  ] DEBUG 2023/05/04 05:30:08 pv charge current: 0A = 0A + -0.822A (189W @ 1p)
[site  ] DEBUG 2023/05/04 05:30:08 ----
[lp-1  ] DEBUG 2023/05/04 05:30:08 charge power: 0W
[site  ] DEBUG 2023/05/04 05:30:08 pv power: 0W
[site  ] DEBUG 2023/05/04 05:30:08 grid power: 192W
[site  ] DEBUG 2023/05/04 05:30:08 site power: 192W
[lp-1  ] DEBUG 2023/05/04 05:30:08 charge currents: [0 0 0]A

Hofyyy, 2023-05-04:
Bitte das log als txt hochladen. wie du siehst ist der auschnitt nur einige Sekunden und nicht die nacht. Da sehe ich leider nichts :-/

Balujuhu, 2023-05-04:
Wird das irgendwo gespeichert oder muss ich das mit einem anderen Befehl selbst anlegen?

VolkerK62, 2023-05-04:
@Balujuhu 
https://docs.evcc.io/docs/guides/setup#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen
Schau mal hier, du kannst sogar einen genauen Zeitraum angeben

Balujuhu, 2023-05-05:
Moin,
habe gestern das log per Start und Endzeit eingerichtet und hoffe das hat geklappt. 

Sitze gerade auf der Arbeit und könnte per VPN auf den raspberry zugreifen. Wie kann ich mir die log Datei anzeigen lassen damit ich den Inhalt hier posten kann?

VolkerK62, 2023-05-05:
wenn du das so gemacht hast
`sudo journalctl -u evcc -S "2023-03-21 07:00:00" -U "2023-03-21 08:00:00" -q > ~/evcc.log`
dann wird die Datei "evcc.log" im Home-Verzeichnis (normal "/pi/home") gespeichert.

Balujuhu, 2023-05-05:
Hm, wenn ich
sudo nano pi/home/evcc.log 

im Terminal eingebe, kommt die Meldung das dass Verzeichnis pi/Home nicht existiert nicht. 

Hab eine ganz normale raspberry pi Installation und nur evcc da drauf am laufen.

VolkerK62, 2023-05-05:
sorry, "/home/pi" 

aber (hab es gerade ausprobiert) das brauchst du gar nicht, einfach "sudo nano evcc.log"

Balujuhu, 2023-05-05:
Irgendwas scheint da nicht geklappt zu haben. Ist ein leerer Inhalt -.-

Als ich das gestern eingegeben habe hat es zwei drei Sekunden geauert und dann kam die Eingabezeile wieder. Bin davon ausgegangen das es dann geklappt hat mit dem "stellen" des Timers für das log.

Balujuhu, 2023-05-05:
Ok, das macht Sinn :D Hoffe damit kann jemand was anfangen:

Last login: Fri May  5 16:09:04 on ttys000
thomasfabricius@Air-von-Thomas ~ % ssh pi@192.168.1.60
pi@192.168.1.60's password: 
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri May  5 15:09:09 2023 from 192.168.1.63

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

pi@raspberrypi:~ $ sudo journalctl -u evcc -S "2023-05-04 23:59:00" -U "2023-05-05 05:30:00" -q > ~/evcc.log
pi@raspberrypi:~ $ sudo nano evcc.log

  GNU nano 5.4                                                                             evcc.log                                                                                      
Mai 05 00:03:42 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:03:42 planned 4h6m9s until 2023-05-05 07:30:00 +0200 CEST at 2300W: total plan duration: 4h6m9s, avg cost: 0.251
Mai 05 00:03:42 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:42   slot from: 2023-05-05 01:00:00 +0200 CEST to 2023-05-05 02:00:00 +0200 CEST cost 0.253
Mai 05 00:03:42 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:42   slot from: 2023-05-05 02:00:00 +0200 CEST to 2023-05-05 03:00:00 +0200 CEST cost 0.249
Mai 05 00:03:42 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:42   slot from: 2023-05-05 03:00:00 +0200 CEST to 2023-05-05 04:00:00 +0200 CEST cost 0.250
Mai 05 00:03:42 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:42   slot from: 2023-05-05 04:00:00 +0200 CEST to 2023-05-05 05:00:00 +0200 CEST cost 0.249
Mai 05 00:03:42 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:42   slot from: 2023-05-05 05:00:00 +0200 CEST to 2023-05-05 05:06:09 +0200 CEST cost 0.260
Mai 05 00:03:42 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:03:42 pv charge current: 0A = 0A + -0.826A (190W @ 1p)
Mai 05 00:03:51 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:03:51 charge power: 0W
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:03:52 charge currents: [0 0 0]A
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:03:52 charge meter total import: 2727.474kWh
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:03:52 charger status: B
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:03:52 planned 4h6m9s until 2023-05-05 07:30:00 +0200 CEST at 2300W: total plan duration: 4h6m9s, avg cost: 0.251
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:52   slot from: 2023-05-05 01:00:00 +0200 CEST to 2023-05-05 02:00:00 +0200 CEST cost 0.253
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:52   slot from: 2023-05-05 02:00:00 +0200 CEST to 2023-05-05 03:00:00 +0200 CEST cost 0.249
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:52   slot from: 2023-05-05 03:00:00 +0200 CEST to 2023-05-05 04:00:00 +0200 CEST cost 0.250
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:52   slot from: 2023-05-05 04:00:00 +0200 CEST to 2023-05-05 05:00:00 +0200 CEST cost 0.249
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:03:52   slot from: 2023-05-05 05:00:00 +0200 CEST to 2023-05-05 05:06:09 +0200 CEST cost 0.260
Mai 05 00:03:52 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:03:52 pv charge current: 0A = 0A + -0.822A (189W @ 1p)
Mai 05 00:04:01 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:01 charge power: 0W
Mai 05 00:04:01 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:01 charge currents: [0 0 0]A
Mai 05 00:04:01 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:01 charge meter total import: 2727.474kWh
Mai 05 00:04:02 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:02 charger status: B
Mai 05 00:04:02 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:02 planned 4h6m9s until 2023-05-05 07:30:00 +0200 CEST at 2300W: total plan duration: 4h6m9s, avg cost: 0.251
Mai 05 00:04:02 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:04:02   slot from: 2023-05-05 01:00:00 +0200 CEST to 2023-05-05 02:00:00 +0200 CEST cost 0.253
Mai 05 00:04:02 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:04:02   slot from: 2023-05-05 02:00:00 +0200 CEST to 2023-05-05 03:00:00 +0200 CEST cost 0.249
Mai 05 00:04:02 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:04:02   slot from: 2023-05-05 03:00:00 +0200 CEST to 2023-05-05 04:00:00 +0200 CEST cost 0.250
Mai 05 00:04:02 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:04:02   slot from: 2023-05-05 04:00:00 +0200 CEST to 2023-05-05 05:00:00 +0200 CEST cost 0.249
Mai 05 00:04:02 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:04:02   slot from: 2023-05-05 05:00:00 +0200 CEST to 2023-05-05 05:06:09 +0200 CEST cost 0.260
Mai 05 00:04:02 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:02 pv charge current: 0A = 0A + -0.839A (193W @ 1p)
Mai 05 00:04:11 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:11 charge power: 0W
Mai 05 00:04:11 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:11 charge currents: [0 0 0]A
[... cut ...]
Mai 05 00:04:12 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:04:12   slot from: 2023-05-05 04:00:00 +0200 CEST to 2023-05-05 05:00:00 +0200 CEST cost 0.249
Mai 05 00:04:12 raspberrypi evcc[2212]: [lp-1  ] TRACE 2023/05/05 00:04:12   slot from: 2023-05-05 05:00:00 +0200 CEST to 2023-05-05 05:06:09 +0200 CEST cost 0.260
Mai 05 00:04:12 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:12 pv charge current: 0A = 0A + -0.822A (189W @ 1p)
Mai 05 00:04:21 raspberrypi evcc[2212]: [lp-1  ] DEBUG 2023/05/05 00:04:21 charge power: 0W

VolkerK62, 2023-05-06:
So wie du eingangs schreibst, wird geladen und irgendwann stoppt es, weil der Fahrzeug-Soc nicht mehr stimmt.
Kannst du herausfinden (evtl übers Ladelog) wann gestoppt wurde?
Der Zeitbereich im Log wäre dann interessant.

Balujuhu, 2023-05-06:
Irgendwann stimmt der angerufene SoC nicht mehr mit dem im Fahrzeig überein, genau. Das müsste an dem Tag aus dem das log stammt eigentlich mit drin sein. 

Letzte Nacht hat es komischerweise problemlos geklappt. 

Evtl hatte Bluelink auch ein Problem über Tage hinweg?

VolkerK62, 2023-05-06:
Das Log zeigt ja nur eine Minute kurz nach Mitternacht. Da wurde ja noch gar nicht geladen.
Laut Planner sollte die Ladung um 3 Uhr starten.
Und zig Std Log durchzusehen um den Zeitpunkt zu finden, wo die Ladung stoppt ist nicht prickelnd ;-). 
Deshalb meine Frage von oben, wann wurde die Ladung gestoppt? (evcc Ladelog, oder go-e App, oder Bluelink ...)

Balujuhu, 2023-05-07:
Moin,

die Nacht wieder passiert, nur ist das nicht wirklich leicht den Zeitpunkt zu finden. Auto hatte 85% SoC und wollte mit dem Planer um 2Uhr anfangen auf 100% SoC zu laden. Ist nur nicht passiert. War lt EVCC UI wieder mehr SoC als es tatsächlich war. 

Last login: Sun May  7 08:29:52 on ttys000
ssh pi@192.168.1.60
 ~ % ssh pi@192.168.1.60
pi@192.168.1.60's password: 
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri May  5 16:09:08 2023 from 192.168.1.63

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

pi@raspberrypi:~ $ sudo journalctl -u evcc -S "2023-05-07 01:40:00" -U "2023-05-07 02:30:00" -q > ~/evcc.log
pi@raspberrypi:~ $ sudo nano evcc.log

  GNU nano 5.4                                                                            evcc.log                                                                                      
Mai 07 02:14:19 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:19 charger status: B
Mai 07 02:14:19 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:19 charger: guard elapse
Mai 07 02:14:19 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:19 guard timer inactive
Mai 07 02:14:19 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:19 charger enable
Mai 07 02:14:19 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:19 wake-up timer: start
Mai 07 02:14:23 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:23 charge power: 70W
Mai 07 02:14:23 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:23 charge currents: [0.7 0 0]A
Mai 07 02:14:23 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:23 charge meter total import: 2775.362kWh
Mai 07 02:14:23 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:23 charger status: C
Mai 07 02:14:23 raspberrypi evcc[531]: [lp-1  ] INFO 2023/05/07 02:14:23 start charging ->
Mai 07 02:14:23 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:23 wake-up timer: stop
Mai 07 02:14:24 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:24 soc estimated: 100.00% (vehicle: 85.00%)
Mai 07 02:14:24 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:24 vehicle soc: 100%
Mai 07 02:14:24 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:24 vehicle soc limit: 100%
Mai 07 02:14:24 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:24 vehicle range: 282km
Mai 07 02:14:33 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:33 charge power: 2200W
Mai 07 02:14:33 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:33 charge currents: [9.3 0 0]A
Mai 07 02:14:33 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:33 detected active phases: 1p
Mai 07 02:14:33 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:33 charge meter total import: 2775.366kWh
Mai 07 02:14:33 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:33 charger status: C
Mai 07 02:14:34 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:34 soc estimated: 100.00% (vehicle: 85.00%)
Mai 07 02:14:34 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:34 vehicle soc: 100%
Mai 07 02:14:34 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:34 vehicle soc limit: 100%
Mai 07 02:14:34 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:34 vehicle range: 282km
Mai 07 02:14:43 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:43 charge power: 2190W
Mai 07 02:14:43 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:43 charge currents: [9.3 0 0]A
Mai 07 02:14:43 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:43 detected active phases: 1p
Mai 07 02:14:43 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:43 charge meter total import: 2775.373kWh
Mai 07 02:14:44 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:44 charger status: C
Mai 07 02:14:44 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:44 soc estimated: 100.00% (vehicle: 85.00%)
Mai 07 02:14:44 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:44 vehicle soc: 100%
Mai 07 02:14:44 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:44 vehicle soc limit: 100%
Mai 07 02:14:44 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:44 vehicle range: 282km
Mai 07 02:14:53 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:53 charge power: 2180W
Mai 07 02:14:53 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:53 charge currents: [9.2 0 0]A
Mai 07 02:14:53 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:53 detected active phases: 1p
Mai 07 02:14:54 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:54 charge meter total import: 2775.379kWh
Mai 07 02:14:54 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:54 charger status: C
Mai 07 02:14:54 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:54 soc estimated: 100.00% (vehicle: 85.00%)
Mai 07 02:14:54 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:54 vehicle soc: 100%
Mai 07 02:14:54 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:14:54 vehicle soc limit: 100%

Hofyyy, 2023-05-07:
Lad mal bitte das ganze log als txt hoch. es ist wichtig zu sehen wo sich ggf. verrechnet wird. Wo bekommt er den SoC denn her?

Balujuhu, 2023-05-07:
@Hofyyy 

Passt das so?
[evcc.txt](https://github.com/evcc-io/evcc/files/11414196/evcc.txt)

Hofyyy, 2023-05-07:
Also da läuft wirklich was schief.

Initial Plan:
 02:00:04 planned 3h16m4s until 2023-05-07 07:30:00 +0200 CEST at 2300W: total plan duration: 3h16m4s
02:00:04   slot from: 2023-05-07 02:00:05 +0200 CEST to 2023-05-07 03:00:00 +0200 CEST cost 0.254
02:00:04   slot from: 2023-05-07 04:00:00 +0200 CEST to 2023-05-07 05:00:00 +0200 CEST cost 0.254
02:00:04   slot from: 2023-05-07 06:00:00 +0200 CEST to 2023-05-07 06:46:09 +0200 CEST cost 0.255
02:00:04   slot from: 2023-05-07 07:00:00 +0200 CEST to 2023-05-07 07:30:00 +0200 CEST cost 0.253
02:00:04 max charge current: 10A

Dann Ladestart:
Mai 07 02:00:14 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:00:14 charge power: 2200W
Mai 07 02:00:14 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:00:14 charge currents: [9.3 0 0]A
Mai 07 02:00:14 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:00:14 charge meter total import: 2775.351kWh
Mai 07 02:00:14 raspberrypi evcc[531]: [lp-1  ] DEBUG 2023/05/07 02:00:14 charger status: C

Dann wird der gradient geupdated und alles geht den Bach runter.

02:00:29 soc gradient updated: soc: 85.0%, socDiff: 13.0%, energyDiff: 2Wh, energyPerSocStep: 0.2Wh, virtualCapacity: 19Wh
02:00:29 vehicle soc: 85%
02:00:29 vehicle soc limit: 100%
02:00:29 vehicle range: 282km
02:00:29 planned 5s until 2023-05-07 07:30:00 +0200 CEST at 2300W: total plan duration: 5s, avg cost: 0.253
02:00:29   slot from: 2023-05-07 07:00:00 +0200 CEST to 2023-05-07 07:00:05 +0200 CEST cost 0.253
02:00:29 continuing until end of slot
02:00:29 charge power: 2200W
02:00:30 charge currents: [9.4 0 0]A
02:00:30 detected active phases: 1p
02:00:30 charge meter total import: 2775.360kWh
02:00:30 charger status: C
02:00:30 soc estimated: 100.00% (vehicle: 85.00%)
02:00:30 vehicle soc: 100%

Fragen:
- Was ist das für ein Fahrzeug? Welche Kapazität ist eingestellt?

Balujuhu, 2023-05-07:
Ein Hyundai Ioniq und ein Kona. Bei beiden hab ich 39 als Kapazität eingestellt.

premultiply, 2023-05-07:
Die Fahrzeug-API meldet erst einen falschen SoC, dann kurz nach Ladestart erst den richtigen?

`estimate: false` sollte das Problem umgehen.

Balujuhu, 2023-05-07:
Was die API meldet weiß ich nicht, kenn mich da nicht aus. Jedenfalls zeigt EVCC irgendwann was anderes an als wenn ich über die Bluelink App von Hyundai gucke. 

was wie und wo mache ich mit estimate false? :D

VolkerK62, 2023-05-07:
Du fügst das (`estimate: false`) deinem Loadpoint hinzu. Einrückungen beachten.
```
mincurrent: 6
maxcurrent: 10
resetOnDisconnect: true
soc:
  estimate: false
```

Balujuhu, 2023-05-07:
Hab es jetzt hinzugefügt, werde berichten. 

Aber warum tritt das Problem plötzlich auf? Und halt auch nicht immer?