evcc-io/evcc #7796: template: tinkerforge-warp um standbypower ergänzen

masterdot, 2023-05-03:
Beim entdecken weiterer Funktionen bin ich über die Shelly Integration gestolpert.
Hier kann man den Eigenverbrauch durch Standby festlegen. Bei der Warp Wallbox habe ich, wenn das Auto angesteckt ist, immer die Anzeige von 2 W.

Wäre vielleicht eine Idee das anzupassen?

andig, 2023-05-03:
/cc @naltatis kleine Leistungen im UI ignorieren falls Status <> C? Physisch sind sie da.

mdkeil, 2023-05-03:
Sieht bei mir ähnlich aus, nur dass meine WB (Hardy Barth Salia) sich 5W gönnt ;) was auch angezeigt wird, wenn kein Auto angeschlossen ist.

VolkerK62, 2023-05-03:
Ähnliches gab es beim Eigenverbrauch des Batterie-WR (der aktuell angezeigt wird)
Als der ignoriert wurde, gab es Hinweise/"Beschwerden", dass `IN` und `OUT` nicht gleich wären.
Um es ganz richtig zu machen, müsste man diese ignorierte Leistung dann zum normalen Verbrauch hinzurechnen.

andig, 2023-05-03:
Guter Punkt. Das wäre dann https://github.com/evcc-io/evcc/issues/7803

closed 2023-05-04 by andig