evcc-io/evcc #7800: Fehlermeldung

spizzl510, 2023-05-03:
[image]

was sagt der Fehler aus? Kann mir eventuell jemand weiterhelfen.

andig, 2023-05-04:
Vmtl. war das Cloud-API für Community Livedaten gerade kurz offline- kannst Du einfach ignorieren.

/cc @naltatis sollten wir die Meldungen vllt. ins Debug schieben?

closed 2023-05-04 by andig

naltatis, 2023-05-04:
@andig würd ungern in der UI nach der Meldung filtern. Besser wäre die Meldung gar nicht erst hochzuspielen bzw. vielleicht ne Art Threshold/Häufigkeit zu haben bevor wir die rausgeben. So ein Mechanismus könnte auch bei anderen Meldungen (out of sync, tesla wakeup, ...) sinnvoll sein bzw. das Rauschen in der UI verringern.

andig, 2023-05-04:
> Besser wäre die Meldung gar nicht erst hochzuspielen

Das war ja meine Frage:

> sollten wir die Meldungen vllt. ins Debug schieben?

Die einfache Lösung wäre: ab ins Debug.