evcc-io/evcc #7801: mqtt status topic sporadically went to offline and receive no updates, other topics are still updated without a problem

eldrik, 2023-05-03:
### Describe the bug

As of version 0.114 maybe some version numbers earlier the evcc mqtt status sporadically wents to offline but the application is still running and other mqtt topics are still updated.

### Steps to reproduce

1. Start evcc
2. Wait for days sometimes one or two weeks
3. mqtt topic goes to offline while application still online and other mqtt topics are updated
...

### Configuration details

```yaml
nothing special here

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7080

log: info
levels:
  cache: error
#  proxy-5200: trace
# unique installation id
plant: 37116dbc07c1e508a3730dbd3a94e1d603da226e639d9daebc1832d03118c510

interval: 10s # control cycle interval

sponsortoken: ***

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

#modbusproxy:
#   - port: 5200
#     uri: 10.0.81.40:502
mqtt:
  broker: localhost:1883
  topic: evcc

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 2
  host: 10.0.81.40
  port: 502
  usage: grid
  storageunit: 1
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 2
  host: 10.0.81.40
  port: 502
  usage: pv
  storageunit: 1
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 2
  host: 10.0.81.40
  port: 502
  usage: battery
  storageunit: 1
  modbus: tcpip
  name: battery3
- type: template
  template: huawei-dongle
  id: 3
  host: 10.0.81.40
  port: 502
  usage: pv
  modbus: tcpip
  name: pv5
- type: template
  template: huawei-dongle
  id: 1
  host: 10.0.81.40
  port: 502
  usage: pv
  modbus: tcpip
  name: pv6
- name: pv7
  type: custom
  power:
   source: mqtt
   broker: localhost:1883
   topic: Balkonkraftwerk/power
   scale: 1000
chargers:
- type: template
  template: keba
  host: 10.0.81.219
  name: wallbox7

loadpoints:
- title: Garage
  charger: wallbox7
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    - pv5
    - pv6
    - pv7
#    battery:
#    - battery3
```

### Log details

```text
...
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.3

andig, 2023-05-04:
Unfortunately, nothing we could do without logfile or chance to look at the problem when it happens :(

closed 2023-05-04 by andig

occtopus, 2023-06-06:
@andig I have the exact same problem. I could see in the logs that this happens when there was a reconnect of my wifi. This results in the syslog message
`avahi-daemon[666]: Registering new address record for 192.168.1.119 on wlan0.IPv4.`
This is followed by the following entries and from there I have the same behavior as described in this issue.

Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/site/homePower: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/chargeCurrent: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/site/greenShare: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/sessionCo2PerKWh: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/sessionEnergy: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/chargeDuration: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/sessionPricePerKWh: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/chargedEnergy: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/charging: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/site/tariffEffectivePrice: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/connected: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/loadpoints/1/enabled: timeout
Jun  5 04:52:23 evcc evcc[737]: [mqtt  ] ERROR 2023/06/05 04:52:23 send: evcc/site/warn: timeout

andig, 2023-06-06:
Neues Issue. Erstens ist der Fehler anders, zweitens fehlen alle relevanten Daten für ein Ticket.