evcc-io/evcc #7805: Buffer SoC leads to unwanted behavior and uncharges home battery at the end of sunny time

Schollino, 2023-05-03:
**Is your feature request related to a problem? Please describe.**
Hi altogether,
I set the bufferSoc of my battery to 80 % to guarantee, that the vehicle loading in PV-mode continues, if there is a cloud or a short time without excess energy. Unfortunately, at the end of the "sunny time", the battery unloads from 100 % to 80 % before the charging of the car stops. I think this is an unwanted behavior, because the whole battery-energy should prior be used for the house consumption at night. 

**Describe the solution you'd like**
I think there should be a "logic" that realizes, that the "sunny time" is going to end very soon and stops the vehicle charging, as soon as there is not enough energy excess to guarantee the min charging power anymore and the battery is still at 100%.

**Describe alternatives you've considered**
The only alternative at the moment is to stop charging manually.

**Additional context**
[image]

TTXaser2, 2023-05-03:
Correct, I also experienced the exact same behaviour.

thebrainkafka, 2023-05-04:
This is the designed way EVCC works.
No bug. Works fine.

TTXaser2, 2023-05-04:
If the goal is to maximize the use of Green Energy then I would definitely NOT declare this behaviour as "works fine"! 
Because in  my case with a relatively small battery this leads to using more energy from the grid at night than it would be necessary  with a full battery at sunset. 20% battery capacity to be exact.

thebrainkafka, 2023-05-04:
Use BufferSoc to adjust based on your needs.

TTXaser2, 2023-05-04:
> Use BufferSoc to adjust based on your needs.

Sorry, this kind of discussion is not productive. It is not practicable to change the config every day at 17 o'clock or in summer at 19 o'clock only to be able to use the Battery during the day as a buffer and filling it at the end of the day.
Following your arguments EVCC is completely unnecessary because you can instead manually change the charging settings of the wallbox depending on the actual available PV surplus. EVCC, in my perception, is there to automate these tasks. The same I would expect with the topic of this thread.

andig, 2023-05-04:
There is already excessive discussion of `bufferSoc` in https://github.com/evcc-io/evcc/discussions/7191, let's continue there in one place. Seems this issue adds a new facette to the discussion.

> Unfortunately, at the end of the "sunny time", the battery unloads from 100 % to 80 % before the charging of the car stops.

closed 2023-05-04 by andig

thebrainkafka, 2023-05-04:
> Sorry, this kind of discussion is not productive. It is not practicable to change the config every day at 17 o'clock or in summer at 19 o'clock only to be able to use the Battery during the day as a [...]

Was not meant as a every day change, but once or e.g summer/winter to get the best results for your needs.

IstNurJunior, 2023-05-10:
Have a look at #7802. This will solve most cases. 
Another way could be to use residualPower with a small value (eg CapBat/SunsetHours). Keep in mind that big values will lead to grid in.