evcc-io/evcc #7813: 3 phase power calculation is wrong.

Roeland54, 2023-05-04:
In the following code the calculation for 3 phase power is multiplying by 3. This is a common misconception to calculate the correct power for a 3 phase load you need to multiply with the square root of 3. 

This is the formula:
P = I * V * 1.73

example if you use when using 6A:
6A * 230V * 1.73 = 2.387kW

Multiplying by 3 results in 4.140kW which is wrong.

https://github.com/evcc-io/evcc/blob/ea314ed11881cfc319ae58704153ddfc4b5cf2ad/assets/js/components/LoadpointSettingsModal.vue#LL212C1-L217C5

closed 2023-05-04 by premultiply

Roeland54, 2023-05-05:
No feedback? Am I wrong in this?

premultiply, 2023-05-05:
This issue form is for reporting issues only! If you want to start a discussion or ask a question, please use the [discussion forum](https://github.com/evcc-io/evcc/discussions).

NOTE Due to amount of issues raised we will close incomplete issues (unclear description what the issue actually is, missing log file, unsupported platform) without further comments.