evcc-io/evcc #7817: Tariff Planner: Request failed with status code 400

abc2006, 2023-05-04:
### Describe the bug

Hab die anderen Issues gelesen. Bei dem einen war ein / zuviel, bei dem anderen war tibber nicht erreichbar. 
Aber: Ich habe die gleiche Fehlermeldung und bei mir ist tibber gar nicht konfiguriert. 
Was mach ich falsch? Vielleicht hab ich was falsch verstanden... 

Danke, 
Stephan 

### Steps to reproduce

1. Ladezeitplan setzen (Uhrzeit angeben)
2. Request failed with status code 400: API request failed http://192.168.0.118:7070/api/tariff/planner

### Configuration details

```yaml
http://192.168.0.118:7070/api/tariff/planner 
liefert als Antwort
{"error":"tariff not available"}

Klar, hab ich nicht konfiguriert. Ich will ja nur, dass morgen um 9 mein Auto geladen ist - egal mit welchem Strom und egal zu welchem Preis.
```

### Log details

```text
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 ----
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:01 charge power: 1392W
evcc    | [lp-2  ] DEBUG 2023/05/04 22:29:01 charge power: 0W
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 pv power: 0W
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 battery 1 power: 1957W
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 battery 1 soc: 79%
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 battery 2 power: 774W
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 battery 2 soc: 90%
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 battery soc: 85%
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 battery power: 2731W
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 grid power: 3W
evcc    | [site  ] DEBUG 2023/05/04 22:29:01 site power: 2834W
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 charge currents: [6.12 0 0]A
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 detected active phases: 1p
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 charge meter total import: 7418.271kWh
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 charger status: C
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 soc estimated: 57.07% (vehicle: 57.00%)
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 vehicle soc: 57%
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 vehicle soc limit: 90%
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 vehicle range: 325km
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 pv charge current: 0A = 6A + -12.3A (2834W @ 1p)
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 site power 2834W >= 0W disable threshold
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 pv disable in 0s
evcc    | [lp-1  ] DEBUG 2023/05/04 22:29:02 pv disable timer remaining: 0s
evcc    | [site  ] DEBUG 2023/05/04 22:29:11 ----
```

### What type of operating system are you running?

Linux

### Version

0.116.7

Hofyyy, 2023-05-04:
Ohne das ich es wirklich weiss:
- Du konntest einen Tarif nehmen, der auf den CO2 Daten beruht. Dann Lädst du, wenn der Strom am besten für die Gesellschaft ist und das Auto ist trotzdem um 8 Uhr voll.
- Oder ich würde tippen es gibt einen zweiten API Call z.B. ohne Tariff, das ist aber nur geraten.

Hofyyy, 2023-05-04:
Beispiel:

```
planner:
  type: grünstromindex
  zip: meine PLZ # PLZ mit führender Null mit in "" setzen
```

Sillium007, 2023-05-05:
@abc2006: Deine Konfig wäre hilfreich.

andig, 2023-05-05:
@abc2006 benutzt Du das UI oder machst Du das "von han". Falls Letzteres: was für einen Request schickst Du denn? 
@Hofyyy der planner sollte eigentlich mit grid Tarif funktionieren wenn sonst nix gesetzt ist.

abc2006, 2023-05-05:
> @abc2006 benutzt Du das UI oder machst Du das "von han". Falls Letzteres: was für einen Request schickst Du denn? 
Ich bin mir nicht sicher, ob ich den Unterschied zwischen "UI" und "von Hand" richtig interpretiere: 
Ich bin auf der Webseite und klicke auf - Tja. leider ist das Auto gerade nicht verbunden.
[image]
Hier in der Mitte unter dem Schieberegler ist ein Link, und wenn ich darauf klicke, kann ich eine Uhrzeit und ein Datum einstellen. 
Ist das "UI" und beantwortet deine Frage? 

@Hofyyy der planner sollte eigentlich mit `grid` Tarif funktionieren wenn sonst nix gesetzt ist. Dafür ist keine Config erforderlich.
Das hätte ich vermutet, deswegen habe ich das so probiert. Wenn das falsch ist, klärt mich doch bitte auf, wo ich die Doku dazu finde - war nämlich erfolglos. 
Config kommt gleich, moment.

abc2006, 2023-05-05:
[evcc.txt](https://github.com/evcc-io/evcc/files/11404354/evcc.txt)

Hier ist die Config, 
Auch hier wieder mein Vorschlag, .yaml-Dateien zum Upload zu erlauben.

Ach ja, vielen Dank für die schnelle und rege Beteiligung/Unterstützung/Hilfe.
Ich schaue mal nach der Doku für tarif/grid - kann aber erst testen, wenn das Auto wieder daheim ist ;) 

Grüße, 
Stephan

andig, 2023-05-05:
> Ich bin auf der Webseite und klicke auf - Tja. leider ist das Auto gerade nicht verbunden.

Alles klar, du verwendest kein `curl` etc. Deine Datei hab ich wieder gelöscht weil Sposortoken etc im Klartext drin waren.

Was ich nicht sehe ist irgendein konfigurierter Tarif; ich vermute es liegt daran. Natürlich sollte es trotzdem möglich sein, das Auto zum Termin zu laden.

/cc @naltatis

andig, 2023-05-05:
@abc2006 

> Habe jetzt einen Tarif hinzugefügt, werde später mal testen, wie das läuft.

Wie gesagt es sollte ohne gehen.

> Request failed with status code 400

Sag uns bitte nochmal *wo genau* Du diesen Fehler siehst. Nur in der Browser Debugkonsole oder irgendwo sonst?

@naltatis ich sehe:

[image]

...die Planung funktioniert aber. Falls hier nur der Fehler in der Konsole stört- haben wir eine Chance, den vllt. durch Handling noch zu verstecken?

naltatis, 2023-05-05:
Auch ohne Tariff sollten wir hier die geplanten Slots sehen. Ich hab mal ein paar kl. Anpassungen gemacht.

abc2006, 2023-05-06:
> @abc2006
> Sag uns bitte nochmal _wo genau_ Du diesen Fehler siehst. Nur in der Browser Debugkonsole oder irgendwo sonst?

Ich hab grade nur zugriff auf die demo-instanz: 
[image]
Oben rechts neben den drei Strichen hab ich einen kringel gemacht. Dort erscheint ein rotes Dreieck mit ausrufezeichen und wenn ich drauf klicke, sehe ich die Fehlermeldungen. 
Ob man den Fehler noch irgendwo anders sieht, weiss ich gerade nicht.

andig, 2023-05-06:
Aber warum ist das so? Dann haben wir irgendwo ein Logging das weg sollte.

andig, 2023-05-06:
> Request failed with status code 400: API request failed http://localhost:7070/api/tariff/planner

Der wird vom UI geloggt

PR #7833 (2023-05-08): Planner: show planned slots without tariff
fix #7817

- return empty list instead of error when no tariff is defined
- adjust bar heights if no price is known

- [x] remove `?` and `unknown price`

[image]
closed 2023-05-08 by andig