evcc-io/evcc #7829: Easee Home nach update auf 0.116.7 Fehler

chrisherzberg, 2023-05-05:
### Describe the bug

Hallo zusammen,

nachdem ich die Version 0.116.7 installiert habe, bekomme ich meine Easee Home nicht mehr an evcc dran.
Username ist natürlich schon auf Email Adresse umgestellt, ansonsten würden ja auch keine Werte zurückkommen.

`invalid message: {\"serialNumber\":\"EH803080\",\"id\":48,\"timestamp\":\"2023-05-05T11:44:26.035385Z\",\"deliveredAt\":null,\"wasAccepted\":false,\"resultCode\":0,\"comment\":\"command expired at 1683287186, my time=1683295345\",\"ticks\":638188838660353850} json: cannot unmarshal number into Go struct field SignalRCommandResponse.ResultCode of type string`

Neuer Fehler, oder sehe ich den Wald vor lauter Bäumen nicht?

Besten Dank
Chris

### Steps to reproduce

1. Start evcc
2. Look at the webpage

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 617f5a16d07d79cbbb0e129c0118b45

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzYy5pbyIsInN1YiI6ImNocmlzaGVyemJlcmcifQ.gEkawM>

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.182.251
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.182.251
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.182.251
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: custom
  power:
[... 408 lines cut ...]
[lp-1  ] DEBUG 2023/05/05 16:36:58 soc estimated: 70.00% (vehicle: 70.00%)
[lp-1  ] DEBUG 2023/05/05 16:36:58 vehicle soc: 70%
[lp-1  ] DEBUG 2023/05/05 16:36:58 vehicle soc limit: 99%
[lp-1  ] DEBUG 2023/05/05 16:36:58 vehicle range: 367km
[lp-1  ] DEBUG 2023/05/05 16:36:58 pv charge current: 5.21A = 6A + -0.786A (542W @ 3p)
[site  ] DEBUG 2023/05/05 16:36:58 telemetry: charge: Δ62/62Wh @ 5696W
[easee ] ERROR 2023/05/05 16:36:59 invalid message: {"serialNumber":"EH803080","id":48,"timestamp":"2023-05-05T11:27:08.280309Z","deliveredAt":null,"wasAccepted":false,"resultCode":0,"comment":"command expired at 1683286148, my time=1683296378","ticks":638188828282803090} json: cannot unmarshal number into Go struct field SignalRCommandResponse.ResultCode of type string
[site  ] DEBUG 2023/05/05 16:37:00 ----
[lp-1  ] DEBUG 2023/05/05 16:37:00 charge power: 5696W
[site  ] DEBUG 2023/05/05 16:37:00 pv power: 7233W
[site  ] DEBUG 2023/05/05 16:37:01 battery soc: 99%
[site  ] DEBUG 2023/05/05 16:37:01 battery power: 412W
[site  ] DEBUG 2023/05/05 16:37:03 grid power: -0W
[easee ] ERROR 2023/05/05 16:37:04 invalid message: {"serialNumber":"EH803080","id":48,"timestamp":"2023-05-05T11:39:51.050601Z","deliveredAt":null,"wasAccepted":false,"resultCode":0,"comment":"command expired at 1683286911, my time=1683297423","ticks":638188835910506010} json: cannot unmarshal number into Go struct field SignalRCommandResponse.ResultCode of type string
[easee ] ERROR 2023/05/05 16:37:04 invalid message: {"serialNumber":"EH803080","id":48,"timestamp":"2023-05-05T11:39:35.061127Z","deliveredAt":null,"wasAccepted":false,"resultCode":0,"comment":"command expired at 1683286895, my time=1683297423","ticks":638188835750611270} json: cannot unmarshal number into Go struct field SignalRCommandResponse.ResultCode of type string
[easee ] ERROR 2023/05/05 16:37:04 invalid message: {"serialNumber":"EH803080","id":48,"timestamp":"2023-05-05T11:40:25.308823Z","deliveredAt":null,"wasAccepted":false,"resultCode":0,"comment":"command expired at 1683286945, my time=1683297423","ticks":638188836253088230} json: cannot unmarshal number into Go struct field SignalRCommandResponse.ResultCode of type string
```

### What type of operating system are you running?

Linux

### Version

0.116.7

Balujuhu, 2023-05-05:
Würde die tokens mal raus machen.

chrisherzberg, 2023-05-05:
Sind doch nicht die echten Tokens. ;-)

andig, 2023-05-05:
> json: cannot unmarshal number into Go struct field SignalRCommandResponse.ResultCode of type string

andig, 2023-05-05:
Fixed in master.

/cc @naltatis

closed 2023-05-05 by andig