evcc-io/evcc #7839: PV Produktion wird nicht korrekt angezeigt (Plenticore 10 Plus)

thorstenwagner, 2023-05-06:
### Describe the bug

Obwohl meine PV Anlage 9.6 KW produziert, werden die in evcc nicht angezeigt. Vorhandener Überschuss wird aber korrekt dargestellt. Der produzierte Strom der Mini-PV Anlage wird ebenfalls korrekt visualisiert.

[image]
[image]

Ich nutze einen Plenticore 10 Plus Wechselrichter und das Kostal Smart Meter. Die Anlage wurde am Donnerstag erst in Betrieb genommen und laut Web-UI ist kein Update verfügbar.

Vermutlich habe ich irgendwas falsch konfiguriert?

### Steps to reproduce

-

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: *

meters:
- name: grid1
  type: template
  template: kostal-ksem-inverter
  usage: grid
  modbus: tcpip
  id: 71
  host: 192.168.0.117
  port: 1502
- type: template
  template: tasmota
  usage: pv
  host: 192.168.0.86
  name: mini-pv
- type: template
  template: kostal-plenticore
  usage: pv
  modbus: tcpip
  host: 192.168.0.117
  id: 71
  port: 1502
  name: maxi-pv

chargers:
- type: template
  template: easee
  timeout: 20s 
  user: *
  password: * 
  charger: *
  name: wallbox2

loadpoints:
- title: Garage
  charger: wallbox2
  mode: now
  vehicle: skoda
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

vehicles:
- name: skoda
  type: template
  timeout: 20s
  template: enyaq
  capacity: 60
  user: *
  password: *

tariffs:
  planner:
    type: grünstromindex
    zip: *

site:
  title: MyHome
  meters:
    grid: grid1
    pvs:
      - mini-pv
```

### Log details

```text
[main  ] INFO 2023/05/06 12:17:24 evcc 0.116.7
[main  ] INFO 2023/05/06 12:17:24 using config file: /etc/evcc.yaml
[main  ] INFO 2023/05/06 12:17:24 starting ui and api at :7070
[db    ] INFO 2023/05/06 12:17:24 using sqlite database: /home/twagner/.evcc/evcc.db
[lp-1  ] WARN 2023/05/06 12:17:34 locking phase config to 3p for switchable charger
[site  ] INFO 2023/05/06 12:17:34 site config:
[site  ] INFO 2023/05/06 12:17:34   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/05/06 12:17:34     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/06 12:17:34     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/05/06 12:17:34   vehicles:
[site  ] INFO 2023/05/06 12:17:34     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/05/06 12:17:34 loadpoint 1:
[lp-1  ] INFO 2023/05/06 12:17:34   mode:        now
[lp-1  ] INFO 2023/05/06 12:17:34   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/05/06 12:17:34   meters:      charge ✓
[lp-1  ] INFO 2023/05/06 12:17:34     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/05/06 12:17:34 phase timer inactive
[lp-1  ] DEBUG 2023/05/06 12:17:34 pv timer inactive
[lp-1  ] DEBUG 2023/05/06 12:17:34 guard timer inactive
[lp-1  ] INFO 2023/05/06 12:17:34 vehicle updated: unknown -> Skoda
[site  ] DEBUG 2023/05/06 12:17:34 ----
[lp-1  ] DEBUG 2023/05/06 12:17:34 charge power: 0W
[site  ] ERROR 2023/05/06 12:17:34 tariff: no matching rate
[site  ] DEBUG 2023/05/06 12:17:34 pv power: 603W
[site  ] DEBUG 2023/05/06 12:17:34 grid power: -9462W
[site  ] DEBUG 2023/05/06 12:17:34 site power: -9462W
[lp-1  ] DEBUG 2023/05/06 12:17:37 vehicle odometer: 7545km
[lp-1  ] DEBUG 2023/05/06 12:17:37 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/06 12:17:37 charge meter total import: 1568.897kWh
[lp-1  ] DEBUG 2023/05/06 12:17:37 charger status: A
[lp-1  ] INFO 2023/05/06 12:17:37 car disconnected
[site  ] DEBUG 2023/05/06 12:17:44 ----
[lp-1  ] DEBUG 2023/05/06 12:17:44 charge power: 0W
[site  ] ERROR 2023/05/06 12:17:44 tariff: no matching rate
[site  ] DEBUG 2023/05/06 12:17:44 pv power: 608W
[site  ] DEBUG 2023/05/06 12:17:44 grid power: -9466W
[site  ] DEBUG 2023/05/06 12:17:44 site power: -9466W
[lp-1  ] DEBUG 2023/05/06 12:17:44 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/06 12:17:44 charge meter total import: 1568.897kWh
[lp-1  ] DEBUG 2023/05/06 12:17:44 charger status: A
[site  ] DEBUG 2023/05/06 12:17:54 ----
[lp-1  ] DEBUG 2023/05/06 12:17:54 charge power: 0W
[site  ] ERROR 2023/05/06 12:17:54 tariff: no matching rate
[site  ] DEBUG 2023/05/06 12:17:54 pv power: 612W
[site  ] DEBUG 2023/05/06 12:17:54 grid power: -9468W
[site  ] DEBUG 2023/05/06 12:17:54 site power: -9468W
[lp-1  ] DEBUG 2023/05/06 12:17:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/06 12:17:54 charge meter total import: 1568.897kWh
[lp-1  ] DEBUG 2023/05/06 12:17:54 charger status: A
[site  ] DEBUG 2023/05/06 12:18:04 ----
[lp-1  ] DEBUG 2023/05/06 12:18:04 charge power: 0W
[site  ] ERROR 2023/05/06 12:18:04 tariff: no matching rate
[site  ] DEBUG 2023/05/06 12:18:04 pv power: 611W
[site  ] DEBUG 2023/05/06 12:18:04 grid power: -9470W
[site  ] DEBUG 2023/05/06 12:18:04 site power: -9470W
[lp-1  ] DEBUG 2023/05/06 12:18:04 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/06 12:18:04 charge meter total import: 1568.897kWh
[lp-1  ] DEBUG 2023/05/06 12:18:04 charger status: A
```

### What type of operating system are you running?

Linux

### Version

0.166.7

StefanSchoof, 2023-05-06:
Du hast in deiner Site nur die mini-pv und nicht die maxi-pv

thorstenwagner, 2023-05-06:
Stupid me. Danke sehr!

closed 2023-05-06 by thorstenwagner

StefanSchoof, 2023-05-06:
@andig Wäre eine Warnung, wenn ein Meter konfiguriert, aber nicht benutzt ist einfach möglich?

andig, 2023-05-07:
Möglich, aber keine Prio...