evcc-io/evcc #7847: With HA update - strconv.ParseFloat: parsing "<nil>": invalid syntax

erdbeertoni, 2023-05-07:
### Describe the bug

Hi all,
I'm wondering if anybody has the same issue, I did a HA update yesterday from 2023.5.1 to 2023.5.2 and it seems since then I get this error in evcc:
`[site  ] ERROR 2023/05/07 09:05:37 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax`

Not sure what happened but it worked perfectly for me before?

Thank you so much.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
  - name: my_pv
    type: template
    template: sma-inverter
    usage: pv
    host: 192.168.10.56 
    password: *** 
  - name: my_grid
    type: custom
    power:
        source: http
        uri: http://192.168.10.41/cm?cmnd=Status%208
        jq: .StatusSNS.curr_w
      
chargers:
  - name: my_charger
    type: template
    template: go-e-v3
    host: 192.168.10.95 # IP-Adresse oder Hostname
```

### Log details

```text
[site  ] ERROR 2023/05/07 09:05:37 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.116.7

andig, 2023-05-07:
Bitte trace log

closed 2023-05-07 by andig

erdbeertoni, 2023-05-07:
sorry, in dieser Art:
```
[lp-1  ] DEBUG 2023/05/07 10:26:20 charge power: 1355W
[site  ] DEBUG 2023/05/07 10:26:20 pv power: 3287W
[http  ] TRACE 2023/05/07 10:26:20 GET http://192.168.10.41/cm?cmnd=Status%208
[http  ] TRACE 2023/05/07 10:26:20 {"StatusSNS":{"Time":"2023-05-07T09:26:21","SML":{"curr_w":-1163}}}
[http  ] TRACE 2023/05/07 10:26:20 GET http://192.168.10.41/cm?cmnd=Status%208
[http  ] TRACE 2023/05/07 10:26:20 {"StatusSNS":{"Time":"2023-05-07T09:26:21","SML":{"curr_w":-1163}}}
[http  ] TRACE 2023/05/07 10:26:21 GET http://192.168.10.41/cm?cmnd=Status%208
[http  ] TRACE 2023/05/07 10:26:21 {"StatusSNS":{"Time":"2023-05-07T09:26:21","SML":{"curr_w":-1163}}}
[site  ] ERROR 2023/05/07 10:26:21 grid meter: strconv.ParseFloat: parsing "<nil>": invalid syntax
[sma   ] TRACE 2023/05/07 10:26:23 login for 192.168.10.56:9522
```

andig, 2023-05-07:
Dein JSON passt einfach nicht zu Deinem JQ Query. Mit evcc hat das- custom meter- aber nichts zu tun. In Dein JQ muss ein .SML mit rein.