evcc-io/evcc #7856: Enhancement documentation

abc2006, 2023-05-08:
### Describe the bug

Error Message "invalid Zipcode" on the website

### Steps to reproduce

1. folge der Doku : 
2. planner:
  type: grünstromindex
  zip: meine PLZ # PLZ mit führender Null mit in "" setzen

### Configuration details

```yaml
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.3395
  feedin:
    type: fixed
    price: 0.0911
  planner:
    type: grünstromindex
    zip: "065931"
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.116.7

abc2006, 2023-05-08:
Lösung: Doku ändern in 
"Wenn die PLZ mit einer 0 beginnt, dann muss sie in "" angegeben werden". 

Einfacher und generischer: PLZ *immer* mit "" angeben

StefanSchoof, 2023-05-08:
Das gehört dann nach https://github.com/evcc-io/docs/issues
Du kannst recht einfach deinen Vorschlag als PR in dem docs einreichen.
[image]

andig, 2023-05-08:
Das wäre perfekt- vielen Dank!

closed 2023-05-08 by andig