evcc-io/evcc #7877: Keep EVCC running when Wallbox is offline

souhl, 2023-05-09:
**Is your feature request related to a problem? Please describe.**
Today the easee cloud is down, but it could be any other charger having issues. Because of this evcc is not coming up again after a restart. This is causing e.g. the modbus proxy not to work and other apps relying on it (e.g. Home Assistant SolarEdge Module) are not functional any longer.

**Describe the solution you'd like**
Mark e.g. Easee as "offline" and send a push message, but run the rest of the evcc services.

**Describe alternatives you've considered**
Not sure how to get the data out of SolarEdge without using the modbus proxy.

**Additional context**
With additional services offered by evcc like the modbus proxy or other chargers for e.g. bikes configured it would be nice to have evcc running even if one of the chargers like e.g. Shelly, Easee, Go-E not beeing available during startup.

andig, 2023-05-09:
> Because of this evcc is not coming up again after a restart

Kann ich nicht nachvollziehen. 

Der Modbus Proxy wird vor den sonstigen Geräte gestartet. Erläuft also, solange evcc läuft. Da evcc erst nach 5min neu startet- und auch dann vom Service restartet wird- ist die aktuelle Lösung bereits eine Alternative.

closed 2023-05-09 by andig

HeikoH78, 2023-05-09:
> > Because of this evcc is not coming up again after a restart
> 
> Kann ich nicht nachvollziehen.
> 
> Der Modbus Proxy wird vor den sonstigen Geräte gestartet. Erläuft also, solange evcc läuft. evcc läuft bis Neustart mit einer Fehlerseite. Da evcc erst nach 5min neu startet- und auch dann vom Service restartet wird- ist die aktuelle Lösung bereits eine Alternative.

Ich glaube das ist vielen nicht klar. Jedoch wäre es tatsächlich gut, wenn die normale Übersichtsseite im Browser angezeigt würde mit einem deutlichen Fehlerhinweis. Denn solange der Fehler wie heute bei der easee besteht, kann man die Anzeige nicht nutzen (und die wird bei weitem nicht nur fürs Laden genutzt ;))