evcc-io/evcc #7892: Tarriffs with currency CHF not working

chadibe0, 2023-05-10:
### Describe the bug

If I change currency: EUR to currency: CHF in the tariffs section the UI display is not complete anymore. The bars that are showing the powers of PV and grid are not displayed anymore.

### Steps to reproduce

1. change currency to CHF
2.
3.
...

### Configuration details

```yaml
tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.323
    zones:
      - days: Mo-Fr
        hours: 20-6
        price: 0.2252
      - days: Sa
        hours: 0-6
        price: 0.2252
      - days: Sa
        hours: 13-23
        price: 0.2252
      - days: So
        price: 0.2252
  feedin: 
    type: fixed
    price: 0.171
    zones:
      - days: Mo-Fr
        hours: 20-6
        price: 0.1315
      - days: Sa
        hours: 0-6
        price: 0.1315
      - days: Sa
        hours: 13-23
        price: 0.1315
      - days: So
        price: 0.1315
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.117.0

andig, 2023-05-10:
> The bars that are showing the powers of PV and grid are not displayed anymore.

Could you share a small screenshot which power bars you're referring to?

romerod, 2023-05-10:
I guess he refers to the bars which show current consumption and production and so on:
[image]

from the console:
minimumFractionDigits value is out of range.
    at new NumberFormat (<anonymous>)
    at Proxy.fmtPricePerKWh (index-f7ddf593.js:75:20033)
    at Proxy.detailsTooltip (index-f7ddf593.js:116:32309)
    at Proxy.$E (index-f7ddf593.js:116:35953)
    at qo (index-f7ddf593.js:1:21833)
    at Gi.D [as fn] (index-f7ddf593.js:1:41914)
    at Gi.run (index-f7ddf593.js:1:10552)
    at G.v.update (index-f7ddf593.js:1:42661)
    at I (index-f7ddf593.js:1:41656)
    at X (index-f7ddf593.js:1:41318)
    
fmtPricePerKWh: function(e=0, t="EUR", n=!1) {
            let r = t
              , o = e
              , s = "auto"
              , i = 3;
            return ["EUR", "USD"].includes(t) && (o *= 100,
            r = "ct",
            s = 1,
            i = 1),
            `${new Intl.NumberFormat(this.$i18n.locale,{
                style: "decimal",
                minimumFractionDigits: s,
                maximumFractionDigits: i
            }).format(o)} ${r}${n ? "" : "/kWh"}`
        }
t is CHF

chadibe0, 2023-05-10:
UI with Currency = EUR
[image]

UD with Currency = CHF
[image]

snh, 2023-05-11:
I'm seeing the same thing as of 0.117.0 when using `AUD` as the currency.

andig, 2023-05-11:
All currencies except EUR/USD are affected, sorry for that. We‘re working on a fix. AUD=Australian Dollar? Exciting to see how far evcc has made it!

closed 2023-05-11 by naltatis

Xardas365, 2025-04-29:
@andig Hi, is there a list of supported currencies? My currency (CZK) might not be supported. Thank you for your response.

PR #7918 (2023-05-11): Replace invalid argument minimumFractionDigits argument "auto"
Fix #7892 as discussed in #7898
commit 89032944 (2023-05-11): Replace invalid argument minimumFractionDigits  argument "auto" (fix #7892) (#7918)