evcc-io/evcc #7922: Einstellwerte Buffer/Priority/BatteryStart Soc

thebrainkafka, 2023-05-11:
Wäre es möglich folgendes wieder, bzw. zu ermöglichen 
Priority Soc > bufferSoc einzustellen
BatteryStart = BufferSoc einzustellen

Mit dem coolen Feature dies direkt in der UI einzugeben ist diese Möglichkeit leider verloren gegangen.

Ziel. Ladestart/ stop pulsen bei wenig Solarertrag minimieren

andig, 2023-05-11:
> BatteryStart = BufferSoc einzustellen

Das bedeutet Hysterese = 0 und ist unsinnig.

> Priority Soc > bufferSoc einzustellen

Entweder soll erst die Hausbatterie voll sein oder das Auto geladen werden. Beides gleichzeitig geht nicht.

frankomio, 2023-05-11:
ich bin verwirrt was die Doku angeht:
was ist denn einzustellen , wenn zuerst das Auto mit Überschuss geladen werden soll anstatt Hausbatterie ?

thebrainkafka, 2023-05-11:
> > BatteryStart = BufferSoc einzustellen
> 
> Das bedeutet Hysterese = 0 und ist unsinnig.
> 
> > Priority Soc > bufferSoc einzustellen
> 
> Entweder soll erst die Hausbatterie voll sein oder das Auto geladen werden. Beides gleichzeitig geht nicht.

Danke, dann hatte ich da nen Denkfehler.
Ich würde gern im PV erst mit der Ladung starten wenn auch Batterie zum Puffern genutzt werden kann.
Was müsste ich dazu einstellen?

Aktuell führt es, nachdem PV Start, wieder aufgenommen wurde zum takten, da bereits ohne Batterieunterstütztung die Ladung startet.

thebrainkafka, 2023-05-11:
[image]

So wie hier abgebildet hätte ich es gerne 😜.
Priority Soc 50
Buffer Soc 60
BatteryStart knapp über 50

Das kann man so aber nur durch Einstellen der Schieber in einer spezifischen Reihenfolge erreichen.
Evtl auch ein Bug .

frankomio, 2023-05-11:
wo finde ich diese Einstellungen bitte ?