evcc-io/evcc #7928: ZCS Azzurro HYD 10KTL-3PH-PRO  Register Update

maxirass, 2023-05-11:
Hallo liebes Evcc Team,

Wir haben seit kurzem den **oben genannten Wechselrichter mit Pylontech Batteriespeicher**.
Seit gestern besitzen wir eine **go-eCharger Gemin v3**, damit ich evcc nutzen kann.
Ich lese die Daten über einen **Waveshare RS485 to eth** aus und sende die Daten an einen ioBroker via Modbus TCP/IP.
Ich habe die Register selbst herausgefunden, damit mein Grafana Dashboard die richtigen Werte anzeigt.
Leider habe ich bei der Konfiguration von evcc festgestellt, dass im Template für den ZCS Azzurro Hybrid Wechselrichter, 
die falschen Register hinterlegt sind und somit falsche Werte raus kommen. Da der Azzurro ein umgelabelter SofarSolar ist,
habe ich das Template sofarsolar-g3 verwendet. Dabei stimmen viele Werte, aber nicht alle.
Deshalb wollte ich auf diesem Wege die richtigen Register mitteilen.
Dabei gilt zu beachten, dass der wahre Verbrauch sich aus "active_load - total_grid_output" zusammensetzt.
Die Registeradressen sind als Dezimalzahl im Excel hinterlegt und nicht als Hexa-Adressen.

VG

[Azzurro_Register.xlsx](https://github.com/evcc-io/evcc/files/11451581/Azzurro_Register.xlsx)

andig, 2023-05-13:
@maxirass könntest Du einen PR mit korrigiertem Template machen?

maxirass, 2023-05-13:
@andig sorry ich bin noch komplett neu in dem Github Thema. Was meinst du mit PR?

andig, 2023-05-13:
Ein "pull request" für die neue Templatedatei: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/about-pull-requests

maxirass, 2023-05-14:
Ich hab leider keinerlei Programmiererfahrungen. Könnte jemand anders das Template korrigieren und einen PR machen.

andig, 2023-05-14:
Kannst du denn aufschreiben konkret welche Register abweichen? Die Mühe muss sich ja auch jemand machen?

maxirass, 2023-05-14:
Ok, also ich habe gerade mal die Register vom sofarsolar-g3 Template überprüft. Die Register stimmen alle, aber in der Web Oberfläche von evcc wird beim Vebrauch der "Hausverbrauch + Batterieladung" angezeigt, aber bei der Batterieladung wird kein Wert angezeigt. Deswegen vermute ich, dass hier nur die Werte falsch verknüpft sind. Ich hoffe ich konnte es verständlich erklären. Ich hänge mal 2 Screenshots von meinem Grafana Dashboard und von der evcc Oberfläche an.
[image]
[image]

andig, 2023-05-14:
/cc @premultiply zusätzlich abgedeckter WR und Fehler im Template?

premultiply, 2023-05-14:
`evcc meter --log trace`

closed 2023-05-14 by premultiply

premultiply, 2023-05-14:
Wahr wohl nur eine falsche Skalierung der Batterieleistung. Bitte mal testen.

maxirass, 2023-05-14:
Jetzt wird der Wert angezeigt, aber um eine 10er Potenz zu viel also eigentlich sind es 4kW und evcc zeigt 40kW an. Und die Ladung und Entladung sind in der Ansicht vertauscht. 
[image]
[image]

premultiply, 2023-05-14:
Die Energiemengen sind doch auch bestimmt falsch skaliert, oder?

maxirass, 2023-05-14:
Die Skalierungen in der Excel weiter oben stimmen in meinem Grafana Dashboard.
Ich hab gerade nochmal abgeglichen:
0x0686 # PV_Generation_Total hat scale 0.1 für kWh
0x069A # Bat_Discharge_Total hat scale 0.1 für kWh
0x0606 # Power_Bat1 hat scale 10 für W
Heißt es stimmen die Skalierungen bis auf Power_Bat1 und in der Web Oberfläche müssen "die Anzeigen getauscht" werden, sodass negative Werte bei Entladen und positive Werte bei Laden angezeigt werden

maxirass, 2023-05-14:
Und 
0x068E # Energy_Purchase_Total hat scale 0.1 für kWh
die Skalierung stimmt also auch

andig, 2023-05-18:
Und ...ist das ein Widerspruch zur Implementierung?

commit 7b6f7f98 (2023-05-14): sofarsolar-g3: fix battery power scale

fix #7928