evcc-io/evcc #7942: Setting up test setup for lastmanagement

hyperbart, 2023-05-11:
@hyperbart pls checkout [here](https://github.com/hoermto/evcc/releases/tag/0.117.0-loadmanagement-power) for a build and some comments about power limiting of this release. Added a docker image for local import.

_Originally posted by @hoermto in https://github.com/evcc-io/evcc/issues/3453#issuecomment-1544265777_

hyperbart, 2023-05-11:
hi @hoermto 
There is a car charging here tonight and I am preparing everything.

Could you take a look at the yml and check if it is valid?
I will try to deploy the last management this weekend.

If I want to revert to the original docker it is as simple as to set the image back to "evcc/evcc:nightly" ? The fact that I downloaded your image and I set it to evcc/evcc:testing does not interfere?

```
interval: 5s

site:
- title: xxxxxx
  meters:
    grid: HuisMeter
    pvs:
    - pvHuis
    - pvTuinHuis
  residualpower: 10

messaging:
  events:
    connect:
      title: Wagen aangesloten 
      msg: |
        Auto is aangesloten op de {{.title}}-lader
        Huidig rijbereik is XXXX km
    disconnect:
      title: Wagen ontkoppeld
      msg: |
        Laadbeurt voltooid
        Er is {{round (divf .chargedEnergy 1000) 2 }} kWh geleverd op {{.chargeDuration}}.
        Auto [INVULLEN AUTO] was {{round (divf .connectedDuration) 8 }} aangesloten op de laadpaal
        Je kan [ZOVEEL] km rijden!

  services:
    - type: telegram
      token: xxxxxx:xxxxxx
      chats:
        - -xxxxxx

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.52 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.12 # [currency]/kWh
  planner:
    type: electricitymaps
    uri: https://api.co2signal.com/
    token: xxxxxx
    zone: BE

meters:
  - name: HuisMeter
    type: modbus
    model: sdm # specific non-sunspec meter
    id: 1
    energy: Sum # only required for charge meter usage
    #via TCP:
    uri: 192.168.X.YYY:8899
    rtu: false # serial modbus rtu (rs485) device connected using simple ethernet adapter
  - name: pvHuis
    type: sma 
    uri: 192.168.X.YYY
    password: xxxxxx
  - name: pvTuinHuis
    type: sma
    uri: 192.168.X.YYY
    password: xxxxxx

sponsortoken: xxxxxx.xxxxxx.xxxxxx-xxxxxx 

log: debug 
levels:
    core: debug
    cc-main: debug

influx:
  url: http://192.168.X.YYY:8086
  database: EVCC
  token: "xxxxxx=="
  org: "xxxxxx"

chargers:
- name: ChargingPointAlfen
  type: alfen
  uri: 192.168.4.178 # TCP ModBus address

loadpoints:
- title: Alfen # display name for UI
  charger: ChargingPointAlfen # Wallbox Referenz
  mode: minpv # charge mode (off, now, minpv, pv)
  phases: 0 
  mincurrent: 6 
  maxcurrent: 16
  resetOnDisconnect: false
  guardduration: 2m
  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 0 # maximum import power (W)
    delay: 3m
  circuit: Fluvius

circuits:
  - name: Fluvius
    maxCurrent: 20
    meter: HuisMeter
    parent:

vehicles:

```

hoermto, 2023-05-12:
> Could you take a look at the yml and check if it is valid? I will try to deploy the last management this weekend.
> 

If you evcc accepts this yaml, should be good to go. I cannot see any issue, however I cannot test this file due to missing infrastructure.
Regarding the limit: You might consider changing or adding to the circuit a power limit (`maxPower` in kW), since I think this is what you where looking for:
```
circuits:
  - name: Fluvius
    maxCurrent: 20 # or set to 0 if you dont want to check for phase currents
    maxPower: 5  # in kW
    meter: HuisMeter
    parent:
```

> If I want to revert to the original docker it is as simple as to set the image back to "evcc/evcc:nightly" ? The fact that I downloaded your image and I set it to evcc/evcc:testing does not interfere?

The benefit of docker is to switch the versions easily. If you have loaded the image, change in your Docker GUI or tooling the image name from ":nightly" to ":testing". If something goes bad, just revert this. You should not run 2 evcc in parallel, unless you know what you are doing.

hyperbart, 2023-05-12:
@hoermto :

[image]

hyperbart, 2023-05-12:
Yeah, that's what I thought also, because it says an old version.
Are you sure that the docker image is correct which you put on GitHub?

hoermto, 2023-05-12:
Just ran the following. I think the file name got replaced by github while upload.
[image]

hyperbart, 2023-05-12:
Okay, but I am not seeing what I should do differently. (I am not able to deduct the action points for me from your image).
Could you clarify further?

hoermto, 2023-05-12:
I have checked the checksum, the file is identical with the uploaded file. I dont know how you start the evcc container, so make sure that you start up the correct version. You have seen the command I am using to start the fresh imported image. You might follow the same command to check. Also check out the evcc documentation to run evcc in docker.
[image]

hoermto, 2023-05-12:
One more thing: my image is for amd64 (intel/amd CPU), not ARM (Raspi & co).
If Docker is a kind of challenge 4u, you can try out using the linux tarballs from releases, extract and run the binary natively, as I have shown previously

hyperbart, 2023-05-12:
Okay, something is/was wrong with tags, don't know why, I renamed your image and I got it up and running, however, it fails on circuit/meter:

[image]

hoermto, 2023-05-12:
> Okay, something is/was wrong with tags, don't know why, I renamed your image and I got it up and running, however, it fails on circuit/meter:
> 

set the `maxCurrent: 0` in your config in the circuit, and only use a value for `maxPower`. Then it should work also with non-phase-enabled meters.
Tbh, havent tested this case so far ...

hyperbart, 2023-05-12:
> maybe your meter has, but not supported atm in evcc to read the phase currents

Clear, understood. So that means that for the Eastron there is not yet a template to read phase currents I presume @andig .

Got a little bit further:

`bart@nas:/EVO500$ docker logs EVCC_evcc.1.2s6f8m25qujogsr31ayhmo4h3 
[main  ] INFO 2023/05/12 16:36:22 evcc 0.117.0-loadmanagement-power (328573eb)
[main  ] INFO 2023/05/12 16:36:22 using config file: /etc/evcc.yaml
[main  ] INFO 2023/05/12 16:36:22 starting ui and api at :7070
[db    ] INFO 2023/05/12 16:36:23 using sqlite database: /root/.evcc/evcc.db
[alfen ] DEBUG 2023/05/12 16:36:23 detected 3p alfen
[circuit-1] DEBUG 2023/05/12 16:36:23 current checking disabled
[em    ] ERROR 2023/05/12 16:36:23 unexpected status: 404 (Not Found)
[main  ] ERROR 2023/05/12 16:36:23 failed configuring planner tariff: cannot create tariff 'electricitymaps': unexpected status: 404 (Not Found)
[site  ] WARN 2023/05/12 16:36:23 bufferStartSoc must be larger than bufferSoc
[site  ] INFO 2023/05/12 16:36:23 site config:
[site  ] INFO 2023/05/12 16:36:23   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/05/12 16:36:23     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/05/12 16:36:23     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/05/12 16:36:23     pv 2:      power ✓ energy ✓ currents ✓
[site  ] INFO 2023/05/12 16:36:23   circuits:
[site  ] INFO 2023/05/12 16:36:23      Power: max 5.0kW, Current: no check
[lp-1  ] INFO 2023/05/12 16:36:23 loadpoint 1:
[lp-1  ] INFO 2023/05/12 16:36:23   mode:        minpv
[lp-1  ] INFO 2023/05/12 16:36:23   circuit:     Fluvius
[lp-1  ] INFO 2023/05/12 16:36:23   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/05/12 16:36:23   meters:      charge ✓
[lp-1  ] INFO 2023/05/12 16:36:23     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/05/12 16:36:23 phase timer inactive
[lp-1  ] DEBUG 2023/05/12 16:36:23 pv timer inactive
[lp-1  ] DEBUG 2023/05/12 16:36:23 guard timer inactive
[site  ] DEBUG 2023/05/12 16:36:23 ----
[lp-1  ] DEBUG 2023/05/12 16:36:24 charge power: 0W
[site  ] DEBUG 2023/05/12 16:36:24 pv power: 5020W
[main  ] INFO 2023/05/12 16:36:24 new version available: 0.117.2
[site  ] DEBUG 2023/05/12 16:36:24 grid power: -4185W
[site  ] DEBUG 2023/05/12 16:36:24 site power: -4175W
[lp-1  ] DEBUG 2023/05/12 16:36:24 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/12 16:36:24 charge meter total import: 3445.163kWh
[lp-1  ] DEBUG 2023/05/12 16:36:24 charger status: A
[lp-1  ] INFO 2023/05/12 16:36:24 car disconnected
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x1bace76]

goroutine 102 [running]:
github.com/evcc-io/evcc/core.(*Circuit).MaxPhasesCurrent(0xc0009ce8c0)
        /build/core/circuit.go:183 +0x36
github.com/evcc-io/evcc/core.(*Circuit).update(0xc0009ce8c0)
        /build/core/circuit.go:133 +0x1e
github.com/evcc-io/evcc/core.(*Site).update(0xc000155c00, {0x309fae8?, 0xc000648c00})
        /build/core/site.go:757 +0x739
github.com/evcc-io/evcc/core.(*Site).Run(0xc000155c00, 0xc000d52000, 0x12a05f200)
        /build/core/site.go:851 +0x136
github.com/evcc-io/evcc/cmd.runRoot.func4()
        /build/cmd/root.go:279 +0x28
created by github.com/evcc-io/evcc/cmd.runRoot
        /build/cmd/root.go:278 +0x192a`

hoermto, 2023-05-12:
hey cool. I am just implementing more test cases, but didnt see this so far. Will come back to you.
Can you post your meter, circuits and LP sections you are finally using?

hyperbart, 2023-05-12:
Sure, no problem:

Meter:
[image]

Circuits:
[image]

[image]

hoermto, 2023-05-12:
@hyperbart I uploaded the `evcc_evcc_testing_fix.tgz` image to the release folder. If you want to attempt a new run?

hyperbart, 2023-05-22:
@hoermto ! 😉 

Got it up and running with your new release, car is charging.
Can you tell me why the car is not charging with more kW? I see it says 1.32kW available?

```
[circuit-1] DEBUG 2023/05/22 19:24:15 circuit using 3.45kW, 1.55kW available
[lp-1  ] DEBUG 2023/05/22 19:24:15 get power limitation from 4140.0W to 3894.4W (5.6A) from circuit
[lp-1  ] DEBUG 2023/05/22 19:24:15 max charge current: 5.64A
[lp-1  ] DEBUG 2023/05/22 19:24:15 guard enable in 45s
[site  ] DEBUG 2023/05/22 19:24:15 ----
[lp-1  ] DEBUG 2023/05/22 19:24:15 charge power: 4075W
[site  ] DEBUG 2023/05/22 19:24:15 pv power: 1457W
[site  ] DEBUG 2023/05/22 19:24:15 grid power: 73W
[site  ] DEBUG 2023/05/22 19:24:15 site power: 83W
[lp-1  ] DEBUG 2023/05/22 19:24:15 charge currents: [5.89 6.07 5.9]A
[lp-1  ] DEBUG 2023/05/22 19:24:15 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:24:16 charge meter total import: 3553.251kWh
[lp-1  ] DEBUG 2023/05/22 19:24:16 charger status: B
[lp-1  ] DEBUG 2023/05/22 19:24:16 final charge energy: 0.116kWh
[lp-1  ] INFO 2023/05/22 19:24:16 stop charging <-
[lp-1  ] DEBUG 2023/05/22 19:24:16 wake-up timer: start
[lp-1  ] DEBUG 2023/05/22 19:24:16 charge meter total import: 3553.252kWh
[lp-1  ] DEBUG 2023/05/22 19:24:16 pv charge current: 0A = 0A + -0.121A (83W @ 3p)
[circuit-1] DEBUG 2023/05/22 19:24:16 circuit using -0.67kW, 5.67kW available
[lp-1  ] DEBUG 2023/05/22 19:24:16 max charge current: 6A
[site  ] DEBUG 2023/05/22 19:24:19 ----
[lp-1  ] DEBUG 2023/05/22 19:24:19 charge power: 0W
[site  ] DEBUG 2023/05/22 19:24:19 pv power: 1457W
[site  ] DEBUG 2023/05/22 19:24:34 grid power: 83W
[site  ] DEBUG 2023/05/22 19:24:34 site power: 93W
[lp-1  ] DEBUG 2023/05/22 19:24:34 charge currents: [0.558 0.431 0.45]A
[lp-1  ] DEBUG 2023/05/22 19:24:34 charge meter total import: 3553.252kWh
[lp-1  ] DEBUG 2023/05/22 19:24:35 charger status: C
[lp-1  ] DEBUG 2023/05/22 19:24:35 charge start energy: 3.55e+03kWh
[lp-1  ] INFO 2023/05/22 19:24:35 start charging ->
[lp-1  ] DEBUG 2023/05/22 19:24:35 wake-up timer: stop
[lp-1  ] DEBUG 2023/05/22 19:24:35 pv charge current: 2.42A = 2.56A + -0.134A (93W @ 3p)
[circuit-1] DEBUG 2023/05/22 19:24:35 circuit using 0.38kW, 4.62kW available
[site  ] DEBUG 2023/05/22 19:24:35 ----
[lp-1  ] DEBUG 2023/05/22 19:24:35 charge power: 736W
[site  ] DEBUG 2023/05/22 19:24:35 pv power: 1444W
[site  ] DEBUG 2023/05/22 19:24:36 grid power: 594W
[site  ] DEBUG 2023/05/22 19:24:36 site power: 604W
[lp-1  ] DEBUG 2023/05/22 19:24:36 charge currents: [1.53 1.57 1.56]A
[lp-1  ] DEBUG 2023/05/22 19:24:36 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:24:36 charge meter total import: 3553.252kWh
[lp-1  ] DEBUG 2023/05/22 19:24:36 charger status: C
[lp-1  ] DEBUG 2023/05/22 19:24:36 pv charge current: 2.66A = 3.53A + -0.875A (604W @ 3p)
[circuit-1] DEBUG 2023/05/22 19:24:37 circuit using 0.95kW, 4.05kW available
[site  ] DEBUG 2023/05/22 19:24:39 ----
[lp-1  ] DEBUG 2023/05/22 19:24:39 charge power: 2168W
[site  ] DEBUG 2023/05/22 19:24:39 pv power: 1432W
[site  ] DEBUG 2023/05/22 19:24:39 grid power: 2070W
[site  ] DEBUG 2023/05/22 19:24:39 site power: 2080W
[lp-1  ] DEBUG 2023/05/22 19:24:39 charge currents: [3 3.15 2.97]A
[lp-1  ] DEBUG 2023/05/22 19:24:39 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:24:40 charge meter total import: 3553.254kWh
[lp-1  ] DEBUG 2023/05/22 19:24:40 charger status: C
[lp-1  ] DEBUG 2023/05/22 19:24:40 pv charge current: 1.98A = 5A + -3.01A (2080W @ 3p)
[circuit-1] DEBUG 2023/05/22 19:24:40 circuit using 2.37kW, 2.63kW available
[site  ] DEBUG 2023/05/22 19:24:44 ----
[lp-1  ] DEBUG 2023/05/22 19:24:44 charge power: 3912W
[site  ] DEBUG 2023/05/22 19:24:44 pv power: 1426W
[site  ] DEBUG 2023/05/22 19:24:44 grid power: 3531W
[site  ] DEBUG 2023/05/22 19:24:44 site power: 3541W
[lp-1  ] DEBUG 2023/05/22 19:24:44 charge currents: [5.82 6.08 5.85]A
[lp-1  ] DEBUG 2023/05/22 19:24:44 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:24:45 charge meter total import: 3553.259kWh
[lp-1  ] DEBUG 2023/05/22 19:24:45 charger status: C
[... 198 lines cut ...]
[lp-1  ] DEBUG 2023/05/22 19:26:57 pv charge current: 0.521A = 6A + -5.48A (3781W @ 3p)
[circuit-1] DEBUG 2023/05/22 19:26:57 circuit using 3.72kW, 1.28kW available
[site  ] DEBUG 2023/05/22 19:26:59 ----
[lp-1  ] DEBUG 2023/05/22 19:26:59 charge power: 4019W
[site  ] DEBUG 2023/05/22 19:26:59 pv power: 1293W
[site  ] DEBUG 2023/05/22 19:26:59 grid power: 3673W
[site  ] DEBUG 2023/05/22 19:26:59 site power: 3683W
[lp-1  ] DEBUG 2023/05/22 19:26:59 charge currents: [5.98 5.97 5.92]A
[lp-1  ] DEBUG 2023/05/22 19:26:59 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:26:59 charge meter total import: 3553.412kWh
[lp-1  ] DEBUG 2023/05/22 19:27:00 charger status: C
[lp-1  ] DEBUG 2023/05/22 19:27:00 pv charge current: 0.663A = 6A + -5.34A (3683W @ 3p)
[circuit-1] DEBUG 2023/05/22 19:27:00 circuit using 3.66kW, 1.34kW available
[site  ] DEBUG 2023/05/22 19:27:04 ----
[lp-1  ] DEBUG 2023/05/22 19:27:04 charge power: 4024W
[site  ] DEBUG 2023/05/22 19:27:04 pv power: 1314W
[site  ] DEBUG 2023/05/22 19:27:04 grid power: 3624W
[site  ] DEBUG 2023/05/22 19:27:04 site power: 3634W
[lp-1  ] DEBUG 2023/05/22 19:27:04 charge currents: [5.99 5.97 5.92]A
[lp-1  ] DEBUG 2023/05/22 19:27:04 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:27:04 charge meter total import: 3553.417kWh
[lp-1  ] DEBUG 2023/05/22 19:27:05 charger status: C
[lp-1  ] DEBUG 2023/05/22 19:27:05 pv charge current: 0.734A = 6A + -5.27A (3634W @ 3p)
[circuit-1] DEBUG 2023/05/22 19:27:05 circuit using 3.68kW, 1.32kW available
```

hyperbart, 2023-05-22:
Switching it from min+solar to fast resulted almost immediately in:

[image]

I think that's a bit too fast?

hyperbart, 2023-05-22:
@hoermto I am very sorry but I will have to stop the test, the Alfen has been clicking/clacking for the last 10 minutes, this is how it looks in Grafana (below) and after the Grafana screenshot you will also find the log.

[image]

```
[lp-1  ] DEBUG 2023/05/22 19:38:17 max charge current: 8.48A
[lp-1  ] DEBUG 2023/05/22 19:38:17 charger: guard elapse
[lp-1  ] DEBUG 2023/05/22 19:38:17 guard timer inactive
[lp-1  ] DEBUG 2023/05/22 19:38:17 charger enable
[lp-1  ] DEBUG 2023/05/22 19:38:17 wake-up timer: start
[site  ] DEBUG 2023/05/22 19:38:19 ----
[lp-1  ] DEBUG 2023/05/22 19:38:19 charge power: 0W
[site  ] DEBUG 2023/05/22 19:38:19 pv power: 852W
[site  ] DEBUG 2023/05/22 19:38:34 grid power: 372W
[site  ] DEBUG 2023/05/22 19:38:34 site power: 382W
[lp-1  ] DEBUG 2023/05/22 19:38:34 charge currents: [0.556 0.431 0.445]A
[lp-1  ] DEBUG 2023/05/22 19:38:34 charge meter total import: 3554.140kWh
[lp-1  ] DEBUG 2023/05/22 19:38:35 charger status: C
[lp-1  ] DEBUG 2023/05/22 19:38:35 charge start energy: 3.55e+03kWh
[lp-1  ] INFO 2023/05/22 19:38:35 start charging ->
[lp-1  ] DEBUG 2023/05/22 19:38:35 wake-up timer: stop
[circuit-1] DEBUG 2023/05/22 19:38:35 circuit using 0.72kW, 4.28kW available
[lp-1  ] DEBUG 2023/05/22 19:38:35 get power limitation from 11040.0W to 4283.2W (6.2A) from circuit
[lp-1  ] DEBUG 2023/05/22 19:38:35 max charge current: 6.21A
[site  ] DEBUG 2023/05/22 19:38:35 ----
[lp-1  ] DEBUG 2023/05/22 19:38:36 charge power: 771W
[site  ] DEBUG 2023/05/22 19:38:36 pv power: 857W
[site  ] DEBUG 2023/05/22 19:38:36 grid power: 918W
[site  ] DEBUG 2023/05/22 19:38:36 site power: 928W
[lp-1  ] DEBUG 2023/05/22 19:38:36 charge currents: [1.63 1.56 1.53]A
[lp-1  ] DEBUG 2023/05/22 19:38:36 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:38:36 charge meter total import: 3554.140kWh
[lp-1  ] DEBUG 2023/05/22 19:38:37 charger status: C
[circuit-1] DEBUG 2023/05/22 19:38:37 circuit using 1.36kW, 3.64kW available
[lp-1  ] DEBUG 2023/05/22 19:38:37 get power limitation from 11040.0W to 4413.1W (6.4A) from circuit
[lp-1  ] DEBUG 2023/05/22 19:38:37 max charge current: 6.4A
[site  ] DEBUG 2023/05/22 19:38:39 ----
[lp-1  ] DEBUG 2023/05/22 19:38:39 charge power: 2207W
[site  ] DEBUG 2023/05/22 19:38:39 pv power: 865W
[site  ] DEBUG 2023/05/22 19:38:54 grid power: 3853W
[site  ] DEBUG 2023/05/22 19:38:54 site power: 3863W
[lp-1  ] DEBUG 2023/05/22 19:38:54 charge currents: [6.2 6.12 6.09]A
[lp-1  ] DEBUG 2023/05/22 19:38:54 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:38:55 charge meter total import: 3554.158kWh
[lp-1  ] DEBUG 2023/05/22 19:38:55 charger status: C
[circuit-1] DEBUG 2023/05/22 19:38:55 circuit using 3.85kW, 1.15kW available
[lp-1  ] DEBUG 2023/05/22 19:38:55 get power limitation from 11040.0W to 3361.8W (4.9A) from circuit
[lp-1  ] DEBUG 2023/05/22 19:38:55 max charge current: 4.87A
[lp-1  ] DEBUG 2023/05/22 19:38:55 charger: guard elapse
[lp-1  ] DEBUG 2023/05/22 19:38:55 guard timer inactive
[lp-1  ] DEBUG 2023/05/22 19:38:55 charger disable
[lp-1  ] DEBUG 2023/05/22 19:38:55 wake-up timer: stop
[site  ] DEBUG 2023/05/22 19:38:55 ----
[lp-1  ] DEBUG 2023/05/22 19:38:56 charge power: 4141W
[site  ] DEBUG 2023/05/22 19:38:56 pv power: 913W
[site  ] DEBUG 2023/05/22 19:38:56 grid power: -305W
[site  ] DEBUG 2023/05/22 19:38:56 site power: -295W
[lp-1  ] DEBUG 2023/05/22 19:38:56 charge currents: [6.19 6.16 6.12]A
[lp-1  ] DEBUG 2023/05/22 19:38:56 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:38:56 charge meter total import: 3554.160kWh
[lp-1  ] DEBUG 2023/05/22 19:38:56 charger status: B
[lp-1  ] DEBUG 2023/05/22 19:38:56 final charge energy: 1.02kWh
[lp-1  ] INFO 2023/05/22 19:38:56 stop charging <-
[lp-1  ] DEBUG 2023/05/22 19:38:57 charge meter total import: 3554.160kWh
[circuit-1] DEBUG 2023/05/22 19:38:57 circuit using -0.31kW, 5.31kW available
[lp-1  ] DEBUG 2023/05/22 19:38:57 get power limitation from 11040.0W to 9453.4W (13.7A) from circuit
[lp-1  ] DEBUG 2023/05/22 19:38:57 max charge current: 13.7A
[lp-1  ] DEBUG 2023/05/22 19:38:57 charger: guard elapse
[lp-1  ] DEBUG 2023/05/22 19:38:57 guard timer inactive
[lp-1  ] DEBUG 2023/05/22 19:38:57 charger enable
[... 958 lines cut ...]
[lp-1  ] DEBUG 2023/05/22 19:48:35 charge meter total import: 3554.430kWh
[lp-1  ] DEBUG 2023/05/22 19:48:35 charger status: C
[lp-1  ] DEBUG 2023/05/22 19:48:35 charge start energy: 3.55e+03kWh
[lp-1  ] INFO 2023/05/22 19:48:35 start charging ->
[lp-1  ] DEBUG 2023/05/22 19:48:35 wake-up timer: stop
[circuit-1] DEBUG 2023/05/22 19:48:35 circuit using 0.65kW, 4.35kW available
[lp-1  ] DEBUG 2023/05/22 19:48:35 get power limitation from 11040.0W to 4350.5W (6.3A) from circuit
[lp-1  ] DEBUG 2023/05/22 19:48:35 max charge current: 6.31A
[site  ] DEBUG 2023/05/22 19:48:36 ----
[lp-1  ] DEBUG 2023/05/22 19:48:36 charge power: 839W
[site  ] DEBUG 2023/05/22 19:48:36 pv power: 1052W
[site  ] DEBUG 2023/05/22 19:48:36 grid power: 895W
[site  ] DEBUG 2023/05/22 19:48:36 site power: 905W
[lp-1  ] DEBUG 2023/05/22 19:48:36 charge currents: [1.68 1.58 1.71]A
[lp-1  ] DEBUG 2023/05/22 19:48:36 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/22 19:48:36 charge meter total import: 3554.430kWh
[lp-1  ] DEBUG 2023/05/22 19:48:37 charger status: C
[circuit-1] DEBUG 2023/05/22 19:48:37 circuit using 1.23kW, 3.77kW available
[lp-1  ] DEBUG 2023/05/22 19:48:37 get power limitation from 11040.0W to 4607.1W (6.7A) from circuit
[lp-1  ] DEBUG 2023/05/22 19:48:37 max charge current: 6.68A
[site  ] DEBUG 2023/05/22 19:48:39 ----
[lp-1  ] DEBUG 2023/05/22 19:48:39 charge power: 2304W
[site  ] DEBUG 2023/05/22 19:48:39 pv power: 1045W
[lp-1  ] DEBUG 2023/05/22 19:48:51 charge meter total import: 3554.444kWh
```

hoermto, 2023-05-22:
your config would help. I suspect either a timing issue (slow reaction from Wallbox/car on changing the current), or an issue receiving the actual meter data?
```
[circuit-1] DEBUG 2023/05/22 19:37:35 circuit using 5.04kW, -0.04kW available
[lp-1  ] DEBUG 2023/05/22 19:37:35 get power limitation from 11040.0W to 4162.0W (6.0A) from circuit
....
[circuit-1] DEBUG 2023/05/22 19:37:36 circuit using -0.16kW, 5.16kW available
```

It shows partially 0 kW load in circuit. If its a virtual meter, the LPs might have an issue providing the current consumption. Maybe yout LP / Wallbox interrupts shortly the charging when changing the current? 
If you want, you might lower the evcc timing to 5 or 10 secs

hoermto, 2023-05-22:
So what does it mean? Does the increase of the `interval` value help to improve the situation? Afaik the "guard" and other timers in the loadpoint only control the complete enable/disable interval, but not the change interval of the current.

hyperbart, 2023-05-22:
> 

Sorry for the confusion, what I meant with "I got you" is that I would post the config asap, just a quick response ;) .

Nothing has changed, so the config as you see above has been set like this during the testing period and is set like this also for the regular build of EVCC.

So it was already at 5s interval. Do you still want me to up it to 10s?

I don't know what in fact causes LB to turn off the charging, just experiencing this behavior without looking into it for now.
I understand what you are saying but I suspect the corrective actions during charging are too strict or not taking into account hysteresis, could that be?

hoermto, 2023-05-23:
Config looks ok. I suspect a timing issue between meter accuracy and LP/Car speed of applying changes.
Yes, increasing the interval would be an option. However, there might be another point:
There is a guard period per LP to prevent the LP to toggle too fast. When applying LM, and the remaining power is < minCurrent() (6A by default), the LM ignores this guard period and powers off immediately (to prevent overload). It a point to discuss. But: how do you deploy the evcc? Build yourself or use a release from download? I could get you a build with where the guard period is still in place, even with LM. Just to test that in *your* case the LP will not switch off too fast.

Regarding your questions:
- the remaining power and current application depends also on timings. The circuit might tell that there is a value of 1.xkW available, but evcc can control only the LP. Other consumers will change this value as well. If the LP / car cannot change the current from evcc fast enough, it will end up in this. We never can meet always a 100% accuracy
- there is no hysteresis for LM. on every cycle it takes the actual values from all meters, LP etc and performs the calculation. There might be ways to "smooth" all meter values for a number of periods, but this might lead to other side effects.

hyperbart, 2023-05-24:
I run it on Docker, so if you publish something, I download it, tag it, rm my current stack, deploy your stack and here we go.

hoermto, 2023-05-25:
@hyperbart I have pushed a new docker image `*_guard` to [releases](https://github.com/hoermto/evcc-tho/releases/tag/0.117.4-loadmanagement). Here it will *not* forcely stop the LP if overload was detected, and wait for the guard time according LP configuration.

If there is a guard time of e.g. 5 min, it will keep potentially for this time overload. 

Your effort and feedback is very much appreciated.

hyperbart, 2023-05-31:
@hoermto : yesterday I deployed your image but the same behavior.