evcc-io/evcc #7950: Too much power is drawn starting in 17.0.0

timostr, 2023-05-12:
### Describe the bug

Since new release 17.0.0 full 11.1 kW is drawn even if the surplus (Überschuss) is much less.
The battery can't compensate. Result is that energy is taken from the grid.

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: https
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: xxx.ml
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: "xx"

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true
plant: xxx

# log settings
log: debug # info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
#  - port: 5200
#    uri: 192.168.188.56:502 # Vom Wechselrichter
#    # rtu: true
#    # readonly: true

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
[... 266 lines cut ...]
[site  ] DEBUG 2023/05/12 11:59:21 pv power: 7334W
[site  ] DEBUG 2023/05/12 11:59:21 battery soc: 100%
[site  ] DEBUG 2023/05/12 11:59:21 battery power: 0W
[site  ] DEBUG 2023/05/12 11:59:21 grid power: 4138W
[site  ] DEBUG 2023/05/12 11:59:21 site power: 4238W
[lp-1  ] DEBUG 2023/05/12 11:59:21 charge currents: [15.5 15.5 15.6]A
[lp-1  ] DEBUG 2023/05/12 11:59:21 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/12 11:59:21 charge meter total import: 1611.507kWh
[lp-1  ] DEBUG 2023/05/12 11:59:21 charger status: C
[lp-1  ] DEBUG 2023/05/12 11:59:21 soc estimated: 78.02% (vehicle: 78.00%)
[lp-1  ] DEBUG 2023/05/12 11:59:21 vehicle soc: 78%
[lp-1  ] DEBUG 2023/05/12 11:59:21 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/05/12 11:59:21 vehicle range: 381km
[lp-1  ] DEBUG 2023/05/12 11:59:21 pv charge current: 6.25A = 12.4A + -6.14A (4238W @ 3p)
[lp-1  ] DEBUG 2023/05/12 11:59:21 max charge current: 6.25A
[site  ] DEBUG 2023/05/12 11:59:21 telemetry: charge: Δ20/31Wh @ 10995W
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.117.2

VolkerK62, 2023-05-12:
Deine Wallbox macht nicht das, was man ihr sagt.
`[lp-1  ] DEBUG 2023/05/12 11:59:05 max charge current: 9.98A`

andig, 2023-05-12:
@timostr could you try setting

    keba: trace

to get more logging?

timostr, 2023-05-12:
Here the trace (Note: I changed meanwhile in the car the setting back to 80%):
`"E total": 16125117,
"Serial": "24165919",
"Sec": 356757
}
[keba  ] TRACE 2023/05/12 14:26:36 send to 192.168.188.99:7090 report 3
[keba  ] TRACE 2023/05/12 14:26:36 recv from 192.168.188.99:7090 {
"ID": "3",
"U1": 237,
"U2": 234,
"U3": 235,
"I1": 15476,
"I2": 15494,
"I3": 15512,
"P": 10949265,
"PF": 999,
"E pres": 2034,
"E total": 16125117,
"Serial": "24165919",
"Sec": 356757
}
[lp-1  ] DEBUG 2023/05/12 14:26:36 charge meter total import: 1612.512kWh
[keba  ] TRACE 2023/05/12 14:26:36 send to 192.168.188.99:7090 report 2
[keba  ] TRACE 2023/05/12 14:26:36 recv from 192.168.188.99:7090 {
"ID": "2",
"State": 3,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 16000,
"Max curr %": 266,
"Curr HW": 16000,
"Curr user": 16000,
"Curr FS": 16000,
"Tmo FS": 30,
"Curr timer": 16000,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 0,
"Input": 0,
"X2 phaseSwitch source": 0,
"X2 phaseSwitch": 0,
"Serial": "24165919",
"Sec": 356757
}
[lp-1  ] DEBUG 2023/05/12 14:26:36 charger status: C
[keba  ] TRACE 2023/05/12 14:26:36 send to 192.168.188.99:7090 report 100
[keba  ] TRACE 2023/05/12 14:26:36 recv from 192.168.188.99:7090 {
"ID": "100",
"Session ID": 187,
"Curr HW": 16000,
"E start": 16123083,
"E pres": 2034,
"started[s]": 1683893603,
"ended[s]": 0,
"started": "2023-05-12 12:13:23.000",
"ended": "0",
"reason": 5,
"timeQ": 0,
"RFID tag": "0000000000000000",
"RFID class": "3312e10f000000000000",
"Serial": "24165919",
"Sec": 356757
}
[lp-1  ] DEBUG 2023/05/12 14:26:36 soc estimated: 79.21% (vehicle: 79.00%)
[lp-1  ] DEBUG 2023/05/12 14:26:36 vehicle soc: 79%
[lp-1  ] DEBUG 2023/05/12 14:26:36 vehicle soc limit: 70%
[lp-1  ] DEBUG 2023/05/12 14:26:36 vehicle range: 385km
[keba  ] TRACE 2023/05/12 14:26:36 send to 192.168.188.99:7090 report 2
[keba  ] TRACE 2023/05/12 14:26:36 recv from 192.168.188.99:7090 {
"ID": "2",
"State": 3,
"Error1": 0,
"Error2": 0,
"Plug": 7,
"AuthON": 0,
"Authreq": 0,
"Enable sys": 1,
"Enable user": 1,
"Max curr": 16000,
"Max curr %": 266,
"Curr HW": 16000,
"Curr user": 16000,
"Curr FS": 16000,
"Tmo FS": 30,
"Curr timer": 16000,
"Tmo CT": 0,
"Setenergy": 0,
"Output": 0,
"Input": 0,
"X2 phaseSwitch source": 0,
"X2 phaseSwitch": 0,
"Serial": "24165919",
"Sec": 356757
}
[lp-1  ] DEBUG 2023/05/12 14:26:36 pv charge current: 0A = 6A + -12.9A (8902W @ 3p)
[modbus] TRACE 2023/05/12 14:26:41 modbus: closing connection due to idle timeout: 5.003487138s`

closed 2023-05-12 by andig