evcc-io/evcc #7964: Massive Timeouts Alpha ESS T10 Speicher über Modbvus RTU RS485 USB-Adapter

Nitrox912, 2023-05-12:
### Describe the bug

Bei der Einbindung eines Alpha ESS T10 Speichers kommt es zumindest bei mir zu massiven Timeout-Problemen bei der Abfrage der Daten über eine Modbus RTU-Verbindung per RS485 USB-Adapter am Raspberry Pi. Die Daten werden dargestellt, jedoch kommt es immer wieder zu Abbrüchen oder Paketfehlern. In dem entsprechenden Augenblick kann das System somit nicht richtig arbeiten.

In einem anderen Anwenderforum wurde berichtet das die Anpassung des Intervalls zwischen zwei Bytes auf 40-50ms eine Verbesserung brachte. Inwiefern das für diesen Fall zutrifft, kann ich als Laie leider nicht beurteilen, halte es aber für eine möglich Spur. 

Ich habe bereits verschiedenen Adapter, Kabel und auch einen anderen Raspberry probiert. Das Ergebnis war immer das Gleiche. So lässt sich EVCC in Verbindung mit diesem Speicher leider nicht vernünftig nutzen und ich bin mit meinem Latein auch am Ende.

### Steps to reproduce

Es ist kein Muster zu erkennen, inwiefern sich die Timeouts reproduzieren lassen. Mal läuft es ein paar Minuten gut, mal fliegen alle 10 Sekunden die Timeout rein.

### Configuration details

```yaml
meters:
- type: template
  template: alpha-ess-smile 
  id: 85  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: grid  
  modbus: rs485serial  
  name: grid1

- type: template
  template: alpha-ess-smile 
  id: 85  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: pv  
  modbus: rs485serial  
  name: pv2

- type: template
  template: alpha-ess-smile 
  id: 85  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: battery  
  modbus: rs485serial  
  name: battery3
```

### Log details

```text
[lp-1  ] DEBUG 2023/05/12 17:28:22 charge power: 0W
[site  ] ERROR 2023/05/12 17:28:25 pv 1 power: add[3]: read failed: serial: timeout
[site  ] DEBUG 2023/05/12 17:28:25 pv power: 0W
[site  ] DEBUG 2023/05/12 17:28:25 battery soc: 100%
[site  ] DEBUG 2023/05/12 17:28:25 battery power: 6438W
[site  ] DEBUG 2023/05/12 17:28:26 grid power: -7201W
[site  ] DEBUG 2023/05/12 17:28:32 ----
[lp-1  ] DEBUG 2023/05/12 17:28:32 charge power: 0W
[site  ] DEBUG 2023/05/12 17:28:33 pv power: 6842W
[site  ] DEBUG 2023/05/12 17:28:33 battery soc: 100%
[site  ] DEBUG 2023/05/12 17:28:33 battery power: 0W
[site  ] DEBUG 2023/05/12 17:28:33 grid power: -4213W
[site  ] DEBUG 2023/05/12 17:28:34 grid currents: [0.8 0.2 0.9]A
[site  ] DEBUG 2023/05/12 17:28:34 site power: -4213W
[lp-1  ] DEBUG 2023/05/12 17:28:34 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/12 17:28:34 charge meter total import: 1571.734kWh
[lp-1  ] DEBUG 2023/05/12 17:28:34 charger status: B
[lp-1  ] DEBUG 2023/05/12 17:28:34 final charge energy: 0.205kWh
[lp-1  ] INFO 2023/05/12 17:28:34 stop charging <-
[lp-1  ] DEBUG 2023/05/12 17:28:34 wake-up timer: start
[lp-1  ] DEBUG 2023/05/12 17:28:34 charge meter total import: 1571.734kWh
[lp-1  ] DEBUG 2023/05/12 17:28:35 soc estimated: 74.23% (vehicle: 74.00%)
[lp-1  ] DEBUG 2023/05/12 17:28:35 vehicle soc: 74%
[lp-1  ] DEBUG 2023/05/12 17:28:35 vehicle soc limit: 90%
[lp-1  ] DEBUG 2023/05/12 17:28:35 vehicle range: 393km
[site  ] DEBUG 2023/05/12 17:28:42 ----
[lp-1  ] DEBUG 2023/05/12 17:28:42 charge power: 0W
[site  ] ERROR 2023/05/12 17:28:45 pv 1 power: add[6]: read failed: serial: timeout
[site  ] DEBUG 2023/05/12 17:28:45 pv power: 0W
[site  ] ERROR 2023/05/12 17:28:46 battery 1 soc: read failed: serial: timeout
[site  ] DEBUG 2023/05/12 17:28:46 battery soc: 0%
[site  ] DEBUG 2023/05/12 17:28:46 battery power: 0W
[site  ] DEBUG 2023/05/12 17:28:46 grid power: -4213W
[site  ] DEBUG 2023/05/12 17:28:52 ----
[lp-1  ] DEBUG 2023/05/12 17:28:52 charge power: 0W
[site  ] DEBUG 2023/05/12 17:28:53 pv power: 6842W
[site  ] DEBUG 2023/05/12 17:28:53 battery soc: 100%
[site  ] DEBUG 2023/05/12 17:28:53 battery power: 0W
[site  ] DEBUG 2023/05/12 17:28:53 grid power: -4207W
[site  ] DEBUG 2023/05/12 17:28:54 grid currents: [0.8 0.2 0.9]A
[site  ] DEBUG 2023/05/12 17:28:54 site power: -4207W
[lp-1  ] DEBUG 2023/05/12 17:28:54 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/12 17:28:54 charge meter total import: 1571.734kWh
[lp-1  ] DEBUG 2023/05/12 17:28:54 charger status: B
```

### What type of operating system are you running?

Linux

### Version

0.117.2

premultiply, 2023-05-13:
Greift ein weiteres System / App etc. parallel zu?

Nitrox912, 2023-05-13:
Nein, evcc ist das einzige System, dass auf den Modbus des Speichers zugreift.

Nitrox912, 2023-05-13:
Sobald ich den RS485 Adapter allerdings abstecke, erscheint zusätzlich der Fehler battery 1 power. Ich würde also darauf zurückführen, dass diese Verbindung im Gegensatz zu _pv 1 power_, _battery 1 soc_ und _grid energy_ stabil läuft.

[image]

Nitrox912, 2023-05-14:
Ich habe mit Modbus Poll den Alpha ESS Speicher ausgelesen und etwas experimentiert. Hier sind überhaupt keine Verbindungsabbrüche aufgetreten. Auch nach dem Verändern des Delays zwischen den Bytes konnte ich keine Probleme feststellen.

premultiply, 2023-05-15:
Mit welchem Datentyp?

andig, 2023-05-15:
Mal `delay` konfigurieren?

Nitrox912, 2023-05-16:
Sorry, aber mir fehlt da das Wissen dies ohne Anleitung so zu konfigurieren. Kann ich ```delay``` einfach so in Meters konfigurieren oder wo soll der Befehl ausgeführt werden?