evcc-io/evcc #7966: iOS / Android App with ionicframework

rene-mueller, 2023-05-12:
I would like to develop an Android and iOS app that connects to the API and performs actions. Is there any interest in this?

andig, 2023-05-12:
Absolut 👍🏻. Magst dich mal bei Slack melden um zu schauen, was Ambition, Abgrenzung, v.a. aber auch Vorteile wären? Go kann man übrigens nach WASM compilieren falls das hilft

rene-mueller, 2023-05-12:
@andig hab dich mal angeschrieben. Wenn du magst können wir kurz dazu mal huddln 🙂

foto-andreas, 2023-05-13:
Ich bin mit der Web-App auf dem Home-Screen ganz zufrieden. Nur das Login drumrum ist aktuell wegen BasicAuth nervig, da das nicht abgespeichert wird und auch noch zweimal angegeben werden muss. Werde da aber mal eine Umstellung auf Traefik mit Token probieren. Vorteile würden mich daher auch interessieren.

rene-mueller, 2023-05-13:
@foto-andreas ich habe es auch aktuell über eine Verknüpfung auf dem Homescreen gelöst.

Eine native App ist direkt installiert und bietet potenzielle Vorteile, was die Verwaltung auf Smartwatches oder Widgets auf den Geräten angeht. Zusätzlich würde eine native App Probleme wie Authentifizierung für weniger technisch versierte Personen verbessern und vereinfachen.

naltatis, 2023-05-13:
Ja find ich gut, mach das. Wir haben auch schon mehrfach diskutiert ob wir eine einfache Remote-Zugriff Option in evcc einbauen/bereitstellen. Ist gerade nicht weit oben auf der Prioritätenliste. @andig hatte mal mit Tailscale experimentiert. Aber so etwas würde sich dann ja ggf. auch gut ergänzen.

Analog zu Andis Angebot: sprich mich gerne auf Slack an! Oder erstell ggf. nen offenen Kanal um das Thema dort zu vertiefen.

rene-mueller, 2023-05-13:
@naltatis ich kann leider keinen Kanal erstellen. Ich habe dich aber angeschrieben 😉

naltatis, 2023-05-13:
Super. Ich hab #android-app angelegt.

rene-mueller, 2023-05-13:
@naltatis wollen wir kurz mal huddln?

andig, 2023-05-13:
Moved to Slack

closed 2023-05-13 by andig