evcc-io/evcc #7968: Odometer shown in session overview

martijndierckx, 2023-05-13:
**Is your feature request related to a problem? Please describe.**
I would like to follow up on my odometer and the estimated usage of kwh/100km (assuming you only charge at home).

**Describe the solution you'd like**
I expect this information as 2 columns in the session overview

**Additional context**
I have a Taycan which is configured as a custom vehicle (because I get the data internally via a different application):
```
vehicles:
  - name: Taycan
    type: custom
    title: Taycan
    capacity: 93 # kWh
    phases: 3
    soc:
      source: http
      uri: http://porsche-ev-influxdb:9040/data
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      insecure: false # set to true to trust self-signed certificates
      jq: .batteryLevel # parse response json
      scale: 1 # floating point factor applied to result, e.g. for kW to W conversion
      timeout: 10s # timeout in golang duration format, see https://golang.
    range:
      source: http
      uri: http://porsche-ev-influxdb:9040/data
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      insecure: false # set to true to trust self-signed certificates
      jq: .remainingElectricRange # parse response json
      scale: 1 # floating point factor applied to result, e.g. for kW to W conversion
      timeout: 10s # timeout in golang duration format, see https://golang.
    status:
      source: http
      uri: http://porsche-ev-influxdb:9040/data
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      insecure: false # set to true to trust self-signed certificates
      jq: if .charging == true then "K" elif .pluggedIn == true then "B" else "A" end
      timeout: 10s # timeout in golang duration format, see https://golang.
    odometer:
      source: http
      uri: http://porsche-ev-influxdb:9040/data
      method: GET # default HTTP method
      headers:
      - content-type: application/json
      insecure: false # set to true to trust self-signed certificates
      jq: .mileage # parse response json
      scale: 1 # floating point factor applied to result, e.g. for kW to W conversion
      timeout: 10s # timeout in golang duration format, see https://golang.
    onIdentify: # set defaults when vehicle is identified
      mode: now
      minSoC: 25
      targetSoC: 85
```

This is my current session overview:
[image]

VolkerK62, 2023-05-13:
click on a session, the detail window comes up, there you have the odometer. The odometer is also listed in the csv-file.

martijndierckx, 2023-05-13:
Great! That works! Feel like an idiot now :-)
Maybe add an icon at the end of each row (on hover) to indicate there’s more to see. Wasn’t clear now.

VolkerK62, 2023-05-13:
>Maybe add an icon at the end of each row (on hover) to indicate there’s more to see. 

Seems to be a good idea
cc/ @naltatis

andig, 2023-05-13:
Ich bin gar nicht glücklich mit der mittlerweile total unübersichtlichen Anzeige. Ich hätte lieber weniger und den Rest im csv oder Detailview.

VolkerK62, 2023-05-13:
es geht ja nur darum, dass man nicht erkennt, dass sich dahinter noch Details verbergen.
Option: Hinweis in der Überschrift: "Ladevorgänge (Klick auf Session öffnet Details)"

andig, 2023-05-13:
Am Desktop sehe ich das geänderte Mausicon, auf Mobile klappt das nicht.

naltatis, 2023-05-15:
> Ich bin gar nicht glücklich mit der mittlerweile total unübersichtlichen Anzeige.

Ja, hatten wir beide doch schon drüber gesprochen. Jetzt wo wir hier auch Preis und % PV Daten sammeln macht es als nächsten Schritt total viel Sinn über eine grafische Auswertung nachzudenken.

Zusätzlich (aber vermutlich weniger präsent) würd ich die Daten immer noch tabellarisch im Web UI verfügbar halten. Alle Ladevorgänge einzeln anklicken oder ein CSV runterladen ist keine wirkliche Alternative.

Das ist aber ein eigener Themenkomplex. Würde ich jetzt hier aus der Diskussion erstmal rausnehmen und mich auf die "Discoverability" der Session Detailansicht beschränken.

andig, 2023-05-16:
> Jetzt wo wir hier auch Preis und % PV Daten sammeln macht es als nächsten Schritt total viel Sinn über eine grafische Auswertung nachzudenken.

Dafür hatte ich den Excelexport gedacht und würde aktuell lieber an anderer Stelle Aufwand spendieren.

> oder ein CSV runterladen ist keine wirkliche Alternative.

Für "alle" Vorgänge oder Werte finde ich das perfekt.

VolkerK62, 2023-05-17:
>>oder ein CSV runterladen ist keine wirkliche Alternative.

>Für "alle" Vorgänge oder Werte finde ich das perfekt.

Das sehe ich (als Excel Fan) auch so.

wobei bei "alle" noch der Soc fehlt #6144  ;-)

StevieC121176, 2023-05-18:
Ich würde eine grafische Auswertung deutlich bevorzugen. Da wir zu Hause nur privat laden hätte man so den Verbrauch und die etwaigen Kosten schön im Blick ohne Umwege über andere Programme.

PR #8586 (2023-06-23): Sessions: only show columns if relevant values are present
- vehicle and loadpoint columns are only shown if multiple values exist (fixes #8561)
- underline time column values to indicate that rows are clickable (fixes #7968)
closed 2023-06-23 by naltatis