evcc-io/evcc #7971: Nicht verbundene Wallbox ist in der Oberfläche nicht erkennbar

micw, 2023-05-13:
### Describe the bug

Ich habe zwei Walboxen (Wattpilot), zu einem der beiden geht immer mal die Verbindung verloren.

Leider kann ich in der Oberfläche nicht erkennen
* zu welchem der beiden die Verbindung verloren geht
* Ob diese wieder da ist

### Steps to reproduce

1. 2 Wallboxen konfigurieren
2. Verbindung zu einer davon trennen
3. Versuchen herauszufinden, welche nicht verbunden ist

Im Event-Log ist nicht zu erkennen: um welche der Wallboxen es sich handelt

[image]

In der Oberfläche ist nicht zu erkennen, dass die Wallbox aktuell nicht verbunden ist:

[image]

### Configuration details

```yaml
chargers:
          - name: Garage
            type: template
            template: fronius-wattpilot
            host: 192.168.1.164
            password: ...

          - name: Carport
            type: template
            template: fronius-wattpilot
            host: 192.168.1.165
            password: ...
```

### Log details

```text
[site  ] INFO 2023/05/13 10:00:34     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/05/13 10:00:34   vehicles:
[site  ] INFO 2023/05/13 10:00:34     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2023/05/13 10:00:34     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/05/13 10:00:34 loadpoint 1:
[lp-1  ] INFO 2023/05/13 10:00:34   mode:        pv
[lp-1  ] INFO 2023/05/13 10:00:34   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/05/13 10:00:34   meters:      charge ✓
[lp-1  ] INFO 2023/05/13 10:00:34     charge:    power ✓ energy ✗ currents ✓
[lp-2  ] INFO 2023/05/13 10:00:34 loadpoint 2:
[lp-2  ] INFO 2023/05/13 10:00:34   mode:        pv
[lp-2  ] INFO 2023/05/13 10:00:34   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-2  ] INFO 2023/05/13 10:00:34   meters:      charge ✓
[lp-2  ] INFO 2023/05/13 10:00:34     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2023/05/13 10:00:34 vehicle updated: unknown -> Leaf
[lp-2  ] INFO 2023/05/13 10:00:34 vehicle updated: unknown -> Enyaq
[lp-1  ] INFO 2023/05/13 10:00:34 car connected
[lp-2  ] INFO 2023/05/13 10:00:45 car connected
[lp-2  ] ERROR 2023/05/13 10:04:54 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:04 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:04 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:04 charge rater: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:04 charger: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:14 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:24 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:24 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:24 charge rater: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:24 charger: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:34 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:44 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:44 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:44 charge rater: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:44 charger: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:05:54 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:06:04 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:06:04 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:06:04 charge rater: connection is not valid
[lp-2  ] ERROR 2023/05/13 10:06:04 charger: connection is not valid
```

### What type of operating system are you running?

Linux

### Version

0.117.2

micw, 2023-05-13:
Im Log sehe ich gerade, dass da "lp-2" steht. D.h. auf Log-Level lässt es sich erkennen. In der Oberfläche jedoch leider nicht

andig, 2023-05-13:
Das ist in der Tat blöd. Langfristig wollen wir das verbessern, ich aber aktuell keine Prio.

naltatis, 2023-07-13:
Ja, ich sehe das auch als Teil bzw. Ausbau von #6029.
Dort werden wir alle konfigurierten Geräte in einer Liste darstellen. An der Stelle könnten wir gut einen "Gesundheitszustand" des jeweiligen Devices anzeigen.

micw, 2023-07-13:
Wenn der Zustand bekannt ist, wäre es schon eine _sehr_ große Hilfe, wenn die Box mit dem Ladepunkt einfach ein deutlich sichtbares Fehlersymbol anzeigt. Dann weiß Frau, das was nicht stimmt und startet nicht überraschend am nächsten Morgen mit einem nicht geladenen Auto.

maikheinrich, 2023-09-24:
Hallo zusammen,

ein weiterer Verbesserungsvorschlag... hilfreich wäre es außerdem, wenn das message system bei einem Fehler auch eine Notification senden würde (email, MQTT, ...) 

LG Maik

closed 2023-10-01 by andig