evcc-io/evcc #7990: Falsche API Adresse für Solax Cloud

DedekDDK, 2023-05-14:
### Describe the bug

Ich konfiguriere meinen Solax Hybrid Wechselrichter die Messungen aus dem Solax Cloud zu lesen (weil Verbindung über Modbus muss ich erst anbinden) und ich bekommen Fehler Internal Server Error (HTTP 500).
Wenn ich in Github gesucht habe, die API Adresse im GO ist andere als wa SolaxCloud in die Doku hat. Das hat sich vielleicht geändert.
Wenn ich über Postman die richtige API verwende, server antwortet richtig.

Ich glaube wenn nur die API Adresse geändert worden ist, alles sollte wieder funktionieren.

API Adresse die mir funktioniert: https://www.solaxcloud.com/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=xxxx&sn=nnn

Die API Adresse im templates/definition/meter/solax-hybrid-cloud.yaml
uri: https://www.eu.solaxcloud.com:9443/proxy/api/getRealtimeInfo.do?tokenId={{ .tokenid }}&sn={{ .serial}}

### Steps to reproduce

1. Konfiguriere Meter als Solax Cloud
2. Eingebe den Token Id (auf den Solax Cloud Account Seiten generiert)
3. Sehe im Logs, wo steht: [site  ] ERROR 2023/05/01 23:57:30 grid meter: unexpected status: 500 (Internal Server Error)

### Configuration details

```yaml
meters:
- type: template
  template: solax-hybrid-cloud 
  usage: grid  
  tokenid: --removed--  
  serial: --removed--
  name: grid1
- type: template
  template: solax-hybrid-cloud 
  usage: pv  
  tokenid: --removed--  
  serial: --removed--  
  name: pv2
- type: template
  template: solax-hybrid-cloud 
  usage: battery  
  tokenid: --removed--  
  serial: --removed--  
  name: battery3
```

### Log details

```text
[eebus ] DEBUG 2023/05/01 23:57:25 restarting mdns search
[eebus ] DEBUG 2023/05/01 23:57:25 delaying connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf by 2.272s to minimize double connection probability
[eebus ] DEBUG 2023/05/01 23:57:27 trying to connect to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123
[eebus ] DEBUG 2023/05/01 23:57:27 initiating connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123:4712
[site  ] INFO 2023/05/01 23:57:27 site config:
[site  ] INFO 2023/05/01 23:57:27   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/05/01 23:57:27     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/01 23:57:27     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/01 23:57:27     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/05/01 23:57:27   vehicles:
[site  ] INFO 2023/05/01 23:57:27     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/05/01 23:57:27 loadpoint 1:
[lp-1  ] INFO 2023/05/01 23:57:27   mode:        off
[lp-1  ] INFO 2023/05/01 23:57:27   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/05/01 23:57:27   meters:      charge ✓
[lp-1  ] INFO 2023/05/01 23:57:27     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2023/05/01 23:57:27 phase timer inactive
[lp-1  ] DEBUG 2023/05/01 23:57:27 pv timer inactive
[lp-1  ] DEBUG 2023/05/01 23:57:27 guard timer inactive
[site  ] DEBUG 2023/05/01 23:57:27 ----
[lp-1  ] DEBUG 2023/05/01 23:57:27 charge power: 0W
[eebus ] DEBUG 2023/05/01 23:57:28 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/05/01 23:57:28 44eb7bb6ef5df511614574951a0518dc1d70daaf websocket read error:  websocket: close 4450: SHIP id mismatch
[eebus ] DEBUG 2023/05/01 23:57:28 restarting mdns search
[eebus ] DEBUG 2023/05/01 23:57:28 delaying connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf by 2.208s to minimize double connection probability
[site  ] ERROR 2023/05/01 23:57:28 pv 1 power: add[0]: unexpected status: 500 (Internal Server Error)
[site  ] DEBUG 2023/05/01 23:57:28 pv power: 0W
[site  ] ERROR 2023/05/01 23:57:29 battery 1 power: unexpected status: 500 (Internal Server Error)
[site  ] ERROR 2023/05/01 23:57:29 battery 1 soc: unexpected status: 500 (Internal Server Error)
[site  ] DEBUG 2023/05/01 23:57:29 battery soc: 0%
[site  ] DEBUG 2023/05/01 23:57:29 battery power: 0W
[site  ] ERROR 2023/05/01 23:57:30 grid meter: unexpected status: 500 (Internal Server Error)
[eebus ] DEBUG 2023/05/01 23:57:30 trying to connect to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123
[eebus ] DEBUG 2023/05/01 23:57:30 initiating connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123:4712
[eebus ] DEBUG 2023/05/01 23:57:30 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/05/01 23:57:30 44eb7bb6ef5df511614574951a0518dc1d70daaf websocket read error:  websocket: close 4450: SHIP id mismatch
[eebus ] DEBUG 2023/05/01 23:57:30 restarting mdns search
[eebus ] DEBUG 2023/05/01 23:57:30 delaying connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf by 2.129s to minimize double connection probability
[eebus ] DEBUG 2023/05/01 23:57:32 trying to connect to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123
[eebus ] DEBUG 2023/05/01 23:57:32 initiating connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123:4712
[eebus ] DEBUG 2023/05/01 23:57:33 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/05/01 23:57:33 44eb7bb6ef5df511614574951a0518dc1d70daaf websocket read error:  websocket: close 4450: SHIP id mismatch
[eebus ] DEBUG 2023/05/01 23:57:33 restarting mdns search
[eebus ] DEBUG 2023/05/01 23:57:33 delaying connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf by 1.316s to minimize double connection probability
[eebus ] DEBUG 2023/05/01 23:57:34 trying to connect to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123
[eebus ] DEBUG 2023/05/01 23:57:34 initiating connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123:4712
[eebus ] DEBUG 2023/05/01 23:57:35 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2023/05/01 23:57:35 44eb7bb6ef5df511614574951a0518dc1d70daaf websocket read error:  websocket: close 4450: SHIP id mismatch
[eebus ] DEBUG 2023/05/01 23:57:35 restarting mdns search
[eebus ] DEBUG 2023/05/01 23:57:35 delaying connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf by 1.77s to minimize double connection probability
[eebus ] DEBUG 2023/05/01 23:57:37 trying to connect to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123
[eebus ] DEBUG 2023/05/01 23:57:37 initiating connection to 44eb7bb6ef5df511614574951a0518dc1d70daaf at 192.168.1.123:4712
[site  ] DEBUG 2023/05/01 23:57:37 ----
[lp-1  ] DEBUG 2023/05/01 23:57:37 charge power: 0W
[eebus ] DEBUG 2023/05/01 23:57:37 Send:  read 1 NodeManagementDetailedDiscoveryData
[site  ] ERROR 2023/05/01 23:57:37 pv 1 power: add[0]: unexpected status: 500 (Internal Server Error)
[site  ] DEBUG 2023/05/01 23:57:37 pv power: 0W
[eebus ] DEBUG 2023/05/01 23:57:37 44eb7bb6ef5df511614574951a0518dc1d70daaf websocket read error:  websocket: close 4450: SHIP id mismatch
[eebus ] DEBUG 2023/05/01 23:57:37 restarting mdns search
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.116.7

andig, 2023-05-14:
Laut https://www.eu.solaxcloud.com/phoebus/resource/files/userGuide/Solax_API.pdf:

> Solax API interface accesses Solax cloud platform data through the Http protocol, and the data is returned in JSON format. Interface address:https://www.solaxcloud.com:9443/proxy/api/

andig, 2023-05-14:
Laut https://www.solaxcloud.com/user_api/SolaxCloud_User_Monitoring_API_V6.1.pdf:

> Solax API interface accesses Solax cloud platform data through the Http protocol, and the data is returned in JSON format. Interface address:https://www.solaxcloud.com/proxyApp/proxy/api/

closed 2023-05-14 by andig

DedekDDK, 2023-05-14:
Danke für die schnelle Änderung!
Ist es möglich diese Parameter irgendwie overriden in der config Datei oder muss ich auf neue Version warten?
David

PR #7991 (2023-05-14): Fix Solax Cloud URL
Fixes #7990