evcc-io/evcc #7992: Ladestand Fahrzeug wird nicht aktualisiert, wenn Ladepunkt nicht abgefragt werden kann

micw, 2023-05-14:
### Describe the bug

Mein Ladepunkt (Wattpilot) ist immer mal nicht erreichbar (anderer Bug). Wenn das der Fall ist, wird allerdings auch nicht mehr abgefragt, wie der Ladestand vom Fahrzeug ist.

[image]

Das Auto hat inzwischen 70%, die Anzeige ist bei 40% stehen geblieben.

### Steps to reproduce

1. Fahrzeug laden
2. TCP-Verbindung zum Ladepunkt unterbrechen
3. Sehen, dass der Ladestand des Fahrzeugs nicht aktualisiert wird

### Configuration details

```yaml
vehicles:
          - name: Leaf
            title: Leaf
            type: template
            template: nissan
            user: anke@wyraz.de
            password: <SECRET>cgsrxgHVuveRwuGtFcfYokU=
            vin: <SECRET>cgsrxgHKqM6H2dGkPYHeo1Wp0ZZjjg==
            capacity: 40
            phases: 1

          - name: Enyaq
            title: Enyaq
            type: template
            template: enyaq
            user: skoda@michael.wyraz.de
            password: <SECRET>cgsrxgHwg+f47PWfAtHi9Az5hec=
            capacity: 80
            phases: 3

        chargers:
          - name: Garage
            type: template
            template: fronius-wattpilot
            host: 192.168.1.164
            password: <SECRET>cgsrxgH2ivCpr8WRE4fq+wa5

          - name: Carport
            type: template
            template: fronius-wattpilot
            host: 192.168.1.165
            password: <SECRET>cgsrxgH2ivCpr8WRE4fq+wa5

        loadpoints:
          - title: Garage
            charger: Garage
            vehicle: Leaf
            mode: pv
            maxCurrent: 14
            phases: 0
            enable: # mind. 10 Minuten lang 300 Watt Einspeisung
              delay: 10m
              threshold: -300
            disable: # mind. 10 Minuten lang 1200 Watt Bezug
              delay: 10m
              threshold: 1200
          - title: Carport
            charger: Carport
            vehicle: Enyaq
            mode: pv
            maxCurrent: 14
            phases: 0
            enable: # mind. 10 Minuten lang 300 Watt Einspeisung
              delay: 10m
              threshold: -300
            disable: # mind. 10 Minuten lang 1200 Watt Bezug
              delay: 10m
              threshold: 1200
```

### Log details

```text
[lp-1  ] DEBUG 2023/05/14 13:22:55 charge power: 0W
[lp-2  ] ERROR 2023/05/14 13:22:55 charge meter: connection is not valid
[site  ] DEBUG 2023/05/14 13:22:55 pv power: 326W
[site  ] DEBUG 2023/05/14 13:22:55 grid power: 10270W
[site  ] DEBUG 2023/05/14 13:22:55 site power: 10270W
[lp-2  ] ERROR 2023/05/14 13:22:55 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/14 13:22:55 charge rater: connection is not valid
[lp-2  ] ERROR 2023/05/14 13:22:55 charger: connection is not valid
[site  ] DEBUG 2023/05/14 13:23:05 ----
[lp-1  ] DEBUG 2023/05/14 13:23:05 charge power: 0W
[lp-2  ] ERROR 2023/05/14 13:23:05 charge meter: connection is not valid
[site  ] DEBUG 2023/05/14 13:23:05 pv power: 325W
[site  ] DEBUG 2023/05/14 13:23:05 grid power: 10363W
[site  ] DEBUG 2023/05/14 13:23:05 site power: 10363W
[lp-1  ] DEBUG 2023/05/14 13:23:05 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/14 13:23:05 charger status: A
[site  ] DEBUG 2023/05/14 13:23:15 ----
[lp-1  ] DEBUG 2023/05/14 13:23:15 charge power: 0W
[lp-2  ] ERROR 2023/05/14 13:23:15 charge meter: connection is not valid
[site  ] DEBUG 2023/05/14 13:23:15 pv power: 323W
[site  ] DEBUG 2023/05/14 13:23:15 grid power: 10254W
[site  ] DEBUG 2023/05/14 13:23:15 site power: 10254W
[lp-2  ] ERROR 2023/05/14 13:23:15 charge meter: connection is not valid
[lp-2  ] ERROR 2023/05/14 13:23:15 charge rater: connection is not valid
[lp-2  ] ERROR 2023/05/14 13:23:15 charger: connection is not valid
[site  ] DEBUG 2023/05/14 13:23:25 ----
[lp-1  ] DEBUG 2023/05/14 13:23:25 charge power: 0W
[lp-2  ] ERROR 2023/05/14 13:23:25 charge meter: connection is not valid
[site  ] DEBUG 2023/05/14 13:23:25 pv power: 323W
[site  ] DEBUG 2023/05/14 13:23:25 grid power: 10344W
[site  ] DEBUG 2023/05/14 13:23:25 site power: 10344W
[lp-1  ] DEBUG 2023/05/14 13:23:25 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/14 13:23:25 charger status: A
```

### What type of operating system are you running?

Docker container

### Version

0.117.2

micw, 2023-05-14:
Relevant ist sicher noch, dass der Ladestand des Fahrzeugs von Skoda kommt und nicht vom Ladepunkt

VolkerK62, 2023-05-14:
Der Soc kommt zwar vom Fahrzeug, aber der Ladepunkt triggert die Abfrage. Wenn der nicht "vorhanden" ist, gibt es auch kein angeschlossenes Fahrzeug und somit keine Soc-Abfrage.
Du könntest versuchen, die Abfrage zu triggern, auch wenn das Fahrzeug nicht verbunden ist.
https://docs.evcc.io/docs/reference/configuration/loadpoints#soc

micw, 2023-05-14:
Verstehe. Ich denke, dann sollte dort aber nach einer gewissen Zeit "offline" oder sowas stehen. Ganz korrekt wäre aber, wenn das Fahrzeug auch abgefragt würde, wenn der Ladepunkt nicht da ist.

VolkerK62, 2023-05-14:
Wie schon gesagt, per default werden nur Fahrzeuge abgefragt, die auch mit einem Ladepunkt verbunden sind. Du kannst das anders einstellen.

micw, 2023-05-14:
Danke, jetzt habe ich den Zusammenhang verstanden. Wenn der Ladepunkt nicht verfügbar ist, läuft die Abfrageeinstellung für "nicht verbunden", welche bei mir auf "nicht abfragen" steht.
Das ist auch so gewollt. Allerdings sollte das Auto dann mindestens als "nicht verbunden" angezeigt werden.

andig, 2023-05-15:
Lasst uns darauf konzentrieren, den Fehler beim Wattpilot als root cause zu fixen.

/cc @mabunixda