evcc-io/evcc #7994: Batterieunterstütztes Laden mit maximaler Ladungleistung

fabiank88, 2023-05-14:
### Describe the bug

Ich habe einen Sonnencharger, PV und Batterie. Das Starten der Ladung klappt, auch wenn nicht genug PV da ist. Nach einiger Zeit wird aber einfach auf die Maximale Ladeleistung gewechselt und diese wird auch nicht mehr runtergeregelt

[image]

### Steps to reproduce

1. Sonnencharger mit Min+PV => 10kW

oder 

1. Sonnencharger im PV Modus, 3.5kw aus PV und Batterie genug um das Laden zu unterstützen ein wenig warten => 10kW

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

interval: 10s # control cycle interval

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 172.16.5.29  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 172.16.5.29  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 172.16.5.29  
  port: 8080  
  name: battery3
- type: custom
  power:
    source: js
    script: |
      var res = 0;
      res;
  name: auxbattery

chargers:
- type: template
  template: etrel 
  host: 172.16.5.40
  port: 502  
  name: wallbox6

vehicles:
- type: template
  template: porsche 
  title: Taycan 4s  
  user: xxx  
  password: xxx 
  vin: xxx  
  capacity: 93  
  name: ev4
- type: template
  template: bmw 
  title: iX50  
  user: xxx
  password: xxx  
  vin: xxx  
  capacity: 105  
  name: ev5

loadpoints:
- title: Garage
  charger: wallbox6
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: Haus
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
    aux: auxbattery
  #prioritySoc: 90
  bufferStartSoc: 95
  bufferSoc: 70
  residualPower: 100

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.42 # [currency]/kWh
  feedin:
    type: fixed
    price: 0.00 # [currency]/kWh
```

### Log details

```text
[lp-1  ] DEBUG 2023/05/14 13:36:23 charge power: 3776W
[site  ] DEBUG 2023/05/14 13:36:23 pv power: 1497W
[site  ] DEBUG 2023/05/14 13:36:23 battery soc: 72%
[site  ] DEBUG 2023/05/14 13:36:23 battery power: 2629W
[site  ] DEBUG 2023/05/14 13:36:23 grid power: 33W
[site  ] DEBUG 2023/05/14 13:36:23 aux power 1: 0W
[site  ] DEBUG 2023/05/14 13:36:23 aux power: 0W
[site  ] DEBUG 2023/05/14 13:36:23 site power: 2762W
[lp-1  ] DEBUG 2023/05/14 13:36:23 charge currents: [5.74 5.82 5.57]A
[lp-1  ] DEBUG 2023/05/14 13:36:23 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/14 13:36:23 charger status: C
[lp-1  ] DEBUG 2023/05/14 13:36:23 soc estimated: 55.10% (vehicle: 55.00%)
[lp-1  ] DEBUG 2023/05/14 13:36:23 vehicle soc: 55%
[lp-1  ] DEBUG 2023/05/14 13:36:23 vehicle range: 217km
[lp-1  ] DEBUG 2023/05/14 13:36:23 pv charge current: 2A = 6A + -4A (2762W @ 3p)
[site  ] DEBUG 2023/05/14 13:36:33 ----
[lp-1  ] DEBUG 2023/05/14 13:36:33 charge power: 10048W
[site  ] DEBUG 2023/05/14 13:36:33 pv power: 1492W
[site  ] DEBUG 2023/05/14 13:36:33 battery soc: 72%
[site  ] DEBUG 2023/05/14 13:36:33 battery power: 4301W
[site  ] DEBUG 2023/05/14 13:36:33 grid power: 5686W
[site  ] DEBUG 2023/05/14 13:36:33 aux power 1: 0W
[site  ] DEBUG 2023/05/14 13:36:33 aux power: 0W
[site  ] DEBUG 2023/05/14 13:36:33 site power: 10087W
[lp-1  ] DEBUG 2023/05/14 13:36:33 charge currents: [15.1 15.3 15]A
[lp-1  ] DEBUG 2023/05/14 13:36:33 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/14 13:36:33 charger status: C
[lp-1  ] DEBUG 2023/05/14 13:36:33 soc estimated: 55.13% (vehicle: 55.00%)
[lp-1  ] DEBUG 2023/05/14 13:36:33 vehicle soc: 55%
[lp-1  ] DEBUG 2023/05/14 13:36:33 vehicle range: 217km
[lp-1  ] DEBUG 2023/05/14 13:36:33 pv charge current: 0A = 6A + -14.6A (10087W @ 3p)
```

### What type of operating system are you running?

Linux

horst789github, 2023-05-15:
das erste was ich vermute ist das du hier was falsch eingestellt hast:
[image]
meine einstellung:
[image]

andig, 2023-05-15:
Ohne Logfile nicht zu beantworten.

closed 2023-05-15 by andig

fabiank88, 2023-05-19:
`Mai 19 17:16:01 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:01 ----
Mai 19 17:16:01 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:01 modbus: send 00 7c 00 00 00 06 01 04 00 1a 00 02
Mai 19 17:16:01 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:01 modbus: recv 00 7c 00 00 00 07 01 04 04 40 75 c2 8f
Mai 19 17:16:01 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:01 charge power: 3840W
Mai 19 17:16:01 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:01 pv power: 1943W
Mai 19 17:16:01 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:01 battery soc: 65%
Mai 19 17:16:01 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:01 battery power: 2573W
Mai 19 17:16:01 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:01 grid power: 6573W
Mai 19 17:16:01 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:01 aux power 1: 1600W
Mai 19 17:16:01 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:01 aux power: 1600W
Mai 19 17:16:01 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:01 site power: 7646W
Mai 19 17:16:01 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:01 modbus: send 00 7d 00 00 00 06 01 04 00 0e 00 06
Mai 19 17:16:01 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:01 modbus: recv 00 7d 00 00 00 0f 01 04 0c 41 73 33 33 41 73 33 33 41 72 b0 21
Mai 19 17:16:01 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:01 charge currents: [15.2 15.2 15.2]A
Mai 19 17:16:01 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:01 detected active phases: 3p
Mai 19 17:16:01 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:01 modbus: send 00 7e 00 00 00 06 01 04 00 00 00 01
Mai 19 17:16:01 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:01 modbus: recv 00 7e 00 00 00 05 01 04 02 00 04
Mai 19 17:16:01 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:01 charger status: C
Mai 19 17:16:01 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:01 soc estimated: 69.02% (vehicle: 69.00%)
Mai 19 17:16:01 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:01 vehicle soc: 69%
Mai 19 17:16:01 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:01 vehicle range: 275km
Mai 19 17:16:01 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:01 modbus: send 00 7f 00 00 00 06 01 04 00 04 00 02
Mai 19 17:16:01 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:01 modbus: recv 00 7f 00 00 00 07 01 04 04 41 80 00 00
Mai 19 17:16:01 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:01 pv charge current: 0A = 6A + -11.1A (7646W @ 3p)
Mai 19 17:16:06 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:06 ----
Mai 19 17:16:06 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:06 modbus: send 00 80 00 00 00 06 01 04 00 1a 00 02
Mai 19 17:16:06 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:06 modbus: recv 00 80 00 00 00 07 01 04 04 41 21 ca c1
Mai 19 17:16:06 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:06 charge power: 10112W
Mai 19 17:16:06 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:06 pv power: 2392W
Mai 19 17:16:06 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:06 battery soc: 65%
Mai 19 17:16:06 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:06 battery power: 4491W
Mai 19 17:16:06 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:06 grid power: 4721W
Mai 19 17:16:06 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:06 aux power 1: 1600W
Mai 19 17:16:06 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:06 aux power: 1600W
Mai 19 17:16:06 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:06 site power: 7712W
Mai 19 17:16:06 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:06 modbus: send 00 81 00 00 00 06 01 04 00 0e 00 06
Mai 19 17:16:06 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:06 modbus: recv 00 81 00 00 00 0f 01 04 0c 41 73 33 33 41 73 33 33 41 72 b0 21
Mai 19 17:16:06 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:06 charge currents: [15.2 15.2 15.2]A
Mai 19 17:16:06 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:06 detected active phases: 3p
Mai 19 17:16:06 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:06 modbus: send 00 82 00 00 00 06 01 04 00 00 00 01
Mai 19 17:16:06 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:06 modbus: recv 00 82 00 00 00 05 01 04 02 00 04
Mai 19 17:16:06 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:06 charger status: C
Mai 19 17:16:06 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:06 soc estimated: 69.03% (vehicle: 69.00%)
Mai 19 17:16:06 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:06 vehicle soc: 69%
Mai 19 17:16:06 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:06 vehicle range: 275km
Mai 19 17:16:06 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:06 modbus: send 00 83 00 00 00 06 01 04 00 04 00 02
Mai 19 17:16:06 fapi evcc[13750]: [etrel ] TRACE 2023/05/19 17:16:06 modbus: recv 00 83 00 00 00 07 01 04 04 41 80 00 00
Mai 19 17:16:06 fapi evcc[13750]: [lp-1  ] DEBUG 2023/05/19 17:16:06 pv charge current: 0A = 6A + -11.2A (7712W @ 3p)
Mai 19 17:16:11 fapi evcc[13750]: [site  ] DEBUG 2023/05/19 17:16:11 ----
`

fabiank88, 2023-05-19:
Ich stelle eigentlich nur auf Min+PV und warte ein bisschen und mache nichts. In dem Logfile ist der Zeitpunkt drin, in dem er dann von 3,8 auf 10kw springt, ohne dass ich irgendetwas mache.

Was bräuchtest du noch an logfile? Davor lädt er stetig mit dem Minimum 3,8kw.

Hatte nochmal eine neue Issue aufgemacht, dachte, dass closed, closed ist.

andig, 2023-05-19:
evcc ist der Meinung, das abgeschaltet werden sollte:

> pv charge current: 0A = 6A + -11.1A (7646W @ 3p)

Warum das nicht passiert erklärt sich nicht.

fabiank88, 2023-05-19:
Wie kann ich am besten helfen das zu debuggen?

andig, 2023-05-19:
Aussagekräftiges Logfile

fabiank88, 2023-05-19:
Also alles auf trace und länger laufen lassen oder noch etwas anderes?