evcc-io/evcc #7998: vehicle not available / can not connect

jwinkler1967, 2023-05-14:
### Describe the bug

Hallo Zusammen,

ich habe auf die Version 0.117.3 update, danach blieb mein Fahrzeug offline und folgendes wurde geloged:
ein Downgrade auf 0.117. beseitigte den Fehler nicht!

ich brauche Hilfe !

### Steps to reproduce

1. update auf 0.117.3
2. Downgrade 0.117.2 
3.
4.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: fiat 
  title: Fiat 500e  
  user: xxxxxxxxxxxxxx 
  password: xxxxxxxxx  
  vin: xxxxxxxxxxxxxxx  
  capacity: 42  
  pin: 'xxxxxx'  
  name: ev1
```

### Log details

```text
Vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'fiat': login failed: Get \"https://loginmyuconnect.fiat.com/accounts.webSdkBootstrap?APIKey=3_mOx_J2dRgjXYCdyhchv3b5lhi54eBcdCTX4BI8MORqmZCoQWhA0mV2PTlptLGUQI&format=json&pageURL=https%3A%2F%2Fmyuconnect.fiat.com%2Fde%2Fde%2Fvehicle-services&sdk=js_latest&sdkBuild=12234\": dial tcp: lookup loginmyuconnect.fiat.com: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

0.117.3 / 0.117.2

Hofyyy, 2023-05-14:
Wie wird evcc betrieben etc? docker? kann evcc sich mit dem internet verbinden? fiat api down? internet down? ist denke kein fehler in evcc.

jwinkler1967, 2023-05-14:
der evcc läuft bei mir auf einem Raspberry,. Internet down!? nein ich habe gerade das Update gezogen und natürlich die Fiat Cloud geprüft

Hofyyy, 2023-05-14:
Die fehlermeldung sagt halt, dass der login nicht klappt :-/

jwinkler1967, 2023-05-14:
Ja, danke, ich habe das INet von Raspberry geprüft, sin da. Ich denke das Auto wollte nicht mehr, auch da gab es mit mal ein großes "Achtung". Vermutlich dummer Zufall , ich vermute das die Fiat Cloud eine Macke hatte, das Auto verbindet sich über Mobilfunk und konnte sich ebenfalls nicht verbinden. doofer zufall das ich gerade das evccc update habe.

Jetzt klappt alles wieder, das für deine schnell Unterstützung

jwinkler1967, 2023-05-14:
alles OK

closed 2023-05-14 by jwinkler1967

lordvalium, 2023-08-22:
> Wie wird evcc betrieben etc? docker? kann evcc sich mit dem internet verbinden? fiat api down? internet down? ist denke kein fehler in evcc.

Wie kann man prüfen ob evcc sich mit dem Internet verbinden kann?