evcc-io/evcc #8006: Maximale Ladeparameter (Strom / Anzahl Phasen) durch kurze Testladung ermitteln

Ansgar66, 2023-05-14:
Hallo Community,
ich habe beobachtet, dass mein SMA-EV-Charger nach dem Setzen eines Ladewunsches ertstmal für kurze Zeit vollgas lädt.
Ich gehe davon aus, dass dies geschieht, um die maximal mögliche Ladeleistung zu testen und anschließend zur Kalkulation des spätesten Ladestarts zum Erreichen des gesetzten Ladeziels heranzuziehen. Wäre das nicht auch eine gute Vorgehensweise in EVCC?
Gruß,
Ansgar

andig, 2023-05-15:
Welchen Nutzen hätte das denn? Bei evcc kannst Du ja die Fahrzeuge schon entsprechend konfigurieren, es gibt also keinen Grund für eine Testladung?

andig, 2023-05-20:
Keine neue Erkenntnis, closing.

closed 2023-05-20 by andig

Ansgar66, 2023-05-21:
Oh! das ging ja schnell :-)
ich sehe doch noch einen kleinen Vorteil bei den Systemen, die oft von Gastfahrzeugen besucht werden. Es fehlte aber dann immer noch der Durchschnittsverbrauch dieser Fahrzeuge, um den Ladestart sinnvoll zu bestimmen...
Ein weiter Vorteil wäre, das am Ende der Strippe hängende Fahrzeug treffsicherer zu identifizieren...
Ein sonniges Wochenende,
Ansgar