evcc-io/evcc #8009: EVCC hängt sich sporadisch (teilweise) auf

Robbe64, 2023-05-15:
### Describe the bug

Aktuell habe ich das Problem, dass sich EVCC nach einigen Stunden bis Tagen ohne erkennbaren Grund aufhängt. Dabei ist die Weboberfläche noch erreichbar, es werden aber veraltete Werte angezeigt. Auch der Lademodus kann nicht gewechselt werden. Diese Requests bleibem im Browser (Chrome) auf "(pending)" stehen. Auffällig ist, dass im Log auch (fast) keine Einträge mehr geschrieben werden. Nur der Eintrag bezüglich der neuen Version ist erschienen. Der Container hat den Status "unhealthy".

### Steps to reproduce

Das hab ich leider noch nicht herausgerunden. Ich werde es aber weiter beobachten um zu sehen ob es eine Regelmäßigkeit gibt.
Beim letzten mal ist das Problem aufgetreten, nachdem ich das Fahrzeugt getrennt habe. Ich werde es wohl weiter beobachten müssen.

### Configuration details

```yaml
#uri: 0.0.0.0:7070 # uri for ui
plant: *****
interval: 20s # control cycle interval

log: debug
levels:
  id: trace
  #easee: trace
#  lp-1: trace
#  tibber: trace

sponsortoken: *****

meters:
  - type: template
    template: kostal-plenticore
    id: 71
    host: 192.168.178.101
    port: 1502
    usage: pv
    modbus: tcpip
    name: pv1
  - type: template
    template: kostal-plenticore
    id: 71
    host: 192.168.178.101
    port: 1502
    usage: battery
    modbus: tcpip
    name: battery2
    capacity: 5.2
  - type: template
    template: kostal-ksem-inverter
    id: 71
    host: 192.168.178.101
    port: 1502
    usage: grid
    modbus: tcpip
    name: grid3

#chargers:
#  - type: template
#    template: fritzdect
#    uri: https://192.168.178.1
#    user: *****
#    password: *****
#    ain: *****
#    standbypower: 15
#    name: fritzDect

chargers:
  - name: easee1
    type: easee
    user: *****
    password: *****
    charger: EHE84FNG

loadpoints:
  - title: Garage
    charger: easee1
    mode: pv
    mincurrent: 6
    maxcurrent: 32
    resetOnDisconnect: true
    soc:
      estimate: false # Interpolation des SoC zwischen den Abfragen zum Auto
    vehicle: egolf
    vehicles:
      - egolf
#      - tesla
    enable:
      delay: 3m
    disable:
      delay: 7m
      threshold: 100
    guardduration: 5m

site:
  title: Mein Zuhause
  meters:
    grid: grid3
    pvs:
      - pv1
    batteries:
      - battery2
  bufferSoC: 95 # Hausbatterie wird oberhalb SoC x% als Puffer genutzt
  prioritySoC: 20 # Hausbatterie bekommt bis zum SoC x% Priorität beim laden
  residualPower: 100 # 100W soll immer eingespeist werden. Nötig für Batteriesystem
  maxGridSupplyWhileBatteryCharging: 50 # Verhindert den Netzbezug bei DC Leistung > AC Leistung und gleichzeitigem Batterieladen

vehicles:
  - name: egolf
    type: vw
    title: eGolf # display name for UI
    capacity: 21
    user: *****
    password: *****
    phases: 1
    onIdentify:
      minCurrent: 7
      maxCurrent: 16
      targetSoC: 90
    #cloud: false
    identifiers: 04DA682ABD7280
#  - name: tesla
#    type: tesla
#    title: Model Y Jürgen # display name for UI
#    capacity: 77 # kWh
#    onIdentify:
#      minCurrent: 6
#      maxCurrent: 24
#    tokens:
#      access: *****
#      refresh: *****
#    #vin: *****
#    identifiers: 04DE682ABD7280

tariffs:
  currency: EUR
  grid:
    type: tibber
    token: *****
    #price: 0.31 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.077 # [currency]/kWh

influx:
  url: http://solaranzeige:8086
  database: evcc

telemetry: true
```

### Log details

```text
Das Log hab ich am 15.05. um ca. 08:35 gezogen:
...
[site  ] DEBUG 2023/05/14 19:27:02 ----
[lp-1  ] DEBUG 2023/05/14 19:27:02 charge power: 0W
[site  ] DEBUG 2023/05/14 19:27:02 pv power: 1235W
[site  ] DEBUG 2023/05/14 19:27:03 battery soc: 92%
[site  ] DEBUG 2023/05/14 19:27:03 battery power: -375W
[site  ] DEBUG 2023/05/14 19:27:03 grid power: -2W
[site  ] DEBUG 2023/05/14 19:27:03 site power: -277W
[lp-1  ] DEBUG 2023/05/14 19:27:03 charge currents: [0.01 0.008 0.008]A
[lp-1  ] DEBUG 2023/05/14 19:27:03 charge meter total import: 3952.929kWh
[lp-1  ] DEBUG 2023/05/14 19:27:03 charger status: B
[lp-1  ] DEBUG 2023/05/14 19:27:03 pv charge current: 1.2A = 0A + 1.2A (-277W @ 1p)
[site  ] DEBUG 2023/05/14 19:27:22 ----
[lp-1  ] DEBUG 2023/05/14 19:27:22 charge power: 0W
[site  ] DEBUG 2023/05/14 19:27:22 pv power: 1098W
[site  ] DEBUG 2023/05/14 19:27:23 battery soc: 92%
[site  ] DEBUG 2023/05/14 19:27:23 battery power: -352W
[site  ] DEBUG 2023/05/14 19:27:23 grid power: 380W
[site  ] DEBUG 2023/05/14 19:27:23 site power: 480W
[lp-1  ] DEBUG 2023/05/14 19:27:23 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/14 19:27:23 charge meter total import: 3952.929kWh
[lp-1  ] DEBUG 2023/05/14 19:27:23 charger status: A
[lp-1  ] INFO 2023/05/14 19:27:23 car disconnected
[lp-1  ] INFO 2023/05/14 19:27:23 vehicle updated: unknown -> eGolf
[lp-1  ] DEBUG 2023/05/14 19:27:23 set min current: 7
[lp-1  ] DEBUG 2023/05/14 19:27:23 set max current: 16
[lp-1  ] DEBUG 2023/05/14 19:27:23 set target soc: 90
[lp-1  ] DEBUG 2023/05/14 19:27:23 set charge mode: pv
[lp-1  ] DEBUG 2023/05/14 19:27:23 set min current: 6
[lp-1  ] DEBUG 2023/05/14 19:27:23 set max current: 32
[lp-1  ] DEBUG 2023/05/14 19:27:23 set min soc: 0
[lp-1  ] DEBUG 2023/05/14 19:27:23 set target soc: 100
[lp-1  ] DEBUG 2023/05/14 19:27:23 set min current: 7
[lp-1  ] DEBUG 2023/05/14 19:27:23 set max current: 16
[main  ] INFO 2023/05/15 00:19:03 new version available: 0.117.3
[main  ] INFO 2023/05/15 06:19:03 new version available: 0.117.3
```

### What type of operating system are you running?

Docker container

### Version

0.117.2

Robbe64, 2023-05-15:
Jetzt ist es sogar direkt nach dem Neustart aufgetreten:

[main  ] INFO 2023/05/15 08:45:23 evcc 0.117.3
[main  ] INFO 2023/05/15 08:45:23 using config file: /etc/evcc.yaml
[main  ] INFO 2023/05/15 08:45:23 starting ui and api at :7070
[db    ] INFO 2023/05/15 08:45:23 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2023/05/15 08:45:29 vehicles option is deprecated
[site  ] INFO 2023/05/15 08:45:29 site config:
[site  ] INFO 2023/05/15 08:45:29   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/05/15 08:45:29     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/15 08:45:29     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/15 08:45:29     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2023/05/15 08:45:29   vehicles:
[site  ] INFO 2023/05/15 08:45:29     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/05/15 08:45:29 loadpoint 1:
[lp-1  ] INFO 2023/05/15 08:45:29   mode:        pv
[lp-1  ] INFO 2023/05/15 08:45:29   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/05/15 08:45:29   meters:      charge ✓
[lp-1  ] INFO 2023/05/15 08:45:29     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/05/15 08:45:29 phase timer inactive
[lp-1  ] DEBUG 2023/05/15 08:45:29 pv timer inactive
[lp-1  ] DEBUG 2023/05/15 08:45:29 guard timer inactive
[lp-1  ] INFO 2023/05/15 08:45:29 vehicle updated: unknown -> eGolf
[lp-1  ] DEBUG 2023/05/15 08:45:29 set min current: 7
[lp-1  ] DEBUG 2023/05/15 08:45:29 set max current: 16
[lp-1  ] DEBUG 2023/05/15 08:45:29 set target soc: 90

Robbe64, 2023-05-16:
Wie mache ich dass denn im Docker Container? Habe als command 'evcc --profile' versucht aber das klappt nicht. Geht das prinzipiell beim EVCC Docker Container nicht oder mache ich was falsch?

andig, 2023-05-16:
Das müsste sein

    /app/evcc --profile (-c <config>)

Robbe64, 2023-05-16:
Es läuft jetzt mit --profile. Ich berichte wenn der Fehler weider aufgetreten ist.

Robbe64, 2023-06-02:
@andig: Es hängt jetzt wieder. Ich weiß bloß nicht genau wonach ich schauen muss.
Habe eine profile Datei mit /debug/pprof/profile?seconds=3 und eine heap datei mit /debug/pprof/heap gezogen.
Für letztere hab ich auch mit `go tool pprof -web` eine pprof.evcc.alloc_objects.alloc_space.inuse_objects.inuse_space.002.pb.gz erzeugt.
Kann ich die Dateien einfach hier hochladen zur Analyse oder soll ich nach etwas bestimmtem suchen?

andig, 2023-06-02:
Gibts noch eine Chance auf Portweiterleitung? Also noch etwas höngen lassen?

josef62, 2023-06-02:
Ich kann das leider bestätigen: auch bei mir "hängt" nach einigen Stunden/wenigen Tagen problemloser Funktion EVCC. Dabei lässt sich der Ladeschalter nicht mehr verändern, Menüaufruf funktioniert noch. Ebenfalls Docker (Synology) und die 0.117.2

Ich muss jetzt leider EVCC aus dem System nehmen, weil meine Regierung das nicht mehr haben will, sodass ich zur Fehlersuche  nicht beitragen kann.

Beste Grüße, Josef

Robbe64, 2023-06-02:
Ja, kann es noch eine weile hängen lassen. Portweiterleitung habe ich aktiviert. Habe dir die Adresse bei Slack geschickt.

Dies waren die letzten Einträge im Log:
```
[site  ] DEBUG 2023/06/01 08:52:15 ----
[lp-1  ] DEBUG 2023/06/01 08:52:15 charge power: 0W
[site  ] DEBUG 2023/06/01 08:52:17 pv power: 837W
[site  ] DEBUG 2023/06/01 08:52:17 battery soc: 83%
[site  ] DEBUG 2023/06/01 08:52:17 battery power: -511W
[site  ] DEBUG 2023/06/01 08:52:17 grid power: 2W
[site  ] DEBUG 2023/06/01 08:52:17 site power: -409W
[lp-1  ] DEBUG 2023/06/01 08:52:17 charge currents: [0.008 0.008 0.009]A
[lp-1  ] DEBUG 2023/06/01 08:52:17 charge meter total import: 4078.890kWh
[lp-1  ] DEBUG 2023/06/01 08:52:17 charger status: B
[lp-1  ] DEBUG 2023/06/01 08:52:17 targetSoc reached: 85.0% > 80%
[site  ] DEBUG 2023/06/01 08:52:35 ----
[lp-1  ] DEBUG 2023/06/01 08:52:35 charge power: 0W
[site  ] DEBUG 2023/06/01 08:52:37 pv power: 830W
[site  ] DEBUG 2023/06/01 08:52:37 battery soc: 83%
[site  ] DEBUG 2023/06/01 08:52:37 battery power: -512W
[site  ] DEBUG 2023/06/01 08:52:37 grid power: 7W
[site  ] DEBUG 2023/06/01 08:52:37 site power: -405W
[lp-1  ] DEBUG 2023/06/01 08:52:37 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/06/01 08:52:37 charge meter total import: 4078.890kWh
[lp-1  ] DEBUG 2023/06/01 08:52:37 charger status: A
[lp-1  ] INFO 2023/06/01 08:52:37 car disconnected
[lp-1  ] DEBUG 2023/06/01 08:52:37 set charge mode: pv
[lp-1  ] DEBUG 2023/06/01 08:52:37 set min current: 6
[lp-1  ] DEBUG 2023/06/01 08:52:37 set max current: 32
[lp-1  ] DEBUG 2023/06/01 08:52:37 set min soc: 0
[lp-1  ] DEBUG 2023/06/01 08:52:37 set target soc: 100
[lp-1  ] DEBUG 2023/06/01 08:52:37 set min current: 7
[lp-1  ] DEBUG 2023/06/01 08:52:37 set max current: 16
[lp-1  ] DEBUG 2023/06/01 08:52:37 set target soc: 90
[easee ] ERROR 2023/06/01 09:07:52 must retry
```

andig, 2023-06-02:
Habt ihr alle eine Easee? Im pprof sehe ich gerade dass der Hänger dort zu passieren scheint:

```
goroutine profile: total 2021
1976 @ 0x43f0b6 0x4504cf 0x4504a6 0x46fae6 0x47f685 0x47f4bb 0x178430a 0x4e268b 0x4e193c 0x11c297f 0x11c27be 0x4743e1
#	0x46fae5	sync.runtime_SemacquireMutex+0x25						/usr/local/go/src/runtime/sema.go:77
#	0x47f684	sync.(*Mutex).lockSlow+0x164							/usr/local/go/src/sync/mutex.go:171
#	0x47f4ba	sync.(*Mutex).Lock+0x3a								/usr/local/go/src/sync/mutex.go:90
#	0x1784309	github.com/evcc-io/evcc/charger.(*Easee).ProductUpdate+0x269			/build/charger/easee.go:270
#	0x4e268a	reflect.Value.call+0xb0a							/usr/local/go/src/reflect/value.go:586
#	0x4e193b	reflect.Value.Call+0xbb								/usr/local/go/src/reflect/value.go:370
#	0x11c297e	github.com/philippseith/signalr.(*loop).handleInvocationMessage.func1.1+0x13e	/go/pkg/mod/github.com/philippseith/signalr@v0.6.2/loop.go:205
#	0x11c27bd	github.com/philippseith/signalr.(*loop).handleInvocationMessage.func1+0x9d	/go/pkg/mod/github.com/philippseith/signalr@v0.6.2/loop.go:206
```

@josef62 ansonsten hilft nur ein Repro. Wenn Du's deaktivierst wird es keine Lösung geben (können).

Robbe64, 2023-06-02:
Ich hatte das Problem auf jeden Fall schon mit der 0.117.2 aber ich meine, dass Problem wäre auch in einer Vorversion schon vorhanden gewesen.
Habe nun erstmal wieder neu gestartet (mit aktuellem nightly). Wenn es wieder auftritt schaue ich nochmal unter /debug/pprof/goroutine?debug=1 nach.

nicx, 2023-06-03:
ich klinke mich mal mit ein: ich habe das problem seit ein/zwei/drei versionen leider auch. im grunde muss ich täglich evcc neu starten. ich nutze 2 easee wallboxen. evcc läuft als docker container auf einem unraid nas.
sobald ich aus dem urlaub wieder zu hause bin kann ich auch logs liefern ;)

andig, 2023-06-03:
Logs nutzen nix. Bitte mit —profile starten und pprof Output liefern.

Robbe64, 2023-06-03:
Hab mir die Version mit dem fix gezogen und teste mal ein paar Tage.

Robbe64, 2023-06-03:
Hat sich direkt beim Start aufgehängt.
[goroutine.txt](https://github.com/evcc-io/evcc/files/11642381/goroutine.txt)

andig, 2023-06-03:
In 15min gibts neues Nightly. Vielleicht (tm) hilfts.

andig, 2023-06-03:
Gibts jemanden, der Hänger hat, aber keine InfluxDB nutzt?

nicx, 2023-06-03:
ich nutze influxdb

josef62, 2023-06-03:
Ich nutze auch influx, das kann aber schnell mal abhängen. Bei mir ist eine Elli pro connect am werkeln. Gesendet von Outlook für Android

andig, 2023-06-04:
Könntet ihr ausprobieren, ob der PR für Euch funktioniert? Falls nicht möglich würde ich den mit ins Nightly nehmen. Hintergrund: Influx fragt den Loadpoint bzgl. Vehicle ab. Das führt zum Deadlock wenn der Loadpoint zu dem Zeitpunkt schon gesperrt ist. Die Sperren sind per PR granularer implementiert.

closed 2023-06-04 by andig

TheNaderio, 2024-04-14:
Ich habe seit einigen Tagen auch das Problem, dass sich, meistens nach ein paar Stunden, evcc „aufhängt“ und dann alte Werte erscheinen. Jetzt grad ist Nacht, aber ich habe angeblich noch 300W Solar Leistung 😅
Habe evcc per Docker auf nem Pi und eine Warp2 wallbox. Werte kommen über mqtt. 
Lade einen MG ZS EV, Habe das Gefühl es passiert wenn ich mich in der MG App einlogge. Bei MG kann es ja immer nur einen Login geben, bekomme dann relativ schnell die Meldung am Handy, dass ein anderes Gerät eine Verbindung aufgebaut hat.

andig, 2024-04-15:
Wenn du die Werte per Mqtt schickst dann aliegt das ja an den geschickten Werten, also nicht an evcc. Damit OT. Ich mache hier mal zu, das Problem gibts seit 1 Jahr nicht mehr!