evcc-io/evcc #8012: Zoned tariff: 'ct' instead of 'Rp.' for "Smart Grid Charging"

josefzahner, 2023-05-15:
### Describe the bug

Im WebGUI wird für "Smart Grid Charging" die Einheit "ct" statt "Rp." angezeigt, obwohl ich als Währung "CHF" ausgewählt habe. Sieht man auch schön am aktuellen Tariff. Schliesslich haben wir hier in der Schweiz keine Cents ;-)...
[image]

### Steps to reproduce

-

### Configuration details

```yaml
tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.3347
    zones:
      - days: Mo-Fr
        hours: 20:00-00:00,00:00-07:00
        price: 0.2396
      - days: Sa
        hours: 00:00-07:00
        price: 0.2396
      - days: Sa
        hours: 13:00-23:00
        price: 0.2396
      - days: So
        price: 0.2396

  feedin:
    type: fixed
    price: 0.1841
    zones:
      - days: Mo-Fr
        hours: 20:00-00:00,00:00-07:00
        price: 0.1436
      - days: Sa
        hours: 00:00-07:00
        price: 0.1436
      - days: Sa
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

andig, 2023-05-15:
/cc @naltatis @romerod

andig, 2023-05-18:
@josefzahner einfach `currency: CHF` beim Tarif konfigurieren. Ich geb zu, dass das nicht offensichtlich ist...

josefzahner, 2023-05-18:
@andig wo genau soll ich den CHF eintragen? Im tariff habe ich die „currency: CHF“ bereits drin, siehst du oben in der config als auch im screenshot wo korrekt CHF drin steht. Trotzdem steht ct im GUI (siehe screenshot).

andig, 2023-05-18:
Bei den Tarifen selbst auch nochmal!

andig, 2023-05-19:
Werden wir, das ist erstmal der Workaround!

andig, 2023-05-20:
Replaced by https://github.com/evcc-io/evcc/issues/8087

closed 2023-05-20 by andig