evcc-io/evcc #8013: Elli Pro falsche Leistungsanzeige

Marty56, 2023-05-15:
### Describe the bug

Bei der Elli Pro Wallbox mit integriertem Zähler (in meinem Fall der ABB Nachrüstzähler), wird oft noch eine sehr hohe, falsche Leistungsanzeige angezeigt, nachdem der Ladevorgang unterbrochen wurde (siehe Bild).
Der korrekte Wert ist 0 W.

Ich habe noch einen weiteren Zähler, der  im Stromkasten sitzt und den Strom in die Wallbox misst, deshalb bin ich sicher, dass dieser Strom nicht mehr fließt, sondern nur noch der Standby Strom der Wallbox von ca. 4 W.  Ich vermute, dass evcc direkt nach dem Ausschalten des Ladestroms etwas zu früh aufhört, den integrierten Zähler abzufragen und der deshalb noch nicht ganz den korrekten Endwert erreicht hat. Der angezeigte Leistungswert schwankt `auch.` Manchmal wird 0 W, manchmal 9 W und im Beispiel sogar 23 W anzeigt.

Der Bug ist ein Minor Issue und vermutlich eher kosmetischer Natur.

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
  - name: wallbox
    type: template
    template: ellipro
    ski: B456 C8F7 5294 1D4D A933 3D84 438F 5B60 848D D805
    ip: 192.168.10.94
```
```

### Log details

```text
log zeigt keine Unauffälligkeiten
```

### What type of operating system are you running?

Linux

### Version

V0.117

DerAndereAndi, 2023-05-15:
Es wird der letzte von der Wallbox übermittelte Werte genommen. Sobald kein EV mehr angeschlossen ist, werden auch keine Messwerte von der Wallbox mehr übermittelt. Es kann auch kein aktualisierter Wert explizit angefordert werden.

Es gibt hier nichts das man tun könnte.

Marty56, 2023-05-15:
Ok. Da kann man wohl nichts machen.

War ja auch nur ein Thema, was nicht so wichtig ist.  Ansonsten bin ich sehr glücklich mit der V.117. Die Übertragungsfehler mit dem Charger Meter sind verschwunden. Heute habe ich 5 Stunden geladen und keinen Fehler bekommen, obwohl ich WLAN benutze. Super Arbeit!

closed 2023-05-15 by andig