evcc-io/evcc #8015: evcc warning: bufferStartSoc must be larger than bufferSoc

emcepe, 2023-05-15:
### Describe the bug

Meldung im Log:

evcc warning: bufferStartSoc must be larger than bufferSoc

sowohl bei:

```
  # continue charging on battery above soc (0 to disable)
  bufferSoc: 95
  # start charging on battery above soc (0 to disable)
  bufferStartSoc: 99
```

als auch bei:

```
  # continue charging on battery above soc (0 to disable)
  bufferSoc: 95
  # start charging on battery above soc (0 to disable)
  bufferStartSoc: 0
```

wenn ich richtig rechnen kann, dann ist 99 > 95

und 0 sollte disabled sein

IMHO in beiden Fällen dann eine falsche Warnung.

### Steps to reproduce

start evcc

### Configuration details

```yaml
...
# site describes the EVU connection, PV and home battery
site:
  title: Mein Zuhause
  meters:
    grid: grid
    pvs:
    - pv
    batteries:
    - battery
  # additional household usage margin
  residualPower: 100
  # give home battery priority up to this soc (empty to disable)
  #prioritySoc: 90
  prioritySoc: 5
  # continue charging on battery above soc (0 to disable)
  bufferSoc: 95
  # start charging on battery above soc (0 to disable)
  bufferStartSoc: 0
  # ignore battery charging if AC consumption is above this value
  maxGridSupplyWhileBatteryCharging: 0
  # set cost limit for automatic charging in PV mode
  smartCostLimit: 0
...
```
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.117.3

andig, 2023-05-15:
Der erste Fall liegt vmtl daran, dass Du im Gui noch was Anderes eingestellt und gespeichert hattest- dann zieht die Config nicht mehr.

closed 2023-05-15 by andig

emcepe, 2023-05-17:
Stimmt. Habe ich aus der evcc.db gelöscht, danach war die eine Warnung weg.