evcc-io/evcc #8018: Nach Stromausfall lädt Auto mit volllast

horst789github, 2023-05-15:
### Describe the bug

Ausgangssituation:
3x LoadPoints an dem 1 Tesla hängt (Lp1 "easeeRechts") der schon sein ziel soc von 85% erreicht hat und deswegen auch nicht lädt. PV Anlage speißt ein weil auch Hausakku voll ist. 

Dann: Stromausfall:
ich hatte gerade einen ungeplanten Stromausfall, meine Wechselrichter hat nach ~30 sekunden in den Inselbetrieb gewechselt. Soweit alles ok, es wurde weiterhin kein Auto geladen. Die NAS auf dem EVCC läuft, war jederzeit an (wegen USV). Nur die Wallbox hat für die Umschaltung jeweils kein Strom gehabt. Nach 5min war Strom wieder da, also hat der WechselRichter vom Inselbetrieb auf Netzbetrieb umgeschwenkt, was wieder einen kurzen Stromausfall von ~10 sekunden zur Folge hatte.

Danach habe ich eine große Last gesehen und mich gewundert was hier mit 11 kW lädt. Kurze prüfung von EVCC, war es LP1 der den Tesla lädt obwohl ja ziel SoC von 85% schon längst erreicht wurde. Auch die anzeige im EVCC hatte das erkannt (die daten aus Tesla API wurden korrekt angezeigt und übertragen).

Es sieht so aus als würde nach einem Stromausfall hier irgendwo das Signal herkommen, dass er "schnell" laden soll, obwohl der Modus weiterhin auf PV stand.

### Steps to reproduce

nach Strom ausfall

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: xxxx

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: error
  lp-1: error
  lp-2: error
  lp-3: error
  cache: error
  db: error
# log settings beschreibung
#     fatal: Nur Nachrichten der Kategorie fatal werden angezeigt. Das sind Fehler, die die Systemausführung verhindern.
#    error: Nur Nachrichten der Kategorie error werden angezeigt. Das sind die wenigsten Nachrichten.
#    warn: Beinhaltet error, zusätzlich werden Nachrichten der Kategorie warn angezeigt.
#    info: Beinhaltet warn, zusätzlich werden Nachrichten der Kategorie info angezeigt.
#    debug: Beinhaltet info, zusätzlich werden Nachrichten der Kategorie debug angezeigt. Dies ist für die Fehleranalyse erforderlich.
#    trace: Beinhaltet debug, zusätzlich werden Nachrichten der Kategorie trace angezeigt. Dies ist die detaillierteste Kategorie und kann zu sehr großen Logdaten führen. In der Regel wird dies nicht benötigt!
[... 225 lines cut ...]
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
[lp-3  ] ERROR 2023/05/15 16:08:27 charge meter: Get "http://192.168.0.157/status": dial tcp 192.168.0.157:80: i/o timeout
[lp-3  ] ERROR 2023/05/15 16:08:42 charge meter: Get "http://192.168.0.157/status": dial tcp 192.168.0.157:80: i/o timeout
[lp-3  ] ERROR 2023/05/15 16:08:53 charge meter: Get "http://192.168.0.157/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[lp-3  ] ERROR 2023/05/15 16:11:02 charge meter: Get "http://192.168.0.157/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[main  ] INFO 2023/05/15 16:16:03 new version available: 0.117.3
```

### What type of operating system are you running?

Docker container

### Version

0.117.2

horst789github, 2023-05-15:
[image]

mucki12, 2023-05-15:
Umfasst deine USV denn die gesamte Netzwerkkette?

horst789github, 2023-05-15:
> Umfasst deine USV denn die gesamte Netzwerkkette?

internet ist dann kurz weg, ansonsten funktioniert das netzwerk durch die usv. 
warum fragst du?

andig, 2023-05-15:
Ohne Logfile lässt sich da wenig sagen. Für den Ausnahmefall "strom weg" würde ich allerdings aktuell auch keinen Aufwand investieren.

closed 2023-05-15 by andig

Hofyyy, 2023-05-15:
Ich kann was sehr ähnliches reproduzieren. Wenn ich die Wallbox per Sicherung kurz vom netz nehme, muss ich danach den evcc docker neustarten. sonst sind die speed limits per evcc nicht richtig gesetzt.

mein usecase war eine 22kw wallbox und ich habe maxcurrent auf 16A gestellt.

Die frage ist ob der usecase überhaupt abgedeckt ist und zu welchen Zeitpunkten diese werte per modbus tcp gesetzt werden.

Also mit evcc neustart läd er mit 16A bei sofort. 

Ohne evcc neustart läd er mit 20A (mehr kann das auto nicht)

andig, 2023-05-15:
Also enabled/disabled wird geprüft. Wenn sich am Strom regelungstechnisch nichts ändert wird der aber nicht neu gesetzt- sonst müssten wir das ja quasi bei jedem Durchgang machen.

Hofyyy, 2023-05-16:
Also ich verstehe dich sondas dies nicht vorgesehen ist. Oder könnte man ohne nachteile in jedem zyklus den speed auslesen und bei abweichung einmal neu durch konfigurieren. So wie out of sync detected, force reconfigure ...

andig, 2023-05-18:
"speed auslesen" ist keine Schnittstelle die wir implementiert haben...