evcc-io/evcc #8022: Bug in "add guard disable timer to status ui  (#7959)

VolkerK62, 2023-05-15:
ref #7959

Da scheint ein bug drin zu sein. 
Es wird nach dem disable-timer weiterhin "Ladevorgang aktiv" angezeigt.
Aber im Log steht jetzt: `guard enable in xx Min`, obwohl geladen wird (deshalb auch keine Anzeige im UI)
```
Mai 14 07:17:41 raspberrypi evcc[25720]: [lp-3  ] DEBUG 2023/05/14 07:17:41 site power 402W >= 0W disable threshold
Mai 14 07:17:41 raspberrypi evcc[25720]: [lp-3  ] DEBUG 2023/05/14 07:17:41 pv disable in 0s
Mai 14 07:17:41 raspberrypi evcc[25720]: [lp-3  ] DEBUG 2023/05/14 07:17:41 pv disable timer elapsed
Mai 14 07:17:41 raspberrypi evcc[25720]: [lp-3  ] DEBUG 2023/05/14 07:17:41 guard enable in 16m3s
Mai 14 07:17:51 raspberrypi evcc[25720]: [site  ] DEBUG 2023/05/14 07:17:51 ----
```

VolkerK62, 2023-05-16:
Es ist LP3. Also im api/state die Nr.2.

```
{"result":{"auth":{"vehicles":{}},"availableVersion":"0.117.3","battery":[{"power":442.5818181818181,"soc":57,"capacity":7}],"batteryCapacity":7,"batteryConfigured":true,"batteryPower":442.5818181818181,"batterySoc":57,"bufferSoc":100,"bufferStartSoc":0,"currency":"EUR","greenShare":0.9365885944872644,"gridConfigured":true,"gridCurrents":[-2.12,2.09,0.93],"gridPower":54,"homePower":312.3839015151515,"loadpoints":[{"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":0,"chargePower":0,"chargeTotalImport":455,"chargeVoltages":[230.4,0,0],"chargedEnergy":0,"chargerFeatureIntegratedDevice":false,"chargerIcon":null,"charging":false,"connected":false,"connectedDuration":9223372036854775807,"enabled":false,"guardAction":"inactive","guardRemaining":0,"maxCurrent":20,"minCurrent":6,"minSoc":0,"mode":"off","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planActive":false,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"targetSoc":100,"targetTime":"0001-01-01T00:00:00Z","title":"Carport","vehicleDetectionActive":false},{"chargeCurrent":0,"chargeDuration":0,"chargePower":0,"chargeRemainingEnergy":694.4444444444445,"chargeTotalImport":1.9628166666666667,"chargedEnergy":0,"chargerFeatureIntegratedDevice":false,"chargerIcon":"","charging":false,"connected":true,"connectedDuration":0,"enabled":false,"guardAction":"inactive","guardRemaining":0,"maxCurrent":1,"minCurrent":1,"minSoc":0,"mode":"off","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planActive":false,"pvAction":"inactive","pvRemaining":0,"sessionCo2PerKWh":null,"sessionEnergy":0,"sessionPrice":null,"sessionPricePerKWh":null,"sessionSolarPercentage":0,"targetEnergy":0,"targetSoc":100,"title":"Garage","vehicleCapacity":0.625,"vehicleDetectionActive":false,"vehicleFeatureOffline":true,"vehicleIcon":"bike","vehiclePresent":true,"vehicleRange":0,"vehicleSoc":0,"vehicleTargetSoc":0,"vehicleTitle":"E-Bikes"},{"chargeCurrent":3,"chargeDuration":28160000000000,"chargePower":539.1979166666666,"chargeRemainingDuration":16718000000000,"chargeRemainingEnergy":2500,"chargeTotalImport":212.4967,"chargedEnergy":4070.4000000000633,"chargerFeatureIntegratedDevice":false,"chargerIcon":"","charging":true,"connected":true,"connectedDuration":0,"enabled":true,"guardAction":"enable","guardRemaining":1448128078241,"maxCurrent":3,"minCurrent":3,"minSoc":0,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":1,"planActive":false,"pvAction":"disable","pvRemaining":0,"sessionCo2PerKWh":5.820351723903223,"sessionEnergy":4070.4000000000633,"sessionPrice":0.45841630084155527,"sessionPricePerKWh":0.1126219292554904,"sessionSolarPercentage":99.00733568982822,"targetEnergy":0,"targetSoc":65,"title":"Warmwasser-WP","vehicleCapacity":15,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleIcon":"waterheater","vehiclePresent":true,"vehicleRange":0,"vehicleSoc":50,"vehicleTargetSoc":0,"vehicleTitle":"Warmwasser"}],"prioritySoc":100,"pv":[{"power":355}],"pvConfigured":true,"pvPower":355,"residualPower":100,"savingsAmount":144.26049277817197,"savingsEffectivePrice":0.15691894737341736,"savingsGridCharged":114.14947950145609,"savingsSelfConsumptionCharged":463.02431423095777,"savingsSelfConsumptionPercent":80.22268496230835,"savingsSince":"2022-12-13T13:35:18+01:00","savingsTotalCharged":577.1737937324139,"siteTitle":"Haus","smartCostAvailable":true,"smartCostLimit":0,"smartCostUnit":"gCO2eq","sponsor":"VolkerK62","sponsorTokenExpires":0,"tariffCo2":793,"tariffEffectiveCo2":50.285244571599335,"tariffEffectivePrice":0.12944288703375534,"tariffFeedIn":0.1095,"tariffGrid":0.424,"vehicles":["Warmwasser","(T)Raumschiff","E-Bikes","ohne Bluelink"],"version":"0.117.2 (bd3b28b3)"}}
```

VolkerK62, 2023-05-16:
Zur Klarstellung: Die Funktionalität ist richtig. Guard Timer läuft runter und dann wird Charger disabled.
Es wird mMn nur fälschlicherweise `guard enable` statt `guard disbale` angezeigt.

```
Mai 16 07:44:41 raspberrypi evcc[25720]: [lp-3  ] DEBUG 2023/05/16 07:44:41 guard enable in 8s

Mai 16 07:45:11 raspberrypi evcc[25720]: [lp-3  ] DEBUG 2023/05/16 07:45:11 charger: guard elapse
Mai 16 07:45:11 raspberrypi evcc[25720]: [lp-3  ] DEBUG 2023/05/16 07:45:11 guard timer inactive
Mai 16 07:45:12 raspberrypi evcc[25720]: [lp-3  ] DEBUG 2023/05/16 07:45:12 charger disable
```

naltatis, 2023-05-17:
Ah ok. Ich hab mir die Implementierung noch in der Regelung noch mal angeschaut. Im Gegensatz zu pv und phase unterschiedet der guard Timer aktuell nicht zwischen an und ausschalten. Heißt wir könnten allgemein anzeigen, dass gerade ein Guard aktiv ist, wissen aber erstmal nicht was nach dem Guard passiert (anschalten oder ausschalten). `guardAction: disabled` gibt es momentan gar nicht.

Ich sehe hier drei Wege:
1. Guard Disable explizit in der Regellogik modellieren und ans API geben
2. Generischere Meldung in der UI: "Nächste Aktion in 47s..."
3. enable/disable Guard Text abhängig vom aktuellen Loadpoint Status (enabled) bestimmen

Der erste Weg ist vermutlich der größte Aufwand aber auch die sauberste Lösung. Dann hätten wir phase/guard/pv Timer ale symmetrisch. Ich weiß aber auch, dass die Regellogik nicht trivial ist und @andig parallel überlegt sie zu vereinfachen.
Der letzte Vorschlag würde das gewünschte Verhalten mit wenig Aufwand bringen ist aber hacky und wir würden dann eine Sonderregel für den guard Timer machen. Davon wäre ich kein großer Fan.

Mein aktueller Favorit ist 2. Die generische Meldung ist nicht schön, ich mag wenns möglichst konkret ist. Ich sehe den Guard Timer aber auch eher als zusätzlichen Schutzmechanismus der idR. nur selten angezeigt wird. Phase und PV Timer sollten normalerweise größer sein sodass sie guard überlagern.

andig, 2023-05-17:
2 oder 3 klingt gut, 1 geht nicht

VolkerK62, 2023-05-17:
Ich seh schon, viele Dinge stelle ich mir immer zu einfach vor ;-)
Lösung 2 wäre für mich auch ok. Dann sieht man wenigstens das im Hintergrund etwas passiert und wundert sich nicht, warum nicht abgeschaltet wird.

naltatis, 2023-05-17:
Guard Disable habe ich wieder reverted. Das hatte ja keinen wirklichen Effekt.

Vorschlag für generischen Text:
```
Switching in 57s...
Umschalten in 57s...
```

Das ist relativ kompakt, passt du den bestehenden Formulierungen (pausing, starting) und beschreibt was nach dem Zeitraum passiert.

VolkerK62, 2023-05-17:
Evtl noch ein "Guard aktiv." davor? Dann ist es klarer. Bei "Umschlaten" könnte man an Phasenumschaltung denken.

`Guard aktiv. Umschalten in 57s....`

premultiply, 2023-05-17:
Bei dem GuardTimer ist es doch eher so, dass sobald und solange der läuft der aktuelle Zustand an/aus beibehalten wird. Phasenumschaltung zählt dabei als an/aus.

naltatis, 2023-05-17:
> Evtl noch ein "Guard aktiv." davor? 

Hab ich angepasst. Ich find das etwas technisch. Gerade weil viele Nutzer nicht wissen werden das das bedeutet. Da der im Default-Fall aber nicht zu sehen sein wird find ich das ok. Wenn man den selbst hochgedreht hat macht das ja Sinn.

andig, 2023-05-17:
Schütz gesperrt statt Guard?

premultiply, 2023-05-17:
Schaltschutz aktiv für x:xx ?

naltatis, 2023-05-17:
Wollen wir dann im Englischen auch gleich "Switching protection" machen? Dass das technisch bei uns `guard` heißt ist ja (zumindest perspektivisch) nicht so wichtig.

naltatis, 2023-05-17:
```
Switching protection active for 33s…
Schaltschutz aktiv für 33s…
```

naltatis, 2023-05-17:
Zu viel reverted. Der generische Guard Text muss angezeigt werden auch wenn noch eine Ladevorgang läuft.

PR #8042 (2023-05-17): Generic guard message
fixes #8022
PR #8100 (2023-05-20): Show guard disable timer
fix #8022
closed 2023-05-20 by naltatis