evcc-io/evcc #8036: Add M-TEC Energy Butler

premultiply, 2023-05-16:
Bin stolzer Besitzer eienr PV-Anlage von M-TEC. Das Produkt heißt Energy Butler. Das ist ein Energiespeicher mit integriertem Hybridwechselrichter in den Größen 6, 8, 10 12, 15 und 20 kW und stapelbaren Modulen bis zu 30,7 kWh.: https://www.mtec-systems.com/batteriespeicher/energy-butler-11-bis-30-kwh/

[20221125_MTEC-EnergyButler_Modbus_RTU_Protkoll.pdf](https://forum.iobroker.net/assets/uploads/files/1681811699113-20221125_mtec-energybutler_modbus_rtu_protkoll.pdf)
Wäre schön, wenn es hier Mitstreiter für eine Anbindung gäbe bzw. jemanden der mir bei der Anbindung behilflich sein kann. JA, ich habe keine Ahnung!!

_Originally posted by @1979ck in https://github.com/evcc-io/evcc/discussions/8034_

itmtechnologies, 2023-05-19:
den habe ich auch du kannst ihn über modbus in den homeassistant und dann in evcc einbinden

modbus esp32 hast du schon angeschlossen?

die modbus kommunikation geht plug ans play nicht!

1979ck, 2023-05-20:
> den habe ich auch du kannst ihn über modbus in den homeassistant und dann in evcc einbinden
> 
> 
> 
> modbus esp32 hast du schon angeschlossen?
> 
> 
> 
> die modbus kommunikation geht plug ans play nicht!

Homeassistant habe ich noch nicht laufen. Auch einen esp32 habe ich nicht am laufen. Daher die Anfrage zur direkten Anbindung über Modbus an EVCC. Wenn das nicht geht dann muss ich erstmal mit Homeassistant anfangen.

sirwastl, 2023-06-13:
Hi, seid ihr weiter gekommen? Mich würde auch eine (direkte) Anbindung über Homeassistant interessieren.

itmtechnologies, 2023-06-13:
ja ich das fix fertig aber du brauchst nen esp32 oder 8266 und nen max485

Message ID: ***@***.***>

sirwastl, 2023-06-17:
> ja ich das fix fertig aber du brauchst nen esp32 oder 8266 und nen max485 Message ID: ***@***.***>

Cool. Wo hast Du den dann angeschlossen. HA hab ich gut am Laufen. Mit ESP bin ich noch nicht so tief drin. Daher würde ich mich über mehr Infos freuen .-)

itmtechnologies, 2023-06-17:
kannst du mir deine handy nummer schicken dann schick ich dir mal fotos schick mir die nummer am besten per mail an ***@***.***

Message ID: ***@***.***>

sirwastl, 2023-06-18:
Hi, ja die email wird leider nicht angezeigt (bzw. ist mit * überschrieben), oder es liegt an mir, dass ich sie nicht sehe...