evcc-io/evcc #8040: Audi/Skoda/VW vehicles - pull status of doors, lights etc. through evcc

stevoboy, 2023-05-16:
The VAG API limits the number of accesses. So when using evcc, Iobroker VAG adapter and apps to access the vehicle status, the user account is locked.

If possible, it would be great if evcc could pull also data on door and window status, and provide the controls for charging and climate control.

This way the evcc adapter of Iobroker could access these functions via evcc and thereby only evcc and VAG apps are using the API. Lockouts can be avoided.

Sillium007, 2023-05-17:
Würde das nicht eher andersrum Sinn machen? Der IOBroker zieht alles und evcc holt sich den Ladestand dort.... für viel mehr muss sich evcc ja eigentlich nicht interessieren, daher denke ich der Aufwand alles zu holen und bereitzustellen was gar nichts mit Laden zu tun hat macht wenig Sinn.

stevoboy, 2023-05-17:
Auf jeden Fall würde das auch Sinn machen, richtig.

Allerdings gibt es ja bereits eine funktionierende Verbindung VAG-EVCC und EVCC-Iobroker.
Da wäre es doch am einfachsten diese Verbindungen um die fehlenden Datensegmente zu erweitern.

Sillium007, 2023-05-17:
Das Problem ist weniger die Machbarkeit, sondern dass es wenig Sinn für EVCC macht sich diese Daten zu holen und bereitzustellen und dann auch "nur" für eine Fahrzeugmarke. Das ist Implementierungs- und Pflegeaufwand für etwas das nichts mit dem eigentlichen Scope von evcc zu tun hat.

Aber ich entscheide das ja nicht :)

stevoboy, 2023-05-17:
Ist die alternativlösung denn besser? Man müsste für den evcc ja entweder einen Adapter zum Iobroker bauen oder alternativ jeden Datenpunkt per http Request einlesen. 

Das wäre immerhin für den kompletten MEB Baukasten verwendbar und andere Fahrzeugmodelle haben ja auch mehr datenpunkte als den soc.
Grundsätzlich muss man sich halt überlegen wie der evcc in der gesamtarchitektur platziert sein soll. Die APIs der Autohersteller sind nun mal limitiert - da kannste nicht 10 leecher dranhängen :-(.

KrysosDE, 2023-05-17:
Also ich habe keine Erfahrung mit ioBroker oder VAG, aber ich lasse alle Daten (Inverter, Fahrzeuge, Zähler) von Home Assistant auslesen und schicke diese dann mit MQTT an EVCC weiter. Das funktioniert ohne Probleme und bietet mir auch einige Vorteile. Eine Home Assistant Integration für VAG Fahrzeuge gibt's ja, sollte also auch ohne großen Aufwand möglich sein.

Das in EVCC einzubauen macht meines Erachtens wenig Sinn, die Zielgruppe die das wirklich brauchen kann ist zu klein und steht somit nicht wirklich im Verhältnis zum Aufwand.