evcc-io/evcc #8052: Automated grid charging plan

rene-mueller, 2023-05-18:
**Is your feature request related to a problem?**
I have a balcony power plant and a Tibber contract. In Evcc there is currently the possibility to charge according to PV availability and price. The parameter from when to charge by price must be set by threshold.

**Describe the solution you'd like**

It would be better to define the loading times automatically on the basis of the forecast. Technically, this could work as follows:

* find the cheapest price
* search the prices around (-1 and +1; -2 and +2; etc.) in a loop until the price is 3ct higher.

Thus the radius increases and the price level is defined automatically.

andig, 2023-05-18:
/cc @VolkerK62 haben wir Ladeplanung eigentlich jenseits der Settings unter `planner` erklärt?

andig, 2023-05-18:
@rene-mueller ich denke das ist die Ladeplanung, gibts also schon.

closed 2023-05-18 by andig

rene-mueller, 2023-05-18:
@andig Die ist aber nicht implementiert...

andig, 2023-05-18:
Steile These ;). Schau mal https://github.com/evcc-io/evcc/discussions/5639

rene-mueller, 2023-05-18:
@andig stimmt, habe mir nur die Stable angeschaut

andig, 2023-05-18:
Auch dort ist es seit vielen Releases drin.