evcc-io/evcc #8071: Weiterbetrieb trotz fehlender Komponenten

itmtechnologies, 2023-05-19:


itmtechnologies, 2023-05-19:
Ich fände es glaube ich besser wenn evcc nicht den Dienst komplett quittieren würde wenn eine Komponente wie zum Beispiel mein Kostal CI 30 weg ist.

Das passiert so gesehen jede Nacht weil der Kostal ohne DC Spannung auch nicht mehr kommuniziert.

Was dabei dann die Folge ist: ein Auto was noch nicht fertig geladen ist das fängt dann ohne die kontrolle von evcc mitten in der nacht an mit 11 kw zu laden.

Eine Warnung innerhalb der Anwendung wäre glaube ich sinniger als das es den Start verhindert.

andig, 2023-05-19:
> Ich fände es glaube ich besser wenn evcc nicht den Dienst komplett quittieren würde 

Gerne erstmal ein Logfile um zu sehen was passiert. Eigentlich (tm) sollte evcc weiter laufen.

closed 2023-05-19 by andig

itmtechnologies, 2023-05-19:
hallo andi schön das du dich gleich meldest

erstmal herzlichen dank für die tolle arbeit ich finde das wahnsinn was du da in der kurzen zeit auf die beine gestellt hast.

ich glaube das log erübrigt sich ich kann dir genau sagen wie es dazu kommt

ich habe evcc als homeassistant addon laufen

dort passiert es ab und an dass das addon neu startet das liegt aber nicht direkt an evcc

kann auch durch ein neustart durch HA kommen und das dadurch der evcc neu startet.

die problematik wäre meiner meinung nach nur lösbar in dem man grundsätzlich den start durch eine „falsche“ konfiguration nicht verhindert 

evcc bemängelt sobald der Wechselrichter aus ist die falsche konfiguration und bricht den start ab. dadurch sind die Wallboxen dann führungslos und laden dann eben wenn ein auto dran ist

im konkreten ist das die technisat wallbox die das tut

andig, 2023-05-19:
> evcc bemängelt sobald der Wechselrichter aus ist die falsche konfiguration und bricht den start ab. dadurch sind die Wallboxen dann führungslos und laden dann eben wenn ein auto dran ist

Ja, das ist folgerichtig. Wir könnten noch überlegen, unkritische Komponenten zu ignorieren. Falls Du grid und pv konfiguriert hast hat PV nur optische Funktion. Ich bin aber sicher dann kommt der nächste und möchte die Meßwerte der pv haben...

Ich fürchte da gibt es keine gute Lösung? /cc @premultiply