evcc-io/evcc #8072: Fehler "persist: attempt to write a readonly database (8)"

m-milinovsky, 2023-05-19:
### Describe the bug

Nach Beenden eines Ladevorganges erscheint folgende Fehlermeldung im GUI und im Log:
```
May 19 10:33:11 openhab evcc[1704617]: [main  ] ERROR 2023/05/19 10:33:11 cannot save settings: attempt to write a readonly database (8)
May 19 10:33:11 openhab evcc[1704617]: [db    ] ERROR 2023/05/19 10:33:11 persist: attempt to write a readonly database (8)
```

Die Rechte auf filesystemebene sind:
`-rw-r--r--  1 root root 28672 18. Mai 19:18 evcc.db`
evcc läuft als Dienst.

### Steps to reproduce

1. evcc starten
2. Ladung starten
3. Ladung beenden

### Configuration details

```yaml
[main  ] INFO 2023/05/19 10:34:48 evcc 0.117.4
[main  ] INFO 2023/05/19 10:34:48 using config file: /etc/evcc.yaml
[db    ] INFO 2023/05/19 10:34:49 using sqlite database: /home/martin/.evcc/evcc.db

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: debug
  lp-1: debug
  fiat: debug
  cache: debug
  charger: debug
  meter: debug
  vehicle: debug

# unique installation id
plant: *****

interval: 10s # control cycle interval
#interval: 30s # control cycle interval

meters:
  - name: pv1
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: 192.168.178.8
  - name: my_grid
    type: template
    template: fronius-solarapi-v1
    usage: grid
    host: 192.168.178.8

chargers:
  - type: template
    template: fronius-wattpilot
    host: 192.168.178.12
    password: *****
    name: wallbox3

vehicles:
  - name: fiat500e
    type: template
    template: fiat
    title: Fiat 500e
    user: *****
    password: *****
    capacity: 42
    mode: pv
    targetSoc: 90

tariffs:
  currency: EUR # (default EUR)
  grid:
#    type: awattar
#    region: at # optional, choose at for Austria
    type: fixed
    price: 0.44
  feedin:
    type: fixed
    price: 0.1789

loadpoints:
  - title: Wallbox links
    charger: wallbox3
    mode: pv
    phases: 0
    mincurrent: 6
    maxcurrent: 16
    resetOnDisconnect: true
    vehicle: fiat500e # Referenz auf Standardfahrzeug

site:
  - title: MiliHome
    meters:
      grid: my_grid
      pv:
      - pv1
# push messages
messaging:
  events:
#    start: # charge start event
#      title: Charge started
#      msg: Started charging ${vehicleTitle} in "${mode}" mode
#    stop: # charge stop event
#      title: Charge finished
#      msg: Finished charging ${vehicleTitle} ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car ${vehicleTitle} connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car ${vehicleTitle} disconnected
      msg: Car disconnected after ${connectedDuration}
#    soc: # vehicle soc update event
#      title: Soc updated
#      msg: Battery of ${vehicleTitle} charged to ${vehicleSoc:%.0f}%
#    guest: # vehicle could not be identified
#      title: Unknown vehicle
#      msg: Unknown vehicle, guest connected?
  services:
   - type: ntfy
     uri: https://ntfy.sh/mili_load_info
     priority: low
     tags: fiat500

sponsortoken: *****

Version: `0.117.4`
```

### Log details

```text
May 19 10:32:56 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:32:56 charge power: 3760W
May 19 10:32:56 openhab evcc[1704617]: [site  ] DEBUG 2023/05/19 10:32:56 pv power: 12503W
May 19 10:32:56 openhab evcc[1704617]: [site  ] DEBUG 2023/05/19 10:32:56 grid power: -6612W
May 19 10:32:56 openhab evcc[1704617]: [site  ] DEBUG 2023/05/19 10:32:56 site power: -6612W
May 19 10:32:56 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:32:56 charge currents: [15.6 0 0]A
May 19 10:32:56 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:32:56 detected active phases: 1p
May 19 10:32:56 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:32:56 charger status: C
May 19 10:32:56 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:32:56 soc estimated: 14.13% (vehicle: 14.00%)
May 19 10:32:56 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:32:56 vehicle soc: 14%
May 19 10:32:56 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:32:56 vehicle range: 26km
May 19 10:32:56 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:32:56 pv charge current: 44.7A = 16A + 28.7A (-6612W @ 1p)
May 19 10:32:56 openhab evcc[1704617]: [site  ] DEBUG 2023/05/19 10:32:56 telemetry: charge: Δ42/42Wh @ 3760W
May 19 10:33:06 openhab evcc[1704617]: [site  ] DEBUG 2023/05/19 10:33:06 ----
May 19 10:33:06 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:33:06 charge power: 3770W
May 19 10:33:06 openhab evcc[1704617]: [site  ] DEBUG 2023/05/19 10:33:06 pv power: 12615W
May 19 10:33:06 openhab evcc[1704617]: [site  ] DEBUG 2023/05/19 10:33:06 grid power: -8187W
May 19 10:33:06 openhab evcc[1704617]: [site  ] DEBUG 2023/05/19 10:33:06 site power: -8187W
May 19 10:33:06 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:33:06 charge currents: [15.7 0 0]A
May 19 10:33:06 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:33:06 detected active phases: 1p
May 19 10:33:06 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:33:06 charger status: C
May 19 10:33:06 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:33:06 soc estimated: 14.15% (vehicle: 14.00%)
May 19 10:33:06 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:33:06 vehicle soc: 14%
May 19 10:33:06 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:33:06 vehicle range: 26km
May 19 10:33:06 openhab evcc[1704617]: [lp-1  ] DEBUG 2023/05/19 10:33:06 pv charge current: 51.6A = 16A + 35.6A (-8187W @ 1p)
May 19 10:33:10 openhab systemd[1]: Stopping evcc...
May 19 10:33:11 openhab evcc[1704617]: [main  ] DEBUG 2023/05/19 10:33:10 telemetry: charge: Δ10/10Wh @ 0W
May 19 10:33:11 openhab evcc[1704617]: [main  ] ERROR 2023/05/19 10:33:11 cannot save settings: attempt to write a readonly database (8)
May 19 10:33:11 openhab evcc[1704617]: [db    ] ERROR 2023/05/19 10:33:11 persist: attempt to write a readonly database (8)
May 19 10:33:11 openhab systemd[1]: evcc.service: Succeeded.
May 19 10:33:11 openhab systemd[1]: Stopped evcc.
May 19 10:33:11 openhab systemd[1]: evcc.service: Consumed 2.579s CPU time.
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.117.4

m-milinovsky, 2023-05-19:
Was ich schon versucht habe:

1. evcc stoppen und neu starten
2. influxdb und grafana stoppen (und service disablen)
3. reboot
4. backup der Datenbank und Ersetzen der DB durch das Backup (wie von manchen empfohlen)
```
sqlite3 /var/lib/evcc/evcc.db
sqlite> .backup main evcc.db.backup
sqlite> .exit
mv evcc.db.backup /var/lib/evcc/evcc.db
```
5. evcc nicht als Service sondern mit `evcc --log debug` starten: hier tritt das Problem nicht auf, allerdings wird hier auch nicht die Datenbak in /var/lib/evcc/ verwendet

andig, 2023-05-19:
Deine DB gehört root, der evcc Dienst läuft aber nicht als root.