evcc-io/evcc #8086: PV Überschuss Modus versucht mit 11kW zu laden

keineantwort, 2023-05-20:
### Describe the bug

Bei geringerem PV-Überschuss als 11kW versucht das System den ioniq5 trotzdem mit 11kW zu laden. Erst lädt das Auto kurz, dann wird der Speicher zu Hilfe genommen, dann wird kurz Strom aus dem Netz bezogen. EVCC sagt dann nach kurzer Zeit, dass zu wenig Überschuss vorhanden sei und stoppt die Ladung. Etwas später geht der Spaß von vorn los. 
Meine Erwartungshaltung: bei 4kW Überschuss wird mit 4KW geladen. Bei 6kW mit 6kW...

### Steps to reproduce

1. Auto anschließen
2. Warten bis Überschuss da ist

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: <id>

interval: 10s # control cycle interval

meters:
- type: template
  template: solarwatt 
  usage: grid  
  host: 192.168.1.x  
  name: grid1
- type: template
  template: solarwatt 
  usage: pv  
  host: 192.168.1.x  
  name: pv2
- type: template
  template: solarwatt 
  usage: battery  
  host: 192.168.1.x  
  name: battery3

chargers:
- type: template
  template: keba 
  host: 192.168.1.x  
  name: wallbox6

vehicles:
- type: template
  template: hyundai 
  title: ioniq5  
  user: <user>
  password: <pass>
  capacity: 72.4  
  language: de
  phases: 3  
  name: ev4
  
- type: template
  template: vw 
  title: e-up  
  user: <user>
  password: <pass>
  capacity: 18.7
  phases: 1  
  name: ev5
[... 682 lines cut ...]
[site  ] DEBUG 2023/05/20 09:58:53 battery power: -388W
[site  ] DEBUG 2023/05/20 09:58:53 grid power: 453W
[site  ] DEBUG 2023/05/20 09:58:54 site power: 65W
[lp-1  ] DEBUG 2023/05/20 09:58:54 charge currents: [15.8 15.7 15.7]A
[lp-1  ] DEBUG 2023/05/20 09:58:54 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/20 09:58:54 charge meter total import: 1037.158kWh
[lp-1  ] DEBUG 2023/05/20 09:58:54 charger status: C
[lp-1  ] DEBUG 2023/05/20 09:58:54 soc estimated: 30.18% (vehicle: 30.00%)
[lp-1  ] DEBUG 2023/05/20 09:58:54 vehicle soc: 30%
[lp-1  ] DEBUG 2023/05/20 09:58:54 vehicle soc limit: 100%
[lp-1  ] DEBUG 2023/05/20 09:58:54 vehicle range: 101km
[lp-1  ] DEBUG 2023/05/20 09:58:54 pv charge current: 5.91A = 6A + -0.0942A (65W @ 3p)
[lp-1  ] DEBUG 2023/05/20 09:58:54 site power 65W >= 0W disable threshold
[lp-1  ] DEBUG 2023/05/20 09:58:54 pv disable in 2m40s
[lp-1  ] DEBUG 2023/05/20 09:58:54 pv disable timer remaining: 2m40s
[lp-1  ] DEBUG 2023/05/20 09:58:55 charge meter total import: 1037.161kWh
```

### What type of operating system are you running?

Linux

### Version

0.117.4

andig, 2023-05-20:
Ich verstehe das Problem nicht:

```
[lp-1  ] DEBUG 2023/05/20 09:57:44 charger status: B
[lp-1  ] DEBUG 2023/05/20 09:57:44 pv charge current: 8.48A = 0A + 8.48A (-5851W @ 3p)
[site  ] DEBUG 2023/05/20 09:57:52 ----
[lp-1  ] DEBUG 2023/05/20 09:57:54 charger status: C
[lp-1  ] INFO 2023/05/20 09:57:54 start charging ->
```

ist erstmal richtig.

Aber: Dein System schwingt, die 10s Intervall sind schlicht zu kurz -> auf 30s setzen!

closed 2023-05-20 by andig

keineantwort, 2023-05-20:
🤔 Danke. Teste ich. 

Was mich wundert: Bis vor ein paar Tagen lief es noch ohne Probleme. Ich habe mit `prioritySoc: 50` und `bufferSoc: 75` experimentiert, damit mir die Batterie nicht beim Laden leergezogen wird. Vor dem Bugreport hatte ich die aber wieder entfernt. Das ändert aber nichts an dem o.g. Verhalten.

andig, 2023-05-20:
Das schwingen ist im Log deutlich sichtbar. Was vorher war kann ich leider nicht kommentieren.

premultiply, 2023-05-20:
`residualPower` fehlt auch.

keineantwort, 2023-05-20:
> Das schwingen ist im Log deutlich sichtbar. Was vorher war kann ich leider nicht kommentieren.
Das hat leider nichts verändert. Ich hab es noch in der Oberfläche gesehen. Als ich ein neues Log aufzeichnen wollte, war die Sonne weg. :/ Soll heute auch nur noch bedingt wieder kommen. Melde mich hier wieder mit einem neuen Log-Ausschnitt.

> `residualPower` fehlt auch.
Hab ich hinzugefügt. Hoffentlich kommt bald die Sonne wieder.

andig, 2023-05-20:
Ich bau dafür eine Warnung ein. ⚠️