evcc-io/evcc #8094: Keba-Modubs not showing real power usage

Duell10111, 2023-05-20:
### Describe the bug

When selecting "Schnell" charging option the WebInterface shows 11 kw, even if the real charging power is much lower.

[image]

WB:
[image]

### Steps to reproduce

1. Selecting Schnell charging option
2. Limiting the charging on vehicle side
3. Charging display incorrect
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: https
  host: charging.hub # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

sponsortoken: CENSORED

interval: 10s # control cycle interval

meters:
- type: template
  template: solaredge-hybrid 
  usage: grid  
  host: 192.168.178.10  
  port: 1502  
  timeout: 10s  
  name: grid1
  modbus: tcpip
  id: 1
- type: template
  template: solaredge-hybrid 
  usage: pv  
  host: 192.168.178.10  
  port: 1502  
  timeout: 10s  
  name: pv2
  modbus: tcpip
  id: 1
- type: template
  template: solaredge-hybrid 
  usage: battery  
  host: 192.168.178.10
  port: 1502  
  timeout: 10s  
  name: battery3
  modbus: tcpip
  id: 1

chargers:
  - name: keba
    type: template
    template: keba-modbus 

    # Modbus TCP
    modbus: tcpip
    id: 255
[... 1329 lines cut ...]
[lp-1  ] DEBUG 2023/05/20 13:12:56 vehicle range: 297km
1000: 00000003
[site  ] DEBUG 2023/05/20 13:13:06 ----
[lp-1  ] DEBUG 2023/05/20 13:13:06 charge power: 11040W
[site  ] DEBUG 2023/05/20 13:13:06 pv power: 5053W
[site  ] DEBUG 2023/05/20 13:13:06 battery soc: 59%
[site  ] DEBUG 2023/05/20 13:13:06 battery power: 1646W
[site  ] DEBUG 2023/05/20 13:13:06 grid power: -17W
[site  ] DEBUG 2023/05/20 13:13:06 site power: 1679W
1004: 00000007
1000: 00000003
[lp-1  ] DEBUG 2023/05/20 13:13:06 charger status: C
[lp-1  ] DEBUG 2023/05/20 13:13:06 vehicle soc: 63%
[lp-1  ] DEBUG 2023/05/20 13:13:06 vehicle soc limit: 78%
[lp-1  ] DEBUG 2023/05/20 13:13:06 vehicle range: 297km
1000: 00000003
```

### What type of operating system are you running?

Linux

### Version

evcc 0.117.4 (ea6d1601)

andig, 2023-05-20:
Was sagt denn 

    evcc charger

beim Laden? Hat die Keba einen Zähler?

Duell10111, 2023-05-20:
Das ist der Output von dem Befehl von dir.

`[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 01 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 01 00 00 00 07 ff 03 04 00 04 ca a5
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 02 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 02 00 00 00 07 ff 03 04 00 00 00 00
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 03 00 00 00 06 ff 03 03 ec 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 03 00 00 00 07 ff 03 04 00 00 00 07
1004: 00000007
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 04 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 04 00 00 00 07 ff 03 04 00 00 00 03
1000: 00000003
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 05 00 00 00 06 ff 03 03 e8 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 05 00 00 00 07 ff 03 04 00 00 00 03
1000: 00000003
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 06 00 00 00 06 ff 03 05 dc 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 06 00 00 00 07 ff 03 04 00 00 00 00
Charge status: C
Enabled:       true
Identifier:    <none>
Diagnostic dump:
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 07 00 00 00 06 ff 03 03 f6 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 07 00 00 00 07 ff 03 04 01 66 2e 57
        Serial: 23473751
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 08 00 00 00 06 ff 03 03 fa 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 08 00 00 00 07 ff 03 04 03 0a 2a 00
        Firmware:       3.10.42
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 09 00 00 00 06 ff 03 03 f8 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 09 00 00 00 07 ff 03 04 00 04 ca a5
        Product:        314021
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 0a 00 00 00 06 ff 03 06 0e 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 0a 00 00 00 07 ff 03 04 00 00 00 00
        Phases source:  0
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 0b 00 00 00 06 ff 03 06 10 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 0b 00 00 00 07 ff 03 04 00 00 00 00
        Phases state:   0
[keba  ] TRACE 2023/05/20 16:44:24 modbus: send 00 0c 00 00 00 06 ff 03 06 42 00 02
[keba  ] TRACE 2023/05/20 16:44:24 modbus: recv 00 0c 00 00 00 07 ff 03 04 00 00 00 1e
        Failsafe timeout:       30s`

premultiply, 2023-05-20:
Scheint keinen Zähler zu haben.
Daher sind die Werte von evcc nur Anhand der gesetzten Stromvorgaben nur geschätzt.

Hast du versehentlich die Green/DE-Edition gekauft?

Duell10111, 2023-05-20:
Kann sein, das ich keinen Zähler habe.
Eigentlich habe ich den SolarEdge Home EV Charger, der auf der Keba P30 basiert. 😅

premultiply, 2023-05-20:
Mmmmh. Zeigt die Box denn sonst irgendwo ihre Leistung und geladene Energie valide an?

Duell10111, 2023-05-20:
Ja also bei der Box kann ich das korrekt abrufen über das WebInterface auf Port 8080.

[image]

andig, 2023-05-20:
Vmtl. ist das decoding von 1016 falsch- die Doku sagt nicht ob das dezimal oder hexadezimal erfolgt. Ich schaus später an...

Duell10111, 2023-05-20:
> Vmtl. ist das decoding von 1016 falsch- die Doku sagt nicht ob das dezimal oder hexadezimal erfolgt:
> 
> ```
> hex: 04caa5
> dez: 314021
> ```
> 
> Das müsste man jetzt mal neben die Doku halten.

Also laut Doku sollte es Dezimal sein, dann würde der Wert auch Sinn machen. 
Siehe https://www.keba.com/download/x/dea7ae6b84/kecontactp30modbustcp_pgen.pdf Seite 12. 🙂

closed 2023-05-20 by andig