evcc-io/evcc #8096: Dynamic load management in "pv" working only after manual switch to "now" and back

cseelmann, 2023-05-20:
### Describe the bug

Hi, 

car is a hybrid with 2 phases, phase 2 is disabled with breaker in the electrical cabinet, so we load with 1 phase 6-16A, only 1 phase is detected due to missing current on 2. 
Wallbox is Innogy ebox Professional with manual rfid auth pre-charging due to backend accounting. 
Meter/Solar is Solaredge with modbustcp. 

Everything running as expected, except the dynamic load management only works if I switch from "pv" to "now" and back again every time I charge (same with "min+pv"). If I stay in "pv" load will only be 6A no matter how much Solar surplus (even far far beyond residual threshold). Did not see anything related to the effect in the logs while switch charging modes. 
 
This effect exists since I use evcc (02/2023), not only since v0.117.4. Any ideas how to fix? Thanks a lot in advance!

### Steps to reproduce

1. Reconnect car, authenticate
2. Excess solar production available, charging power is not being raised in pv to cover surplus production
3. Switch to "now" and after a few intervals back to "pv", dynamic loading is working as expected

### Configuration details

```yaml
sponsortoken: xxx
plant: xxx
network:
  schema: http
  host: 10.10.8.100
  port: 7070

interval: 10s

levels:
  site: trace
  LP1: trace
  e300de: trace

site:
  - title: home
    meters:
      grid: SolarEdge-grid
      pv: SolarEdge-pv
    residualPower: 200

loadpoints:
- title: LP1
  charger: ebox
  vehicle: mercedes
  mode: pv
  minCurrent: 6
  maxCurrent: 16
  phases: 1
  resetOnDisconnect: true
  soc:
    poll:
      mode: always
      interval: 60m
    estimate: true

meters:
  - name: SolarEdge-pv
    type: template
    template: solaredge-inverter
    usage: pv
    modbus: tcpip
    id: 1
    host: 10.10.7.40
    port: 502

  - name: SolarEdge-grid
    type: template
    template: solaredge-hybrid
    usage: grid
    modbus: tcpip
    id: 1
    host: 10.10.7.40
    port: 502

chargers:
  - name: ebox
    type: template
    template: innogy-ebox
    modbus: tcpip
    id: 1
    host: 10.10.6.30
    port: 502

vehicles:
- name: mercedes
  type: mercedes
  title: E300de
  clientId: xxx
  clientSecret: xxx
  vin: xxx
  phases: 1
  capacity: 13.5

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.39

  feedin:
    type: fixed
    price: 0.08
```

### Log details

```text
[main  ] INFO 2023/05/20 13:52:52 evcc 0.117.4
[main  ] INFO 2023/05/20 13:52:52 using config file: /etc/evcc.yaml
[main  ] INFO 2023/05/20 13:52:52 starting ui and api at :7070
[db    ] INFO 2023/05/20 13:52:52 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2023/05/20 13:52:53 poll mode '{always 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] INFO 2023/05/20 13:52:53 site config:
[site  ] INFO 2023/05/20 13:52:53   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/05/20 13:52:53     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/20 13:52:53     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/20 13:52:53   vehicles:
[site  ] INFO 2023/05/20 13:52:53     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/05/20 13:52:53 loadpoint 1:
[lp-1  ] INFO 2023/05/20 13:52:53   mode:        pv
[lp-1  ] INFO 2023/05/20 13:52:53   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/05/20 13:52:53   meters:      charge ✓
[lp-1  ] INFO 2023/05/20 13:52:53     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] INFO 2023/05/20 13:52:53 vehicle updated: unknown -> E300de
[main  ] INFO 2023/05/20 13:52:54 ensure the oauth client redirect/callback is configured for E300de: http://10.10.8.100:7070/oauth/vehicles/1/callback
[site  ] DEBUG 2023/05/20 13:52:54 ----
[site  ] DEBUG 2023/05/20 13:52:54 pv power: 3054W
[site  ] DEBUG 2023/05/20 13:52:54 grid power: -404W
[site  ] DEBUG 2023/05/20 13:52:54 site power: -204W
[lp-1  ] INFO 2023/05/20 13:52:54 car connected
[lp-1  ] INFO 2023/05/20 13:52:54 start charging ->
[lp-1  ] ERROR 2023/05/20 13:52:54 vehicle soc: Get "https://api.mercedes-benz.com/vehicledata/v2/vehicles/W1Kxxx/containers/electricvehicle": oauth2: token expired and refresh token is not set
[site  ] DEBUG 2023/05/20 13:53:04 ----
[site  ] DEBUG 2023/05/20 13:53:04 pv power: 2776W
[site  ] DEBUG 2023/05/20 13:53:04 grid power: -993W
[site  ] DEBUG 2023/05/20 13:53:04 site power: -793W
[site  ] DEBUG 2023/05/20 13:53:14 ----
[site  ] DEBUG 2023/05/20 13:53:14 pv power: 2778W
[site  ] DEBUG 2023/05/20 13:53:14 grid power: -81W
[site  ] DEBUG 2023/05/20 13:53:14 site power: 119W
[site  ] DEBUG 2023/05/20 13:53:24 ----
[site  ] DEBUG 2023/05/20 13:53:24 pv power: 2768W
[site  ] DEBUG 2023/05/20 13:53:24 grid power: -311W
[site  ] DEBUG 2023/05/20 13:53:24 site power: -111W
[site  ] DEBUG 2023/05/20 13:53:34 ----
[site  ] DEBUG 2023/05/20 13:53:34 pv power: 2755W
[site  ] DEBUG 2023/05/20 13:53:34 grid power: -114W
[site  ] DEBUG 2023/05/20 13:53:34 site power: 86W
[site  ] DEBUG 2023/05/20 13:53:44 ----
[site  ] DEBUG 2023/05/20 13:53:44 pv power: 2883W
[site  ] DEBUG 2023/05/20 13:53:44 grid power: -400W
[site  ] DEBUG 2023/05/20 13:53:44 site power: -200W
```

### What type of operating system are you running?

Docker container

### Version

0.117.4

andig, 2023-05-20:
Bitte debug log vom Loadpoint ergänzen.

cseelmann, 2023-05-22:
Hier das Log vom Verhalten, Überschuss ist vorhanden, mode ist pv. Start in pv, bleibt dabei stehen, erst nach Wechsel auf now (17:03:53) und zurück auf pv (17:04:23) wird dynamisch geregelt. 
```
2023-05-22T17:00:23.608597+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:23 ----
2023-05-22T17:00:23.613663+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:23 charge power: 0W
2023-05-22T17:00:23.671415+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:23 pv power: 2589W
2023-05-22T17:00:23.719663+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:23 grid power: -2438W
2023-05-22T17:00:23.720005+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:23 site power: -2438W
2023-05-22T17:00:23.723713+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:23 charge currents: [0.06 0 0]A
2023-05-22T17:00:23.725704+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:23 charger status: A
2023-05-22T17:00:23.726047+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:23 next soc poll remaining time: 22m9s
2023-05-22T17:00:33.608601+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:33 ----
2023-05-22T17:00:33.613490+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:33 charge power: 0W
2023-05-22T17:00:33.642963+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:33 pv power: 2427W
2023-05-22T17:00:33.672996+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:33 grid power: -2255W
2023-05-22T17:00:33.673750+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:33 site power: -2255W
2023-05-22T17:00:33.678171+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:33 charge currents: [0.06 0 0]A
2023-05-22T17:00:33.679594+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:33 charger status: A
2023-05-22T17:00:33.679809+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:33 next soc poll remaining time: 22m0s
2023-05-22T17:00:43.608839+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:43 ----
2023-05-22T17:00:43.613368+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:43 charge power: 0W
2023-05-22T17:00:43.640805+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:43 pv power: 2486W
2023-05-22T17:00:43.663236+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:43 grid power: -2329W
2023-05-22T17:00:43.663618+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:43 site power: -2329W
2023-05-22T17:00:43.667154+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:43 charge currents: [0.12 0 0]A
2023-05-22T17:00:43.668709+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:43 charger status: B
2023-05-22T17:00:43.668937+02:00 smarthome evcc[13002]: [lp-1  ] INFO 2023/05/22 17:00:43 car connected
2023-05-22T17:00:43.960423+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:43 vehicle soc: 2%
2023-05-22T17:00:44.105792+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:44 vehicle range: 0km
2023-05-22T17:00:44.117619+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:44 pv charge current: 10.1A = 0A + 10.1A (-2329W @ 1p)
2023-05-22T17:00:44.118227+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:44 site power -2329W <= 0W enable threshold
2023-05-22T17:00:44.118632+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:44 pv enable in 0s
2023-05-22T17:00:44.118756+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:44 pv enable timer elapsed
2023-05-22T17:00:44.341114+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:44 max charge current: 6A
2023-05-22T17:00:44.488341+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:44 charger enable
2023-05-22T17:00:44.488542+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:44 wake-up timer: start
2023-05-22T17:00:53.608240+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:53 ----
2023-05-22T17:00:53.613662+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:53 charge power: 0W
2023-05-22T17:00:53.682149+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:53 pv power: 2285W
2023-05-22T17:00:53.703159+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:53 grid power: -2133W
2023-05-22T17:00:53.703941+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:00:53 site power: -2133W
2023-05-22T17:00:53.707817+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:53 charge currents: [0.06 0 0]A
2023-05-22T17:00:53.709306+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:53 charger status: B
2023-05-22T17:00:53.709499+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:53 next soc poll remaining time: 59m49s
2023-05-22T17:00:53.725568+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:53 pv charge current: 9.27A = 0A + 9.27A (-2133W @ 1p)
2023-05-22T17:00:53.725900+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:53 pv timer reset
2023-05-22T17:00:53.726031+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:53 pv timer inactive
2023-05-22T17:00:54.180098+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:00:54 max charge current: 9.27A
2023-05-22T17:01:03.608923+02:00 smarthome evcc[13002]: [site  ] DEBUG 2023/05/22 17:01:03 ----
2023-05-22T17:01:03.616398+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:01:03 charge power: 1373W
[... cut ...]
2023-05-22T17:05:03.700859+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:05:03 soc estimated: 2.80% (vehicle: 2.00%)
2023-05-22T17:05:03.700962+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:05:03 vehicle soc: 3%
2023-05-22T17:05:03.701360+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:05:03 vehicle range: 0km
2023-05-22T17:05:03.715667+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:05:03 pv charge current: 9.76A = 10.4A + -0.591A (136W @ 1p)
2023-05-22T17:05:03.893104+02:00 smarthome evcc[13002]: [lp-1  ] DEBUG 2023/05/22 17:05:03 max charge current: 9.76A

```

cseelmann, 2023-06-11:
@andig hast Du die Ergänzung des Logs mitbekommen? 
Bleiben wir hier im Issue oder soll ich ein neues aufmachen?

Zwischenzeitlich habe ich auf der eBox getestet, ob die Einstellungen das Lastmanagements irgendeine Auswirkung haben, leider nicht. Egal ob eingeschaltet und verschiedene max. A oder ausgeschaltet, EVCC bleibt bei 6A Ladeleistung bei Start im PV Modus auch bei reichlich Überschuss und startet die Dymanik erst nach Switch auf now und zurück zu pv|minpv. Danke

PeterPablo, 2023-06-14:
gleiches wie @cseelmann meine ich jüngst (seit Update auf v0.118.0?!) auch zu beobachten

andig, 2023-06-14:
Aus dem Log ist nichts zu erkennen. evcc regelt brav vor sich hin, die Box ignoriert es. Braucht einen Mitschnitt der Kommunikation zur Box...

cseelmann, 2023-06-18:
[ecu_log.csv](https://github.com/evcc-io/evcc/files/11781024/ecu_log.csv)

Hier das Log der Box, ich habe alle aus meiner Sicht irrelevanten Infos entfernt (alle Modbus Connects/Disconnects, occp Kommunikation, RFID auths etc.). 

Ca. ab Zeile 95 geht es mit der Load Adaption los, die Box meldet auch brav den DC_Change, setzt diesen aber nicht um. Auch später ab Zeile 380 bei der Adoption auf 10A nicht, erst nach Zeile 484 nach manueller "now" auf 16A wird anschließend im "pv" korrekt geregelt. Ich sehe im Log der Box keine Fehlermeldungen nach jeder Load Adoption. 

pv charge current wird im evcc log correct angezeigt, nur die charge currents bleiben bei knapp 6A. 

Ich hatte neulich den Effekt, dass bei üppigem Überschuss von mehr als ein paar hundert Watt im "pv" Modus die Dynamik ohne Switch auf "now" und zurück auf "pv" direkt funktionierte, das werde ich bei Gelgenheit noch mal testen. 

@andig: hast Du noch eine Idee, was getestet werden kann? Ich habe probehalber den Intervall auf 30s gesetzt, um der Box für den ersten DC_Change von 6A (in diesem Fall auf 9A) mehr Zeit zu geben, ohne Erfolg.

andig, 2023-06-23:
> Es zeigte sich, dass die Dynamik irgendwann anspringt, leider nicht besonders reproduzierbar 

Also dann stimmt das Ansprechen der Box schonmal.

> Zuvor hatte ich vermutet, dass die Ladesteuerung im Fahrzeug vielleicht mit 16A angeschubst werden muss

Das klingt nach Zoe und Blindleistung, ist aber keien Zoe :(

> Überschuss von einigen A war zum Begin des Ladens stets vorhanden, gefühlt scheint es mindestens 12A zu benötigen, bevor die Regelung greift.

Soweit ich im Log sehe, greift die Regelung immer.

Ich hab leider überhaupt keine Idee, warum da Auto die Leistung nicht aufnimmt. Bliebe nur mal Test mit anderem Fahrzeug. Mangels irgendeiner Idee mache ich zu, tut mir leid :(

closed 2023-06-23 by andig

cseelmann, 2023-07-04:
Danke für Deinen Support @andig!

Nachdem klar war, dass weder EVCC noch Box in Frage kommen, konnte es nur das Auto sein. 
Anruf bei der Mercedes Hotline: Hilfe/Ticket nur, wenn der Freundliche nicht weiterkommt. 
Anruf beim Freundlichen und Frage, ob es dort Elektrospezis gibt: Ja, jeder Meister, aber aktuell keine Zeit. Rückruf erfolgte nie. 

Ich erinnerte mich daran, dass ich damals vor der Installation der Wallbox eine Einstellung im EQ Menü für den Maximalen Ladestrom an Haushaltssteckdosen auf 6A gestellt hatte, um beim Laden mit dem Ziegelstein die Leitung zu schonen:

[image]

Siehe da, nachdem ich hier auf Maximum gestellt hatte, regelt das Fahrzeug ab 6A an der Wallbox wie es soll. 
In der Betriebsanleitung oder Online habe ich keine Erläuterung dieses Menüs oder des Verhaltens gefunden. 

Für die SuFu und andere mit dem Problem: 
Es handelt sich um einen Mercedes E300de W213 Facelift BJ 2020