evcc-io/evcc #8114: No message IDs available

cburghardt, 2023-05-21:
### Describe the bug

This morning evcc completey locked up, I got an endless loop of the following error message in the logs:
May 21 08:59:16 evcc evcc[6003]: [mqtt  ] ERROR 2023/05/21 08:52:32 send: evcc/site/error: no message IDs available

The memory increased constantly and the host was then not available anymore. I had to reboot the machine to make it work again.
mqtt was available without issues, also no network errors during this time.

### Steps to reproduce

I am not able to reproduce it.

### Configuration details

```yaml
meters:
  - name: pv
    type: custom
    power:
      source: mqtt
      topic: energy/pv/power
      timeout: 40s  
    currents:
      - source: mqtt    
        topic: energy/pv/currentl1
        timeout: 40s  
      - source: mqtt    
        topic: energy/pv/currentl2
        timeout: 40s  
      - source: mqtt    
        topic: energy/pv/currentl3
        timeout: 40s  
  - name: grid      
    type: custom
    power:  
      source: mqtt
      topic: energy/meter/power
      scale: -1
      timeout: 40s  
    currents:
      - source: mqtt    
        topic: energy/meter/currentl1
        timeout: 40s  
      - source: mqtt    
        topic: energy/meter/currentl2
        timeout: 40s  
      - source: mqtt    
        topic: energy/meter/currentl3
        timeout: 40s  
  - name: battery
    type: custom
    power:
      source: mqtt
      topic: energy/battery/power
      scale: -1
      timeout: 40s  
    soc:
      source: mqtt
      topic: energy/battery/soc
      timeout: 40s
```

### Log details

```text
May 21 08:59:16 evcc evcc[6003]: [mqtt  ] ERROR 2023/05/21 08:52:32 send: evcc/site/error: no message IDs available
```

### What type of operating system are you running?

Linux

### Version

0.117.4

andig, 2023-05-21:
Dein mqtt Server nimmt die Nachrichten von evcc nicht an- irgendwann geht das dann nicht mehr, siehe https://github.com/eclipse/paho.mqtt.golang/commit/fbd79e4a3eca76c934da91ba8e1e9d1cf5876d53

closed 2023-05-21 by andig

cburghardt, 2023-05-21:
Der mosquito server bereit allerdings mit keiner anderen Software Probleme und ich habe sehr viele integration parallel laufen.

andig, 2023-05-21:
> I am not able to reproduce it.

Und Du bist Dir sicher dass das auch genau in diesem einen Fall so war? Ohne Repro gibts nix zu tun und die Fehlermeldung ist eindeutig.