evcc-io/evcc #8116: Ocpp: handle chargepoint stopping transaction

andig, 2023-05-21:
It seems we do not handle cp stopping transaction correctly:

```
[site  ] DEBUG 2023/05/20 10:55:45 ----
[lp-1  ] DEBUG 2023/05/20 10:55:45 charge power: 3785W
[lp-2  ] DEBUG 2023/05/20 10:55:45 charge power: 20W
[site  ] DEBUG 2023/05/20 10:55:45 pv power: 4069W
[site  ] DEBUG 2023/05/20 10:55:45 grid power: -55W
[site  ] DEBUG 2023/05/20 10:55:45 site power: -55W
[lp-2  ] DEBUG 2023/05/20 10:55:45 charge meter total import: 0.012kWh
[lp-2  ] DEBUG 2023/05/20 10:55:45 charger status: C
[lp-2  ] DEBUG 2023/05/20 10:55:45 pv charge current: 5.05A = 4.81A + 0.239A (-55W @ 1p)
[lp-2  ] DEBUG 2023/05/20 10:55:45 max charge current: 5.05A
[ocpp  ] TRACE 2023/05/20 10:56:25 received JSON message from EVB-P20252147: [2,"2bfb47b9-df13-4670-b78b-408bb598fce3","Heartbeat",{}]
[ocpp  ] TRACE 2023/05/20 10:56:25 sent JSON message to EVB-P20252147: [3,"2bfb47b9-df13-4670-b78b-408bb598fce3",{"currentTime":"2023-05-20T08:56:25Z"}]
[ocpp  ] TRACE 2023/05/20 10:56:41 received JSON message from EVB-P20252147: [2,"2bfb47c9-df13-4670-b78b-408be7c61347","StopTransaction",{"transactionId":1,"idTag":"04C6A80ABF6680","timestamp":"2023-05-20T08:56:41Z","meterStop":2034698,"reason":"EVDisconnected"}]
[ocpp  ] TRACE 2023/05/20 10:56:41 sent JSON message to EVB-P20252147: [3,"2bfb47c9-df13-4670-b78b-408be7c61347",{"idTagInfo":{"status":"Accepted"}}]
[ocpp  ] TRACE 2023/05/20 10:56:42 received JSON message from EVB-P20252147: [2,"2bfb47ca-df13-4670-b78b-408b0501664b","StatusNotification",{"connectorId":1,"status":"Available","errorCode":"NoError","info":"EV side disconnected.","timestamp":"2023-05-20T08:56:42Z"}]
[ocpp  ] TRACE 2023/05/20 10:56:42 sent JSON message to EVB-P20252147: [3,"2bfb47ca-df13-4670-b78b-408b0501664b",{}]
[ocpp  ] TRACE 2023/05/20 10:56:42 received JSON message from EVB-P20252147: [2,"2bfb47ca-df13-4670-b78b-408bd63bb04b","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Available,NoError,EV side disconnected.,0,Off,0,{0,8000,100},{2034698,0},{A,12145,12220,61568},0,0,340,2023-05-20T08:56:42Z,0,60,{238,237,238,0,1,0,996,996,0},260,0,0,115,0,270,900,5008"}]
[ocpp  ] TRACE 2023/05/20 10:56:42 sent JSON message to EVB-P20252147: [3,"2bfb47ca-df13-4670-b78b-408bd63bb04b",{"status":"Accepted"}]
[site  ] DEBUG 2023/05/20 10:56:45 ----
[lp-1  ] DEBUG 2023/05/20 10:56:45 charge power: 3785W
[lp-2  ] DEBUG 2023/05/20 10:56:45 charge power: 20W
[site  ] DEBUG 2023/05/20 10:56:45 pv power: 4142W
[site  ] DEBUG 2023/05/20 10:56:45 grid power: -104W
[site  ] DEBUG 2023/05/20 10:56:45 site power: -104W
[lp-1  ] DEBUG 2023/05/20 10:56:45 charger status: A
[lp-1  ] INFO 2023/05/20 10:56:45 stop charging <-
[lp-1  ] DEBUG 2023/05/20 10:56:45 wake-up timer: start
[lp-1  ] INFO 2023/05/20 10:56:45 car disconnected
[lp-1  ] WARN 2023/05/20 10:56:45 charger out of sync: expected enabled, got disabled
[ocpp  ] TRACE 2023/05/20 10:56:45 sent JSON message to EVB-P20252147: [2,"2931443816","RemoteStartTransaction",{"connectorId":1,"idTag":"04C6A80ABF6680","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":8.228260869565219}]}}}]
[ocpp  ] TRACE 2023/05/20 10:56:45 received JSON message from EVB-P20252147: [3,"2931443816",{"status":"Accepted"}]
[lp-1  ] DEBUG 2023/05/20 10:56:45 charger: guard elapse
[lp-1  ] DEBUG 2023/05/20 10:56:45 guard timer inactive
[ocpp  ] TRACE 2023/05/20 10:56:45 sent JSON message to EVB-P20252147: [2,"237438955","RemoteStopTransaction",{"transactionId":0}]
[ocpp  ] TRACE 2023/05/20 10:56:45 received JSON message from EVB-P20252147: [4,"237438955","ProtocolError","Sender's message does not comply with protocol specification.",{}]
[lp-1  ] ERROR 2023/05/20 10:56:45 charger disable: ocpp message (237438955): ProtocolError - Sender's message does not comply with protocol specification.
```

andig, 2023-05-29:
Hier beendet das Fzg die Ladung:

    [ocpp  ] TRACE 2023/05/20 10:56:42 received JSON message from EVB-P20252147: [2,"2bfb47ca-df13-4670-b78b-408b0501664b","StatusNotification",{"connectorId":1,"status":"Available","errorCode":"NoError","info":"EV side disconnected.","timestamp":"2023-05-20T08:56:42Z"}]
    [ocpp  ] TRACE 2023/05/20 10:56:42 sent JSON message to EVB-P20252147: [3,"2bfb47ca-df13-4670-b78b-408b0501664b",{}]

Damit gilt Status A (`Available`). Das passt nicht zum erwarteten "enabled" Bild:

    [lp-1  ] WARN 2023/05/20 10:56:45 charger out of sync: expected enabled, got disabled

Also schalten wir wieder ein, was bei Status A aber unsinnig ist:

    [ocpp  ] TRACE 2023/05/20 10:56:45 sent JSON message to EVB-P20252147: [2,"2931443816","RemoteStartTransaction",{"connectorId":1,"idTag":"04C6A80ABF6680","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":8.228260869565219}]}}}]

Merkwürdig ist, dass die Wallbox das akzeptiert- Firmware bug?:

    [ocpp  ] TRACE 2023/05/20 10:56:45 received JSON message from EVB-P20252147: [3,"2931443816",{"status":"Accepted"}]

Das postwendende Abschalten akzeptiert sie allerdings nicht mehr. Unklar ist, warum evcc hier mit `txnId: 0` abgeschaltet - das müsste egtl. die id der gestartet Transaktion sein:

    [ocpp  ] TRACE 2023/05/20 10:56:45 sent JSON message to EVB-P20252147: [2,"237438955","RemoteStopTransaction",{"transactionId":0}]
    [ocpp  ] TRACE 2023/05/20 10:56:45 received JSON message from EVB-P20252147: [4,"237438955","ProtocolError","Sender's message does not comply with protocol specification.",{}]
    [lp-1  ] ERROR 2023/05/20 10:56:45 charger disable: ocpp message (237438955): ProtocolError - Sender's message does not comply with protocol specification.

andig, 2023-05-29:
Bleibt die Frage, wie das zu fixen ist.

@premultiply sollten wir vllt. bei Status A einfach drauf verzichten, einen `enabled` Status wieder herzustellen der mangels Auto ohnehin sinnlos ist und den "out of sync" Error in diesem Fall einfach ignorieren? Das wäre dann analog https://github.com/evcc-io/evcc/pull/8126/files#diff-c456526d4bd9022a488ce32de208b40d86016e2c40b0c81a2dffe557c186377dR634 auch für das `Enable()` bei abgestecktem Fahrzeug umzusetzen?

premultiply, 2023-05-29:
Soweit ich OCPP verstehe müsste ja eigentlich(TM) beim Neuanstecken eines Fahrzeugs ja immer erstmal disabled sein bis das CS die Freigabe erteilt.

Von daher wäre es in der Tat bei abgestecktem Fahrzeug sinnvoll auf eine Prüfung/Änderung dahingehend zu verzichten.

andig, 2023-05-29:
> Soweit ich OCPP verstehe müsste ja eigentlich(TM) beim Neuanstecken eines Fahrzeugs ja immer erstmal disabled sein bis das CS die Freigabe erteilt.

Check, aber darum geht es hier nicht: der Fall hier ist ja ein Ladevorgang bei dem das Fahrzeug abgesteckt wird.

> Von daher wäre es in der Tat bei abgestecktem Fahrzeug sinnvoll auf eine Prüfung/Änderung dahingehend zu verzichten.

Siehe PR- das wäre global und nicht für Ocpp.

premultiply, 2023-05-29:
Ja, lass uns das mal probieren.

e-fahrer-jochen, 2023-06-11:
Hallo
Ist dieses Verhalten schon in der 118er Version enthalten?

Habe an meiner EV-Box Elvi diese Fehlermeldung:

charger: ConnectorLockFailure: No car attached.

Die Box startet immer automatisch obwohl ich den Autostart in der Box ausgeschaltet habe.
Eine Freigabe muss ich nicht ausführen.
Im everon Portal war die Autostartfunktion noch aktiv. Das habe ich einfach durch die neue evcc Verbindung getrennt.

Gruß.
(weiter so. Habe bald alle Problemkinder (Smart, EV-Box) am Start. Nur der Kostal WR hat sofort komplett funktioniert)

andig, 2023-06-11:
> ConnectorLockFailure

hat bix hiermit zu tun

rtrbt, 2023-06-16:
Wenn ich das richtig verstehe, löst der PR dann auch https://github.com/evcc-io/evcc/issues/5055#issuecomment-1305359228

binzman, 2023-07-15:
Ich habe eben noch durch Zufall entdeckt, dass es bei meiner Wallbox (evbox) eine Einstellung gibt, ob der Ladevorgang bei Ziehen des Kabels aus dem Auto beendet werden soll.

Dieser Wert ist aktuell aktiviert:

`{"key":"StopTransactionOnEVSideDisconnect","readonly":false,"value":"true"}`

Erklärt, warum die evbox den Ladevorgang von sich aus beendet, wenn der Stecker gezogen wird und dies dann an evcc meldet. Finde ich prinzipiell aber auch gut so

e-fahrer-jochen, 2023-07-30:
bei meiner evBox klappt das jetzt mit dem "Ziehen des Kabels aus dem Auto"  
[image]

Jul 30 11:27:02 openhab evcc[393]: [lp-1  ] INFO 2023/07/30 11:27:02 car connected
Jul 30 11:27:17 openhab evcc[393]: [lp-1  ] INFO 2023/07/30 11:27:17 start charging ->
Jul 30 11:28:03 openhab evcc[393]: [lp-1  ] INFO 2023/07/30 11:28:03 vehicle updated: unknown -> Smart
Jul 30 11:53:32 openhab evcc[393]: [lp-1  ] INFO 2023/07/30 11:53:32 stop charging <-
Jul 30 11:53:47 openhab evcc[393]: [lp-1  ] INFO 2023/07/30 11:53:47 car disconnected
Jul 30 11:53:47 openhab evcc[393]: [lp-1  ] INFO 2023/07/30 11:53:47 vehicle updated: Smart -> unknown

https://github.com/evcc-io/evcc/pull/9075

Danke. War aktuell mein vorletztes Problem.

andig, 2023-07-30:
Oh, das ist ein schöner Nebeneffekt :)

sweber, 2023-08-04:
Also ich habe mit meiner Wallbox Pulsar Plus genau das Problem.
Wenn bei einem aktiven Ladevorgang das Kabel vom Auto getrennt wird kommt danach Evcc nicht mehr klar.

 
```
charger: ocpp message (386875446): ProtocolError - validation of RemoteStartTransaction failed
charger out of sync: expected enabled, got disabled
```

sweber, 2023-08-11:
Wenn es hilft kann ich gerne ein Log erzeugen wo ich den Fehler reproduziere.

andig, 2023-08-13:
@sweber es wäre tatsächlich schön, nochmal ein Log vom aktuellen Nightly zu erzeugen (debug,ocpp:trace). Danke!

sweber, 2023-08-13:
Habe eben versucht mit dem aktuellen Nightly das Problem zu repoduzieren. Ist mir nicht gelungen. Ich werde es morgen nachmittag mit PV Strom nochmal probieren. ggfs. passiert es nur damit. Beim händischen anschalten (min + PV oder Schnell) hat sich jetzt alles korrekt verhalten.

Gab es im nightly noch eine Anpassung im Bereich von OCPP?

andig, 2023-08-13:
Gabs nicht, aber Berichte, dass es noch da ist.

SKB-CGN, 2023-08-18:
Diese Fehlermeldung tritt bei mir mit der PulsarPlus auch auf und lässt sich nur durch Neustart des evcc beheben. Sonst pendelt sie in der OutOfSync-Schleife bzw meldet den RemoteStartTransaction Fehler.

Gibt es eine absehbare Zeit, wann das Update zur Verfügung steht?

dscho, 2023-08-23:
> nochmal ein Log vom aktuellen Nightly zu erzeugen (debug,ocpp:trace)

Jetzt habe sogar ich (endlich!) kapiert, was mit dieser kurzen, äusserst würzigen Nebenbemerkung gemeint war. Meine `evcc.yml` enthält jetzt folgenden Abschnitt:

```yml
log: debug
levels:
  cache: error
  ocpp: trace
```

Nur so für die Nachwelt, falls jemand anderes das gleiche Verständnisproblem hat.

Aber den Fehler, den ich so oft sehe, den habe ich gerade nicht nachstellen können, nachdem das Laden aufgehört hat weil 80% erreicht waren, und ich den Stecker ziehe, hält `evcc` gerade still (Vorführeffekt!):

```
[...]
Aug 23 12:35:25 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:25 ----
Aug 23 12:35:25 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:25 charge power: 4140W
Aug 23 12:35:25 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:25 pv power: 3970W
Aug 23 12:35:25 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:25 battery soc: 70%
Aug 23 12:35:25 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:25 battery power: 720W
Aug 23 12:35:25 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:25 grid power: -0W
Aug 23 12:35:25 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:25 site power: 1020W
Aug 23 12:35:25 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:25 charger status: C
Aug 23 12:35:25 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:25 soc estimated: 79.98% (vehicle: 79.00%)
Aug 23 12:35:25 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:25 vehicle soc: 80%
Aug 23 12:35:25 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:25 vehicle soc limit: 80%
Aug 23 12:35:25 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:25 vehicle range: 284km
Aug 23 12:35:25 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:25 pv charge current: 4.52A = 6A + -1.48A (1020W @ 3p)
Aug 23 12:35:26 raspberrypi evcc[14684]: [ocpp  ] TRACE 2023/08/23 12:35:26 received JSON message from evcc: [2,"268724853","Heartbeat",{}]
Aug 23 12:35:26 raspberrypi evcc[14684]: [ocpp  ] TRACE 2023/08/23 12:35:26 sent JSON message to evcc: [3,"268724853",{"currentTime":"2023-08-23T11:35:26Z"}]
Aug 23 12:35:35 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:35 ----
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 charge power: 4140W
Aug 23 12:35:35 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:35 pv power: 3567W
Aug 23 12:35:35 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:35 battery soc: 70%
Aug 23 12:35:35 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:35 battery power: 969W
Aug 23 12:35:35 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:35 grid power: 124W
Aug 23 12:35:35 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:35 site power: 1392W
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 charger status: C
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 soc estimated: 80.00% (vehicle: 79.00%)
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 vehicle soc: 80%
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 vehicle soc limit: 80%
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 vehicle range: 284km
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 targetSoc reached: 80.0% > 80%
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 charger: guard elapse
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 guard timer inactive
Aug 23 12:35:35 raspberrypi evcc[14684]: [ocpp  ] TRACE 2023/08/23 12:35:35 sent JSON message to evcc: [2,"815096732","RemoteStopTransaction",{"transactionId":1}]
Aug 23 12:35:35 raspberrypi evcc[14684]: [ocpp  ] TRACE 2023/08/23 12:35:35 received JSON message from evcc: [3,"815096732",{"status": "Accepted"}]
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 charger disable
Aug 23 12:35:35 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:35 wake-up timer: stop
Aug 23 12:35:35 raspberrypi evcc[14684]: [ocpp  ] TRACE 2023/08/23 12:35:35 received JSON message from evcc: [2,"942853241","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Finishing","timestamp": "2023-08-23T11:35:35Z"}]
Aug 23 12:35:35 raspberrypi evcc[14684]: [ocpp  ] TRACE 2023/08/23 12:35:35 sent JSON message to evcc: [3,"942853241",{}]
Aug 23 12:35:35 raspberrypi evcc[14684]: [ocpp  ] TRACE 2023/08/23 12:35:35 received JSON message from evcc: [2,"1450003826","StopTransaction",{"meterStop": 3645846,"timestamp": "2023-08-23T11:35:35Z","idTag": "","transactionData": [],"reason": "Remote","transactionId": 1}]
Aug 23 12:35:35 raspberrypi evcc[14684]: [ocpp  ] TRACE 2023/08/23 12:35:35 sent JSON message to evcc: [3,"1450003826",{"idTagInfo":{"status":"Accepted"}}]
Aug 23 12:35:45 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:45 ----
Aug 23 12:35:45 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:45 charge power: 0W
Aug 23 12:35:45 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:45 pv power: 3936W
Aug 23 12:35:45 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:45 battery soc: 70%
Aug 23 12:35:45 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:35:45 battery power: 1W
[... cut ...]
Aug 23 12:35:55 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:35:55 targetSoc reached: 80.0% > 80%
Aug 23 12:36:05 raspberrypi evcc[14684]: [site  ] DEBUG 2023/08/23 12:36:05 ----
Aug 23 12:36:05 raspberrypi evcc[14684]: [lp-1  ] DEBUG 2023/08/23 12:36:05 charge power: 0W
[...]
```

Ich werde das auf jeden Fall weiter mit dieser Einstellung laufen lassen. Ach ja, das ist mit 0.118.11, also nicht mit 0.119.0, aber in den Release Notes steht ja nichts über die Pulsar Plus oder OCPP drin, das sollte also gehüpft wie getüpft sein.

dscho, 2023-09-01:
So, aber nun habe ich ein Log von dem Problem!

```
[... car connected, but for a while no longer charging already because it had reached 80% ...]
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:25 ----
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:25 charge power: 0W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:25 pv power: 631W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:25 battery soc: 70%
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:25 battery power: 969W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:25 grid power: 742W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:25 site power: 2012W
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:25 charger status: B
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:25 pv charge current: 0A = 0A + -2.92A (2012W @ 3p)
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:35 ----
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:35 charge power: 0W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:35 pv power: 656W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:35 battery soc: 70%
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:35 battery power: 969W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:35 grid power: 654W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:35 site power: 1923W
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:35 charger status: B
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:35 pv charge current: 0A = 0A + -2.79A (1923W @ 3p)
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:36 received JSON message from evcc: [2,"681804617","Heartbeat",{}]
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:36 sent JSON message to evcc: [3,"681804617",{"currentTime":"2023-08-31T11:43:36Z"}]
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:37 received JSON message from evcc: [2,"1747185126","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Finishing","timestamp": "2023-08-31T11:43:36Z"}]
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:37 sent JSON message to evcc: [3,"1747185126",{}]
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:38 received JSON message from evcc: [2,"1127599200","StopTransaction",{"meterStop": 3654023,"timestamp": "2023-08-31T11:43:37Z","idTag": "","transactionData": [],"reason": "EVDisconnected","transactionId": 5}]
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:38 sent JSON message to evcc: [3,"1127599200",{"idTagInfo":{"status":"Accepted"}}]
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:38 received JSON message from evcc: [2,"1378664633","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Available","timestamp": "2023-08-31T11:43:37Z"}]
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:38 sent JSON message to evcc: [3,"1378664633",{}]
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 ----
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 charge power: 0W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 pv power: 673W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 battery soc: 70%
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 battery power: 970W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 grid power: 638W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 site power: 1908W
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 charger status: A
evcc[14684]: [lp-1  ] INFO 2023/08/31 12:43:45 car disconnected
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 set charge mode: pv
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 set min current: 6
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 set max current: 16
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 set min soc: 0
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 set target soc: 100
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 set priority: 0
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:45 sent JSON message to evcc: [2,"2322263944","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:45 received JSON message from evcc: [3,"2322263944",{"status": "Accepted"}]
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 charger: guard elapse
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 guard timer inactive
evcc[14684]: [lp-1  ] ERROR 2023/08/31 12:43:45 charger disable: unknown transaction running, cannot disable
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 ----
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 charge power: 0W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 pv power: 673W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 battery soc: 70%
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 battery power: 970W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 grid power: 638W
evcc[14684]: [site  ] DEBUG 2023/08/31 12:43:45 site power: 1908W
evcc[14684]: [lp-1  ] DEBUG 2023/08/31 12:43:45 charger status: A
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:45 sent JSON message to evcc: [2,"3260672042","RemoteStartTransaction",{"connectorId":1,"idTag":"evcc","chargingProfile":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
evcc[14684]: [ocpp  ] TRACE 2023/08/31 12:43:45 received JSON message from evcc: [3,"3260672042",{"status": "Rejected"}]
[... cut ...] eKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
Sep 01 08:08:07 raspberrypi evcc[14684]: [ocpp  ] TRACE 2023/09/01 08:08:07 received JSON message from evcc: [3,"1217778617",{"status": "Rejected"}]
Sep 01 08:08:07 raspberrypi evcc[14684]: [lp-1  ] ERROR 2023/09/01 08:08:07 charger: Rejected
[... this is when I took a snapshot of the log ...]
```

Das ist mit evcc 0.118.11, ich werde jetzt das Update starten und weiter loggen.

dscho, 2023-09-08:
@andig hilft Dir das, das Problem besser zu verstehen?

andig, 2023-09-15:
Ich hab mir das jetzt (endlich) nochmal in Ruhe angeschaut. Das Problem ist eigentlich ganz einfach:

```go
// Enabled implements the api.Charger interface
func (c *OCPP) Enabled() (bool, error) {
	txn, err := c.cp.TransactionID()
	return txn > 0, err
}
```

Wir sind immer "enabled" wenn es eine laufende Transaktion gibt. 

Aber:

```sh
[ocpp  ] TRACE 2023/05/20 10:56:41 received JSON message from EVB-P20252147: [2,"2bfb47c9-df13-4670-b78b-408be7c61347","StopTransaction",{"transactionId":1,"idTag":"04C6A80ABF6680","timestamp":"2023-05-20T08:56:41Z","meterStop":2034698,"reason":"EVDisconnected"}]
[ocpp  ] TRACE 2023/05/20 10:56:41 sent JSON message to EVB-P20252147: [3,"2bfb47c9-df13-4670-b78b-408be7c61347",{"idTagInfo":{"status":"Accepted"}}]
```

Wenn sich das Auto(!) bei uns abmeldet wird die Transaktion auch gestoppt:

```go
func (cp *CP) StopTransaction(request *core.StopTransactionRequest) (*core.StopTransactionConfirmation, error) {
	if request != nil {
		cp.mu.Lock()
		defer cp.mu.Unlock()

		// reset transaction
		if time.Since(request.Timestamp.Time) < transactionExpiry { // only respect transactions in the last hour
			// log mismatching id but close transaction anyway
			if request.TransactionId != cp.txnId {
				cp.log.ERROR.Printf("stop transaction: invalid id %d", request.TransactionId)
			}

			cp.txnId = 0
		}
	}
}
```

Wir müssen also unterscheiden, wer die TXN gestoppt hat und dem Loadpoint den passenden Status zurück spiegeln.

/cc @GrimmiMeloni ein bisschen das gleiche Spiel wie bei der Easee ;)

andig, 2023-09-15:
Gäbe es jemanden der den PR mal lokal testen könnte? Ladung starten, Auto abziehen und später wieder anstecken und neue Ladung beginnen sollte schon reichen.

closed 2023-09-16 by andig

binzman, 2023-09-17:
Ich war die letzten Tage unterwegs und kann auch erst frühestens Mittwoch testen. Wenn ich es richtig sehe, ist es inzwischen ja schon in der 1.20.0 drin.

Werde es testen und gebe Rückmeldung

dscho, 2023-09-18:
> Werde es testen und gebe Rückmeldung

Ich auch, sobald mal wieder genügend Sonne da ist zum Laden (grmbl).

binzman, 2023-09-23:
Ich konnte es heute kurz mit v0.120.0 testen. Sieht so aus, als wäre der Fehler nun nicht mehr da.

Ich hatte allerdings ein anderes kleines Problem. Vermutlich, weil ich das Auto zu schnell wieder angeschlossen habe.

Hier das Log dazu:
- evcc stand auf Min+PV
- Das Log beginnt mitten in einem Ladevorgang
- Ich habe dann das Laden über das Auto beendet und das Kabel abgezogen
- Es kamen keine Fehlermeldungen
- Ich habe dann (vermutlich zu) kurz gewartet und das Kabel wieder angeschlossen
- evcc hat zweimal versucht mit Min + PV einen Ladevorgang mit 6A zu starten
- Die Wallbox hat beide Versuche rejected
- ich habe dann in evcc per Weboberfläche auf "Schnell" geschaltet
- evcc hat versucht, einen Ladevorgang mit 16A zu starten. Das war erfolgreich
- Danach habe ich auf "Min+PV" umgestellt und seitdem läuft der Ladevorgang gut (Dieser Teil ist nicht mehr im Log)

<details><summary>Logfile</summary>
<p>

```
Sep 23 12:37:53 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:53 received JSON message from EVB-P20252147: [2,"2ca17c80-df13-4670-b78b-408b5e19b28e","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Charging,NoError,No error.,1,Blue,1,{20,8000,13},{2422332,1349},{C,12160,5763,11888},0,0,320,2023-09-23T10:37:52Z,1,57,{226,0,9,60,0,0,993,0,0},220,5,0,116,0,240,78,5003"}]
Sep 23 12:37:53 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:53 sent JSON message to EVB-P20252147: [3,"2ca17c80-df13-4670-b78b-408b5e19b28e",{"status":"Accepted"}]
Sep 23 12:37:53 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:53 sent JSON message to EVB-P20252147: [2,"3488170167","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":7.1}]}}}]
Sep 23 12:37:53 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:53 received JSON message from EVB-P20252147: [3,"3488170167",{"status":"Accepted"}]
Sep 23 12:37:58 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:58 received JSON message from EVB-P20252147: [2,"2ca17c86-df13-4670-b78b-408b471c58ae","StopTransaction",{"transactionId":1,"idTag":"evcc","timestamp":"2023-09-23T10:37:58Z","meterStop":2422333,"reason":"EVDisconnected"}]
Sep 23 12:37:58 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:58 sent JSON message to EVB-P20252147: [3,"2ca17c86-df13-4670-b78b-408b471c58ae",{"idTagInfo":{"status":"Accepted"}}]
Sep 23 12:37:59 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:59 received JSON message from EVB-P20252147: [2,"2ca17c87-df13-4670-b78b-408b2b2678b4","StatusNotification",{"connectorId":1,"status":"Available","errorCode":"NoError","info":"EV side disconnected.","timestamp":"2023-09-23T10:37:59Z"}]
Sep 23 12:37:59 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:59 sent JSON message to EVB-P20252147: [3,"2ca17c87-df13-4670-b78b-408b2b2678b4",{}]
Sep 23 12:37:59 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:59 received JSON message from EVB-P20252147: [2,"2ca17c87-df13-4670-b78b-408b1b02dbb4","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Available,NoError,EV side disconnected.,0,Off,0,{0,8000,100},{2422333,4},{A,12160,12220,61568},0,0,320,2023-09-23T10:37:59Z,0,57,{230,65,25,2,0,0,993,0,0},230,0,0,116,0,240,900,5003"}]
Sep 23 12:37:59 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:37:59 sent JSON message to EVB-P20252147: [3,"2ca17c87-df13-4670-b78b-408b1b02dbb4",{"status":"Accepted"}]
Sep 23 12:38:47 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:38:47 received JSON message from EVB-P20252147: [2,"2ca17cb7-df13-4670-b78b-408bca41a1d1","Heartbeat",{}]
Sep 23 12:38:47 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:38:47 sent JSON message to EVB-P20252147: [3,"2ca17cb7-df13-4670-b78b-408bca41a1d1",{"currentTime":"2023-09-23T10:38:47Z"}]
Sep 23 12:38:53 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:38:53 received JSON message from EVB-P20252147: [2,"2ca17cbd-df13-4670-b78b-408b57a3b9f6","StatusNotification",{"connectorId":1,"status":"Preparing","errorCode":"NoError","timestamp":"2023-09-23T10:38:53Z"}]
Sep 23 12:38:53 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:38:53 sent JSON message to EVB-P20252147: [3,"2ca17cbd-df13-4670-b78b-408b57a3b9f6",{}]
Sep 23 12:38:53 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:38:53 received JSON message from EVB-P20252147: [2,"2ca17cbd-df13-4670-b78b-408b891541f7","DataTransfer",{"vendorId":"EV-BOX","messageId":"evbStatusNotification","data":"1,Preparing,NoError,No error.,0,Off,0,{20,8000,100},{2422333,0},{B,12144,8860,61584},0,0,320,2023-09-23T10:38:53Z,0,57,{231,65,25,0,0,0,0,0,0},230,0,0,116,0,240,900,5003"}]
Sep 23 12:38:53 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:38:53 sent JSON message to EVB-P20252147: [3,"2ca17cbd-df13-4670-b78b-408b891541f7",{"status":"Accepted"}]
Sep 23 12:39:13 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:39:13 sent JSON message to EVB-P20252147: [2,"3774901393","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":0,"chargingProfilePurpose":"TxProfile","chargingProfileKind":"Relative","chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
Sep 23 12:39:13 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:39:13 received JSON message from EVB-P20252147: [3,"3774901393",{"status":"Rejected"}]
[... cut ...]
Sep 23 12:40:12 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:40:12 received JSON message from EVB-P20252147: [2,"2ca17d0b-df13-4670-b78b-408bef6f38ca","StatusNotification",{"connectorId":1,"status":"Charging","errorCode":"NoError","info":"C;320","timestamp":"2023-09-23T10:40:11Z"}]
Sep 23 12:40:12 evcc evcc[4674]: [ocpp  ] TRACE 2023/09/23 12:40:12 sent JSON message to EVB-P20252147: [3,"2ca17d0b-df13-4670-b78b-408bef6f38ca",{}]
```
</p>
</details> 

Ich werde bei der nächsten Gelegenheit noch einmal einen Test machen, bei dem ich nach dem Beenden des Ladevorgangs und vor dem Starten des nächsten länger warte...

andig, 2023-09-23:
> Die Wallbox hat beide Versuche rejected

Mach bitte ein neues Issue und Logfile- hier fehlt der komplette Kontext von Ladepunkt und Site.

binzman, 2023-09-23:
Ich wollte hier nur bestätigen, dass der issue erfolgreich gelöst ist.

Sollte ich den anderen Fehler reproduzieren können, mache ich ein ordentliches Log und ein neues issue

andig, 2023-09-23:
Super, danke!