evcc-io/evcc #8120: Tuxedo Distro ubuntu 22.04 Derivat

hbf12, 2023-05-21:
### Describe the bug

Sorry, muss mich diesbezüglich nochmal melden.
Der Kollege hatte das in den bereits geschlossenen Tread reingeschrieben, hätte ich auch gleich erwähnen können.

Genau aber bringt das was wenn ich in den geschlossenen noch etwas reinschreibe ?
Ich habe versucht die Distro zu ändern in ubuntu 22.04 das ging leider nicht.
Beim Versuch das mit der Installationsanleitung zu ändern kommt auch wieder eine Fehlermeldung.

### Steps to reproduce

1.Installation EVCC bricht ab
2.Laut Anleitung versucht Distro zu ändern, leider kommt da auch eine Fehlermeldung 
3.   OK: Importing 'evcc/stable' repository GPG keys ...
 NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=tuxedo  version=22.04  codename=jammy  arch=x86_64  
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as Ubuntu/Xenial (16.04), use the following:
 >>>>:
 >>>>:   <curl command> | distro=ubuntu version=16.04 codename=xenial sudo -E bash
 >>>>:
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact evcc for further assistance.
 !!!!: 
 !!!!: URL: https://github.com/evcc-io/evcc
 !!!!: 
 !!!!: Last 20 log lines from .csm_setup_FuO8iSusP (might not be errors, nor even relevant):
 !!!!:
 !!!!: > ii  gnupg                                         2.2.27-3ubuntu2.1
 !!!!: > all          GNU privacy guard - a free PGP replacement
 !!!!: > ii  gnupg-agent                                   2.2.27-3ubuntu2.1
 !!!!: > all          GNU privacy guard - cryptographic agent (dummy
 !!!!: > transitional package)
 !!!!: > ii  gnupg-l10n                                    2.2.27-3ubuntu2.1
 !!!!: > all          GNU privacy guard - localization files
 !!!!: > ii  gnupg-utils                                   2.2.27-3ubuntu2.1
 !!!!: > amd64        GNU privacy guard - utility programs

:~$ <curl command> | distro=ubuntu version=22.04 codename=jammy jellyfish sudo -E bash
bash: Syntaxfehler beim unerwarteten Wort »|«
...

### Configuration details

```yaml
so weit komme ich nicht
```

### Log details

```text
gibt es keinen
```

### What type of operating system are you running?

Linux

premultiply, 2023-05-21:
`distro=ubuntu version=22.04 codename=jammy jellyfish sudo -E bash`

closed 2023-05-21 by premultiply

hbf12, 2023-05-21:
Besten Dank, so hat es funktioniert