evcc-io/evcc #8128: Keba-Modbus: Chargesession never ends...

Duell10111, 2023-05-21:
### Describe the bug

Sorry for bothering you again, the last fix worked and the correct power is now shown in the Web-Interface.
I now sadly found a new bug.
The charging session currently never ends even if a new car is plugged in and the WB is clearing the current session.

[image]

WB Data:

[image]

### Steps to reproduce

1. Starting a charge session with a vehicle
2. Stopping and unplugging car
3. Plug in again and start charging again
4. EVCC still counts up as if the first session never ended

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: https
  host: charging.hub # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

sponsortoken: CENSORED

interval: 10s # control cycle interval

meters:
- type: template
  template: solaredge-hybrid 
  usage: grid  
  host: 192.168.178.10  
  port: 1502  
  timeout: 10s  
  name: grid1
  modbus: tcpip
  id: 1
- type: template
  template: solaredge-hybrid 
  usage: pv  
  host: 192.168.178.10  
  port: 1502  
  timeout: 10s  
  name: pv2
  modbus: tcpip
  id: 1
- type: template
  template: solaredge-hybrid 
  usage: battery  
  host: 192.168.178.10
  port: 1502  
  timeout: 10s  
  name: battery3
  modbus: tcpip
  id: 1

chargers:
  - name: keba
    type: template
    template: keba-modbus 

    # Modbus TCP
    modbus: tcpip
    id: 255
    host: 192.168.178.79
    port: 502 # Port 
#  - name: wb
#    type: template
#    template: ocpp 
#    idtag: f5eh6qml76xasy5ec82p
#    stationid: 99002061

vehicles:
- name: Tesla_Model_3_P
  type: template
  template: tesla
  title: Konstantins Tesla
  accessToken: ****
  refreshToken: ****
  vin: ****
  #capacity: 81 # Akku-Kapazität in kWh # Optional 
- name: Tesla_Model_Y_LR
  type: template
  template: tesla
  title: Papas Tesla
  accessToken: ****
  refreshToken: ****
  vin: ****
  #capacity: 81 # Akku-Kapazität in kWh # Optional 

loadpoints:
- title: Garage
  charger: keba
  mode: now
  phases: 3
  mincurrent: 5
  maxcurrent: 16
  resetOnDisconnect: true
  disable:
    threshold: 200 # maximum import power (W)
    delay: 10m

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  bufferSoC: 85
  prioritySoC: 50
  maxGridSupplyWhileBatteryCharging: 200
  residualPower: 50

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.35 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0641 # [currency]/kWh
```

### Log details

```text
[site  ] DEBUG 2023/05/21 21:22:48 ----
[lp-1  ] DEBUG 2023/05/21 21:22:48 charge power: 0W
[site  ] DEBUG 2023/05/21 21:22:48 pv power: 0W
[site  ] DEBUG 2023/05/21 21:22:48 battery soc: 82%
[site  ] DEBUG 2023/05/21 21:22:48 battery power: 2548W
[site  ] DEBUG 2023/05/21 21:22:48 grid power: -0W
[site  ] DEBUG 2023/05/21 21:22:48 site power: 2598W
[lp-1  ] DEBUG 2023/05/21 21:22:48 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:22:48 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:22:48 charger status: B
[site  ] DEBUG 2023/05/21 21:22:58 ----
[lp-1  ] DEBUG 2023/05/21 21:22:58 charge power: 0W
[site  ] DEBUG 2023/05/21 21:22:58 pv power: 0W
[site  ] DEBUG 2023/05/21 21:22:58 battery soc: 82%
[site  ] DEBUG 2023/05/21 21:22:58 battery power: 2560W
[site  ] DEBUG 2023/05/21 21:22:58 grid power: -9W
[site  ] DEBUG 2023/05/21 21:22:58 site power: 2601W
[lp-1  ] DEBUG 2023/05/21 21:22:59 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:22:59 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:22:59 charger status: B
[site  ] DEBUG 2023/05/21 21:23:08 ----
[lp-1  ] DEBUG 2023/05/21 21:23:08 charge power: 0W
[site  ] DEBUG 2023/05/21 21:23:08 pv power: 0W
[site  ] DEBUG 2023/05/21 21:23:08 battery soc: 82%
[site  ] DEBUG 2023/05/21 21:23:08 battery power: 2545W
[site  ] DEBUG 2023/05/21 21:23:08 grid power: 6W
[site  ] DEBUG 2023/05/21 21:23:08 site power: 2601W
[lp-1  ] DEBUG 2023/05/21 21:23:08 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:23:08 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:23:08 charger status: B
[site  ] DEBUG 2023/05/21 21:23:18 ----
[lp-1  ] DEBUG 2023/05/21 21:23:18 charge power: 0W
[site  ] DEBUG 2023/05/21 21:23:18 pv power: 0W
[site  ] DEBUG 2023/05/21 21:23:18 battery soc: 82%
[site  ] DEBUG 2023/05/21 21:23:18 battery power: 2551W
[site  ] DEBUG 2023/05/21 21:23:18 grid power: -3W
[site  ] DEBUG 2023/05/21 21:23:18 site power: 2598W
[lp-1  ] DEBUG 2023/05/21 21:23:18 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:23:18 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:23:18 charger status: B
[site  ] DEBUG 2023/05/21 21:23:28 ----
[lp-1  ] DEBUG 2023/05/21 21:23:28 charge power: 0W
[site  ] DEBUG 2023/05/21 21:23:28 pv power: 0W
[site  ] DEBUG 2023/05/21 21:23:28 battery soc: 82%
[site  ] DEBUG 2023/05/21 21:23:28 battery power: 2545W
[site  ] DEBUG 2023/05/21 21:23:28 grid power: 2W
[site  ] DEBUG 2023/05/21 21:23:28 site power: 2597W
[lp-1  ] DEBUG 2023/05/21 21:23:28 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:23:28 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:23:28 charger status: B
[site  ] DEBUG 2023/05/21 21:23:38 ----
[lp-1  ] DEBUG 2023/05/21 21:23:38 charge power: 0W
[site  ] DEBUG 2023/05/21 21:23:38 pv power: 0W
[site  ] DEBUG 2023/05/21 21:23:38 battery soc: 81%
[site  ] DEBUG 2023/05/21 21:23:38 battery power: 2545W
[site  ] DEBUG 2023/05/21 21:23:38 grid power: -0W
[site  ] DEBUG 2023/05/21 21:23:38 site power: 2595W
[lp-1  ] DEBUG 2023/05/21 21:23:39 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:23:39 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:23:39 charger status: B
[site  ] DEBUG 2023/05/21 21:23:48 ----
[lp-1  ] DEBUG 2023/05/21 21:23:48 charge power: 0W
[site  ] DEBUG 2023/05/21 21:23:48 pv power: 0W
[site  ] DEBUG 2023/05/21 21:23:48 battery soc: 81%
[site  ] DEBUG 2023/05/21 21:23:48 battery power: 2545W
[site  ] DEBUG 2023/05/21 21:23:49 grid power: 1W
[site  ] DEBUG 2023/05/21 21:23:49 site power: 2596W
[lp-1  ] DEBUG 2023/05/21 21:23:49 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:23:49 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:23:49 charger status: B
[site  ] DEBUG 2023/05/21 21:23:58 ----
[lp-1  ] DEBUG 2023/05/21 21:23:58 charge power: 0W
[site  ] DEBUG 2023/05/21 21:23:58 pv power: 0W
[site  ] DEBUG 2023/05/21 21:23:58 battery soc: 81%
[site  ] DEBUG 2023/05/21 21:23:58 battery power: 2541W
[site  ] DEBUG 2023/05/21 21:23:58 grid power: 6W
[site  ] DEBUG 2023/05/21 21:23:58 site power: 2597W
[lp-1  ] DEBUG 2023/05/21 21:23:58 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:23:58 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:23:58 charger status: B
[site  ] DEBUG 2023/05/21 21:24:08 ----
[lp-1  ] DEBUG 2023/05/21 21:24:08 charge power: 0W
[site  ] DEBUG 2023/05/21 21:24:08 pv power: 0W
[site  ] DEBUG 2023/05/21 21:24:09 battery soc: 81%
[site  ] DEBUG 2023/05/21 21:24:09 battery power: 2549W
[site  ] DEBUG 2023/05/21 21:24:09 grid power: -0W
[site  ] DEBUG 2023/05/21 21:24:09 site power: 2599W
[lp-1  ] DEBUG 2023/05/21 21:24:09 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:24:09 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:24:09 charger status: B
[site  ] DEBUG 2023/05/21 21:24:18 ----
[lp-1  ] DEBUG 2023/05/21 21:24:18 charge power: 0W
[site  ] DEBUG 2023/05/21 21:24:18 pv power: 0W
[site  ] DEBUG 2023/05/21 21:24:18 battery soc: 81%
[site  ] DEBUG 2023/05/21 21:24:18 battery power: 2549W
[site  ] DEBUG 2023/05/21 21:24:18 grid power: 2W
[site  ] DEBUG 2023/05/21 21:24:18 site power: 2601W
[lp-1  ] DEBUG 2023/05/21 21:24:18 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 21:24:18 charge meter total import: 37069.105kWh
[lp-1  ] DEBUG 2023/05/21 21:24:18 charger status: B
```

### What type of operating system are you running?

Linux

### Version

0.117.4 (c3989733)

premultiply, 2023-05-21:
Looks like there is a bug in scaling the raw modbus meter total energy value.

andig, 2023-05-21:
Logfile doesnt show new car plugged.

closed 2023-05-21 by andig

premultiply, 2023-05-21:
TotalEnergy ist offensichtlich trotzdem um Faktor 10 falsch. ;)

andig, 2023-05-22:
Ich habs ja schonmal geschrieben. Implementiert wie Doku. Keba fragen oder einfach ändern. Bisher kein Feedback darauf…

premultiply, 2023-05-22:
Habs nur oben im Log vs. Web-IF gesehen.

andig, 2023-05-22:
Also Session und Total / 10?

premultiply, 2023-05-22:
Session sieht man da im Log nicht. Wäre aber auch erstmal zu vermuten, oder?

Duell10111, 2023-05-22:
Habe mal die Logs noch rausgesucht wo das Auto abgesteckt wird wenn das euch weiter hilft.

````
[site  ] DEBUG 2023/05/21 13:00:38 ----
[lp-1  ] DEBUG 2023/05/21 13:00:38 charge power: 3492W
[site  ] DEBUG 2023/05/21 13:00:38 pv power: 5050W
[site  ] DEBUG 2023/05/21 13:00:38 battery soc: 99%
[site  ] DEBUG 2023/05/21 13:00:38 battery power: -288W
[site  ] DEBUG 2023/05/21 13:00:38 grid power: -32W
[site  ] DEBUG 2023/05/21 13:00:38 site power: -270W
[lp-1  ] DEBUG 2023/05/21 13:00:38 charge currents: [5.09 5.12 5.07]A
[lp-1  ] DEBUG 2023/05/21 13:00:38 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/21 13:00:39 charge meter total import: 37020.766kWh
[lp-1  ] DEBUG 2023/05/21 13:00:39 charger status: C
[lp-1  ] DEBUG 2023/05/21 13:00:39 vehicle soc: 79%
[lp-1  ] DEBUG 2023/05/21 13:00:39 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/05/21 13:00:39 vehicle range: 388km
[site  ] DEBUG 2023/05/21 13:00:48 ----
[lp-1  ] DEBUG 2023/05/21 13:00:48 charge power: 3446W
[site  ] DEBUG 2023/05/21 13:00:48 pv power: 5263W
[site  ] DEBUG 2023/05/21 13:00:48 battery soc: 99%
[site  ] DEBUG 2023/05/21 13:00:48 battery power: -537W
[site  ] DEBUG 2023/05/21 13:00:48 grid power: -60W
[site  ] DEBUG 2023/05/21 13:00:48 site power: -547W
[lp-1  ] DEBUG 2023/05/21 13:00:48 charge currents: [5.03 5.05 5]A
[lp-1  ] DEBUG 2023/05/21 13:00:48 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/21 13:00:48 charge meter total import: 37020.863kWh
[lp-1  ] DEBUG 2023/05/21 13:00:48 charger status: C
[lp-1  ] DEBUG 2023/05/21 13:00:48 vehicle soc: 79%
[lp-1  ] DEBUG 2023/05/21 13:00:48 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/05/21 13:00:48 vehicle range: 388km
[site  ] DEBUG 2023/05/21 13:00:58 ----
[lp-1  ] DEBUG 2023/05/21 13:00:58 charge power: 3494W
[site  ] DEBUG 2023/05/21 13:01:07 pv power: 5679W
[site  ] DEBUG 2023/05/21 13:01:07 battery soc: 99%
[site  ] DEBUG 2023/05/21 13:01:07 battery power: -899W
[site  ] DEBUG 2023/05/21 13:01:07 grid power: -131W
[site  ] DEBUG 2023/05/21 13:01:07 site power: -980W
[lp-1  ] DEBUG 2023/05/21 13:01:07 charge currents: [5.09 5.12 5.08]A
[lp-1  ] DEBUG 2023/05/21 13:01:07 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/21 13:01:08 charge meter total import: 37021.047kWh
[lp-1  ] DEBUG 2023/05/21 13:01:08 charger status: C
[lp-1  ] DEBUG 2023/05/21 13:01:08 vehicle soc: 79%
[lp-1  ] DEBUG 2023/05/21 13:01:08 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/05/21 13:01:08 vehicle range: 388km
[site  ] DEBUG 2023/05/21 13:01:08 ----
[lp-1  ] DEBUG 2023/05/21 13:01:08 charge power: 3495W
[site  ] DEBUG 2023/05/21 13:01:08 pv power: 5763W
[site  ] DEBUG 2023/05/21 13:01:08 battery soc: 99%
[site  ] DEBUG 2023/05/21 13:01:08 battery power: -1013W
[site  ] DEBUG 2023/05/21 13:01:08 grid power: -38W
[site  ] DEBUG 2023/05/21 13:01:08 site power: -1001W
[lp-1  ] DEBUG 2023/05/21 13:01:08 charge currents: [5.09 5.12 5.08]A
[lp-1  ] DEBUG 2023/05/21 13:01:08 detected active phases: 3p
[lp-1  ] DEBUG 2023/05/21 13:01:08 charge meter total import: 37021.057kWh
[lp-1  ] DEBUG 2023/05/21 13:01:08 charger status: C
[lp-1  ] DEBUG 2023/05/21 13:01:09 vehicle soc: 79%
[lp-1  ] DEBUG 2023/05/21 13:01:09 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/05/21 13:01:09 vehicle range: 388km
[site  ] DEBUG 2023/05/21 13:01:18 ----
[lp-1  ] DEBUG 2023/05/21 13:01:18 charge power: 0W
[site  ] DEBUG 2023/05/21 13:01:18 pv power: 5879W
[site  ] DEBUG 2023/05/21 13:01:18 battery soc: 99%
[site  ] DEBUG 2023/05/21 13:01:18 battery power: -4629W
[site  ] DEBUG 2023/05/21 13:01:19 grid power: -65W
[site  ] DEBUG 2023/05/21 13:01:19 site power: -4644W
[lp-1  ] DEBUG 2023/05/21 13:01:19 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/21 13:01:19 charge meter total import: 37021.128kWh
[lp-1  ] DEBUG 2023/05/21 13:01:19 charger status: B
[lp-1  ] INFO 2023/05/21 13:01:19 stop charging <-
[... 661 lines cut ...]
[lp-1  ] INFO 2023/05/21 14:22:49 start charging ->
[lp-1  ] DEBUG 2023/05/21 14:22:49 wake-up timer: stop
[lp-1  ] DEBUG 2023/05/21 14:22:49 vehicle soc: 77%
[lp-1  ] DEBUG 2023/05/21 14:22:49 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/05/21 14:22:49 vehicle range: 378km
[lp-1  ] DEBUG 2023/05/21 14:22:49 pv charge current: 10.8A = 2.27A + 8.52A (-5879W @ 3p)
[lp-1  ] DEBUG 2023/05/21 14:22:49 max charge current: 10.8A
[site  ] DEBUG 2023/05/21 14:22:58 ----
[lp-1  ] DEBUG 2023/05/21 14:22:58 charge power: 382W
[site  ] DEBUG 2023/05/21 14:22:58 pv power: 7144W
[site  ] DEBUG 2023/05/21 14:22:58 battery soc: 100%
[site  ] DEBUG 2023/05/21 14:22:58 battery power: 0W
[site  ] DEBUG 2023/05/21 14:22:59 grid power: -5636W
[site  ] DEBUG 2023/05/21 14:22:59 site power: -5586W
[lp-1  ] DEBUG 2023/05/21 14:22:59 charge currents: [1.77 0.28 0.268]A
[lp-1  ] DEBUG 2023/05/21 14:22:59 detected active phases: 1p
[lp-1  ] DEBUG 2023/05/21 14:22:59 charge meter total import: 37021.129kWh
[lp-1  ] DEBUG 2023/05/21 14:22:59 charger status: C
[lp-1  ] DEBUG 2023/05/21 14:22:59 vehicle soc: 77%
[lp-1  ] DEBUG 2023/05/21 14:22:59 vehicle soc limit: 80%
[lp-1  ] DEBUG 2023/05/21 14:22:59 vehicle range: 378km
[lp-1  ] DEBUG 2023/05/21 14:22:59 pv charge current: 28.1A = 3.77A + 24.3A (-5586W @ 1p)
[lp-1  ] DEBUG 2023/05/21 14:22:59 max charge current: 16A
[site  ] DEBUG 2023/05/21 14:23:08 ----
```

andig, 2023-05-22:
Probieren wirs mal: eebdacaf338b0e2fc8307239c02bdafd1aace539

Duell10111, 2023-05-22:
Ich habe noch einen kleinen Fehler gefunden beim parsen des ChargingState gefunden, sodass es immer im State B ist.

Siehe Doku Seite 10.
https://www.keba.com/download/x/dea7ae6b84/kecontactp30modbustcp_pgen.pdf

Habe dazu mal ein PR auf:
https://github.com/evcc-io/evcc/pull/8131

Duell10111, 2023-05-22:
Hab das ganze mal lokal gebaut und bei mir klappt das jetzt auch. :D

````
[main  ] INFO 2023/05/22 12:26:54 evcc 0.117.4 (7f64ba74)
[main  ] INFO 2023/05/22 12:26:54 using config file: /etc/evcc.yaml
[main  ] INFO 2023/05/22 12:26:54 starting ui and api at :7070
[db    ] INFO 2023/05/22 12:26:55 using sqlite database: /root/.evcc/evcc.db
[site  ] WARN 2023/05/22 12:26:58 deprecated: use 'pv' instead of 'pvs'
[site  ] WARN 2023/05/22 12:26:58 deprecated: use 'battery' instead of 'batteries'
[site  ] INFO 2023/05/22 12:26:58 site config:
[site  ] INFO 2023/05/22 12:26:58   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/05/22 12:26:58     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/22 12:26:58     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/05/22 12:26:58     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/05/22 12:26:58   vehicles:
[site  ] INFO 2023/05/22 12:26:58     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] INFO 2023/05/22 12:26:58     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/05/22 12:26:58 loadpoint 1:
[lp-1  ] INFO 2023/05/22 12:26:58   mode:        now
[lp-1  ] INFO 2023/05/22 12:26:58   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/05/22 12:26:58   meters:      charge ✓
[lp-1  ] INFO 2023/05/22 12:26:58     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/05/22 12:26:58 phase timer inactive
[lp-1  ] DEBUG 2023/05/22 12:26:58 pv timer inactive
[lp-1  ] DEBUG 2023/05/22 12:26:58 guard timer inactive
[lp-1  ] DEBUG 2023/05/22 12:26:58 max charge current: 5A
[site  ] DEBUG 2023/05/22 12:26:58 ----
[lp-1  ] DEBUG 2023/05/22 12:26:58 charge power: 0W
[site  ] DEBUG 2023/05/22 12:26:58 pv power: 7860W
[site  ] DEBUG 2023/05/22 12:26:58 battery soc: 100%
[site  ] DEBUG 2023/05/22 12:26:58 battery power: 0W
[site  ] DEBUG 2023/05/22 12:26:58 grid power: -6691W
[site  ] DEBUG 2023/05/22 12:26:58 site power: -6641W
[lp-1  ] DEBUG 2023/05/22 12:26:58 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/05/22 12:26:58 charge meter total import: 3708.444kWh
[lp-1  ] DEBUG 2023/05/22 12:26:58 charger status: A
[lp-1  ] INFO 2023/05/22 12:26:58 car disconnected
```

Bezüglich deiner Änderungen muss ich noch mein Auto anstecken und schauen was das verändert. 😅