evcc-io/evcc #8130: Tesla: SoC wird nicht aktualisiert

sebnaf, 2023-05-22:
### Describe the bug

Ich habe mir mal die Mühe gemacht evcc basierend auf aktueller Doku einzurichten. Interessanterweise aktualisiert Tesla nicht richtig: ich stecke das Auto Abends an, es wird erkannt. Dann jedoch über 24h der SoC nicht aktualisiert. Das Auto ist wach. Tesla-App und Tronity zeigt es richtig an. Wechsle ich zu Tronity und wieder zurück wird der SoC richtig ausgelesen und die Ladung beginnt.
Ich hatte früher einen Cache-Parameter aktiv welcher heute nicht mehr erkannt wird.

Das Tracelog zeigt, dass beim
- Einstecken der SoC gelesen wird
- beim manuellen Wechsel zwischen den Autos
dann aber nicht mehr bis PV Überschuss vorhanden ist und geladen wird.

Hier liegt das Problem: Wenn ich mit 80% SoC einstecke und auf 80% SoC laden möchte wird die Ladung nie aktiv.

Idee: Alle n Minuten nach dem SoC fragen.
Ich glaube, das war in der alten Config sogar ein Parameter in Verbindung mit "Cache"

[image]
[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - type: template
    template: tesla
    title: EM-IL 55 E
    capacity: 78
    accessToken: ey
    refreshToken: ey
    vin: LRW
    phases: 0
    icon: car
    mode: pv
    minSoc: 20
    targetSoc: 80
    minCurrent: 6
    maxCurrent: 16
    name: tesla
    #cache: 15m

  - type: template
    template: tronity
    title: EM-IL 55 E (Tronity)
    user: 6e
    password: 31
    capacity: 78
    phases: 0
    icon: car
    mode: pv
    minSoc: 20
    targetSoc: 80
    minCurrent: 6
    maxCurrent: 16
    name: ev3
```

### Log details

```text
[tesla ] TRACE 2023/05/21 21:48:06 GET https://owner-api.teslamotors.com/api/1/vehicles

und dann wieder

[tesla ] TRACE 2023/05/22 09:16:48 GET https://owner-api.teslamotors.com/api/1/vehicles/
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.117.3

andig, 2023-05-27:
Without full log and config it's unclear what's going on.

andig, 2023-06-06:
Config fehlt weiterhin :(

closed 2023-06-06 by andig

sebnaf, 2023-06-06:
Sorry, ich dachte der Auszug oben reicht.

``` 
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 10s # control cycle interval
plant: 2d22a

sponsortoken: eyJhb

telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: debug
  db: debug
  vehicle: debug
  tesla: trace
  elektronenpumpe: debug

# meter definitions
meters:
  - name: grid
    type: movingaverage
    decay: 0.2
    meter:
      type: custom
      power:
        source: modbus
        model: sunspec
        uri: 10.10.20.7:502
        id: 1
        subdevice: 1
        value: 203:W
        scale: -1.09
        # decay: 0.7
        #timeout: 2000
  - name: pv
    type: custom
    power:
      source: modbus
      model: sunspec
      uri: 10.10.20.7:502
      id: 1
      value: 101:W
      #timeout: 2000

# charger definitions
chargers:
  - name: elektronenpumpe
    type: go-e
    uri: http://10.10.20.22
    # token: 
    # cache: 10s # go-e cloud API cache duration

# vehicle definitions
vehicles:
  - type: template
    template: tesla
    title: EM-IL 55 E
    capacity: 78
    accessToken: 
    refreshToken: 
[... 128 lines cut ...]
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
    - type: pushover
      app: a
      recipients:
        - u
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
```

RTTTC, 2023-06-06:
Use `poll` `connected` configuration in `loadpoints:` ?
https://github.com/evcc-io/evcc/blob/master/evcc.dist.yaml

andig, 2023-06-06:
Exactly. You have `poll: charging` and that's exactly what happens. Seems you'd want `poll: connected`. We'd still need to see how/when you would wake up the Tesla then, similar to https://github.com/evcc-io/evcc/issues/8254.