evcc-io/evcc #8140: EVCC request to charger EASEE cloud returns "message":"Forbidden" - #8083 continued

wita-sec, 2023-05-22:
### Describe the bug

Please find issue details in #8083.
I talked to easee support.
They said, that my IP has been blocked by their firewall.
The reason was that evcc posts too many requests to the easee cloud api. 
They claim that evcc needs to adopt code in order to comply with easee.  
They told me that they already have contacted evcc dev team about this issue.

evcc team, please can you confirm, and please can you give me a timeline for solving this problem? 

Here is an extract of my emailing with easee support:

<<<<<<
Von:	Abdel (Easee) <support.international@easee.com>
Gesendet:	Montag, 22. Mai 2023 10:48
An:	Michael-jj Wolf
Betreff:	Ich möchte gerne den Zugriff/Berechtigung ändern

Wir haben deine Anfrage (159139) aktualisiert. Bitte antworte auf diese E-Mail, wenn du weitere 
Kommentare oder Fragen hast ??
 
Abdel (Easee) 
22. Mai 2023, 10:47 MESZ 
Hallo Michael,  
  
da evcc den Code anpassen muss, haben wir diese Infos bereits an EVCC über unsere 
Entwicklern kommuniziert.  
Die Dokumentation der API ist hier zu finden: https://developer.easee.cloud/docs  
 
Die anzahl zulässiger Befehle hängt von den Endpunkten ab. 
  
Wenn du weitere Fragen hast, wende dich bitte an EVCC. 
Mit freundlichen Grüßen,
Abdelmjid
  
https://support.easee.com
>>>>>>>>

### Steps to reproduce

1. see #8083

### Configuration details

```yaml
see #8083
```

### Log details

```text
see #8083
```

### What type of operating system are you running?

Linux

### Version

0.112.5

RTTTC, 2023-05-23:
If I understood correctly - in your previous, now closed, issue you were given advice to upgrade software?

Is there a reason you are running an old evcc version (20 versions behind) and keep asking evcc team "to fix something"?
Maybe it is already fixed in current version?
Your version 0.112.5
Current version 0.117.4 https://github.com/evcc-io/evcc/releases

Another point you may be missing here - ask Easee support what is "not too many" ? (The reason was that evcc posts too many requests to the easee cloud api.) So how many requests say per 24 hour period is OK?
Then you could adjust your control `interval` accordingly.

PS: Obviously - something changed on Easee end, since you are on the same version of evcc, which previoiusly worked good for you..
PPS: There was a discussion here that due to insufficient IP4 address numbers in Germany some internet providers use IP address sharing, that would effectively be seen as one IP that contacts Easee server way too often, when in fact it is many different people that are trying to do that separately. Do a search here for more info and solutions.

And finally my condolences on purchasing easee, I would advice selling it to someone that would use it as a dumb charger, and getting yourself a bettere one, with local control only..

wita-sec, 2023-05-23:
RTTC, thank you for investigation. 
You said:
Another point you may be missing here - ask Easee support what is "not too many" ? (The reason was that evcc posts too many requests to the easee cloud api.) So how many requests say per 24 hour period is OK?
Then you could adjust your control interval accordingly.

I had asked easee that question, however they did not give me a clear answer. Rather they said, that depends on the type of request, which in fact is not very helpful. 
My application runs evcc -c <config file> charger twice per hour. Not sure whether thats too much for easee. I will change this.
Another amount of evcc requests probably are posted when easee charger is charging.  

I have meantime upgraded my evcc from 0.122.5 to 0.117.4. 
I seem to remember that earlier evcc did the upgrade on its own. Looks like I was wrong with this.
Sorry for that.

RTTTC, 2023-05-23:
Hopefully this will get it going. Let us know if you solved the issue.