evcc-io/evcc #8143: Fahrzeug über API aufwecken wenn CP Unterbrechung angefordert wird.

MarkusGH, 2023-05-23:
Porsche PHEV Fahrzeuge scheinen das Verhalten zu haben, dass sie sich nach 5 vergeblichen Ladeversuchen oder Ladeunterbrechungen schlafen legen und auch bei CP Unterbrechung nicht aufwachen, falls Sie an einer Wallbox hängen, die das IEC 61851 Protokoll (Kommunikation über PWM) verwendet.

Aufwecken über die Porsche API oder jede andere Aktion, die zum Erwachen des Fahrzeugs führt geht aber und löst das Problem, da das Fahrzeug beim Einschlafen alle vorherigen Fehler "vergisst".

Wenn der Poll Mode auf "charging" steht passiert das leider nicht.

Vorschlag:

Fahrzeug einmalig pollen, wenn bei der Wallbox eine CP Unterbrechung angefordert wird.

andig, 2023-05-23:
> Fahrzeug einmalig pollen, wenn bei der Wallbox eine CP Unterbrechung angefordert wird.

Das machen wir heute tatsächlich nicht sondern entweder oder:

```go
func (lp *Loadpoint) wakeUpVehicle() {
	// charger
	if c, ok := lp.charger.(api.Resurrector); ok {
		if err := c.WakeUp(); err != nil {
			lp.log.ERROR.Printf("wake-up charger: %v", err)
		}
		return
	}

	// vehicle
	if vs, ok := lp.vehicle.(api.Resurrector); ok {
		if err := vs.WakeUp(); err != nil {
			lp.log.ERROR.Printf("wake-up vehicle: %v", err)
		}
	}
}
```

Würde das Problem für Porsche mangels `api.Resurrector` aber nicht lösen. Stand heute haben wir das nur für Bluelink/ Ford/ Renault.

MarkusGH, 2023-05-23:
OK, das würde bedeuten:

1) wakeUpVehicle ändern, so dass beide Methoden versucht werden
2) api.Resurrector für den Porsche Provider implementieren

Sehe ich richtig, dass für 2) schlicht in porsche/provider.go analog zu Ford/provider.go eine Funktion WakeUp implementiert werden muss, die dann z.B. die Porsche API Funktion Status aufruft?

MarkusGH, 2023-05-23:
Naja - ob Status reicht kann ich aktuell nicht sagen. Ein Neustart von evcc scheint das Fahrzeug nicht zuverlässig zu wecken.
Verriegeln (obwohl schon verriegelt war) über die App hat geholfen.
Gibt es irgendwo eine Doku der Schnittstellen oder weiß irgendjemand was man da zum Aufwecken hernehmen könnte?

andig, 2023-05-27:
Keine neuen Erkenntnisse, closing.

closed 2023-05-27 by andig

MarkusGH, 2023-05-28:
Implementiert in #8204