evcc-io/evcc #8146: Enphase Envoy with token-based authentication

salz3n, 2023-05-23:
**BRAUCHE UNTERSTÜTZUNG BEI DER PROGRAMMIERUNG**

In Issue https://github.com/evcc-io/evcc/issues/7779 wurde ein Template für das Enphase Envoy-S implementiert. 

Leider wurde bei der neueren Enphase Envoy Firmware der Zugriff auf eine Token-Authentifizierung umgestellt. Daher geht das aktuelle Template https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/enphase.yaml nicht mehr für die neuesten Enphase-Versionen.

Ich habe das Template leicht angepasst und konnte es zumindest als Proof-of-Concept wieder zum Laufen bringen:

`/templates/definition/meter/enphase.yaml`
```
template: enphase
products:
  - brand: Enphase
    description:
      generic: IQ Envoy
params:
  - name: usage
    choice: ["pv"]
  - name: host
    required: true
render: |
  type: custom
  power:
    source: http
    uri: https://{{ .host }}/api/v1/production
    method: GET
    auth:
      type: bearer
      password: <THE TOKEN>
    insecure: true
    jq: .wattsNow
  energy:
    source: http
    uri: https://{{ .host }}/api/v1/production
    method: GET
    auth:
      type: bearer
      password: <THE TOKEN>
    insecure: true
    jq: .wattHoursLifetime
    scale: 0.001
```

**Jedoch gibt es noch zwei Probleme, welche vor einem Pull Request gelöst werden müssten. Vielleicht findet sich ja jemand, der mich unterstützen kann, da ich noch nie in Go programmiert habe.**

Problem 1: Token-Generierung
---

Das Authentifizierung-Token (oben im Code mit `<THE TOKEN>` benannt) hat nur eine Laufzeit von 1 Jahr und müsste daher am besten automatisiert vom evcc Code erzeugt und vor Ablauf erneuert werden.

Der genaue Prozess ist hier sehr gut beschrieben: https://enphase.com/download/iq-gateway-access-using-local-apis-or-local-ui-token-based-authentication-tech-brief

Es gibt auch ein funktionierendes Skript in Python. Dieses müsste vermutlich in Go umgeschrieben und dann irgendwie in den evcc Code integriert werden:

```
import json
import requests
user=’<UserName>’
password='<Password>'
envoy_serial=’< Envoy_Serial_No>’
data = {'user[email]': user, 'user[password]': password}
response = requests.post('https://enlighten.enphaseenergy.com/login/login.json?', data=data) response_data = json.loads(response.text)
data = {'session_id': response_data['session_id'], 'serial_num': envoy_serial, 'username': user}
response = requests.post('https://entrez.enphaseenergy.com/tokens', json=data) token_raw = response.text
```

Problem 2: Endpoint production.json funktioniert nicht mehr
---

Der Endpoint `/production.json` ist leider zur Zeit noch nicht mit Hilfe des Bearer Tokens abrufbar. Dies scheint ein Bug von der Enphase API zu sein, da nur manche API-Endpunkte mit dem Token funktionieren - andere aber nur mit einem Session-Cookie (Erst im Browser lokal in den Envoy mit Hilfe vom Bearer Token einloggen, dann den Cookie-Wert `sessionId` auslesen).

Es gibt jedoch noch den Endpoint `/api/v1/production`. Dieser scheint ähnliche - aber nicht die selben Werte zu liefern. Ich bin mir nicht sicher, ob das die korrekten Werte sind. Eventuell kann man ja einfach auf diesen Endpoint wechseln.

```
{
  "wattHoursToday": 3954,
  "wattHoursSevenDays": 17063,
  "wattHoursLifetime": 21017,
  "wattsNow": 979
}
```

Ich würde mich sehr freuen, wenn sich jemand findet, mit dem ich das Template fertig entwickeln kann.

andig, 2023-05-23:
Ich sehe nicht wie wir Token Auth generisch implementieren könnten. Das würde dann wohl eine eigene Go Umsetzung erfordern, Beispiele mit HTTP finden sich diverse, z.B. die Easee (OAuth2).

pitware2005, 2023-05-28:
@salz3n danke für Deine Arbeit.

Meine Meinung zu Deinen 2 Punkten wäre:
1. Das Token ist 12 Monate gültig, das sollte also kein Problem darstellen es 1x im Jahr von Hand zu erneuern.
2. Bei dem anderen Endpunkt gibt es da überhaupt alle wichtigen Daten? Also Produktion und Bezug?

Kannst Du bitte die anderen Punkte auch einbauen? Also Bezug usw. Eben was wichtig ist.

@andig könnten bitte in dem aktuellen Template auch der Bezug mit aufgenommen werden. Damit man es nicht über irgendwelche Umwege bauen muss?!

andig, 2023-05-28:
Ein PR für das Template wäre schön

salz3n, 2023-05-31:
Bei mir wurde gestern das neuste Update im Enphase Envoy installiert und nun bekomme ich endlich alle Werte in der API korrekt ausgeliefert. Ich werde im Laufe der Woche noch ein PR erstellen. 

Wenn ich das richtig sehe, dann kann ich mit dem Template dann `meters` für `pv`, `grid` und `battery` implementieren. Jedoch habe ich keine Batterie und könnte somit nur `pv` und `grid` testen. 

@pitware2005 Hast Du eine Enphase-Installation mit Software 7.x und Batterie zum Testen von meinem PR?

bgrigoriu, 2023-06-04:
Hi everyone 
Just trying to set-up my enphase with evcc and had some difficulties . Can you make a shot to to list for IQ with 7.X firmware  for newbies like me ?

bgrigoriu, 2023-06-04:
HI everyone,

A have some difficulties in setting up an IQ enphase with 7.X firmware. Can you make a short step by syep to-do list with the settings or a link to a newbie style tutorial ?

salz3n, 2023-06-04:
@bgrigoriu It shouldn't be too complex, but you will have to wait until the support of firmware 7.X is added to the code. The maintainers of the project need to include the pull request https://github.com/evcc-io/evcc/pull/8247 into a release of evcc. Until this change is not included the Enphase Envoy 7.x is not supported by evcc.

bgrigoriu, 2023-06-04:
Thanks for the comment.
I will wait. 
Unfortunately I am not at all a Go programmer and I cannot help. 
For the 5.XX firmware a wrote a routine in C++  which gets a stream reading from the IQ envoy with 2x reads/ sec on the local network
I can write somthing similar thing for the 7.X but since it uses  a token for authethican it has to pass through their servers and it is not possible to get a such granular reading ( but I am not sure at all that this is necesary). 
When the issue is fixed Please let me know. I am craving tu use the system which seams to very very well designed.