evcc-io/evcc #8150: Improve health check with additional information

prior-philip, 2023-05-23:
**Is your feature request related to a problem? Please describe.**
Im getting a health check error with no further information what's causing problem.

**Describe the solution you'd like**
A log snipped or reason code would help a lot.

**Describe alternatives you've considered**
Grabbing thru the logs by consulting the documentation first.

andig, 2023-05-23:
The point of the health check is to catch application deadlocks and terminal errors. Most likely, there won‘t be any structured error information. The last error is available via /api/state.

closed 2023-05-23 by andig

prior-philip, 2023-05-24:
The point is.... Im getting an unsuccessful health check with no information about whats causing the issue.
`evcc health
[main  ] INFO 2023/05/24 10:03:08 evcc 0.117.4
[main  ] ERROR 2023/05/24 10:03:08 health check failed`

while the error logs are clean and not reporting errors and /api/health is also fine.
```
curl -X GET http://localhost:8083/api/health
OK
```
BTW: Why do you mention /api/state ? I do not get any system health information with that call, or do I?

I still believe that enriching health output with details from log would help users a lot.

prior-philip, 2023-05-24:
Sure.
Did a restart few minutes before creating logs to reduce log size. Ensured health check still reports error.
Pls find log attached.
[evcc.log](https://github.com/evcc-io/evcc/files/11552730/evcc.log)

prior-philip, 2023-05-24:
Maybe a config dump is even more helpful.
[config.txt](https://github.com/evcc-io/evcc/files/11552790/config.txt)

RTTTC, 2023-05-25:
This does not look right?
Try min 6 . 
Max 16, (or 12, 10 ..)
(Charging does not work below 6A according to the standard)
```
vehicles:
..
..
  minCurrent: 3  
  maxCurrent: 6

```

prior-philip, 2023-05-25:
Yes, doesnt make sense indeed. Changed it back to 6a min / 16a max.
Anyhow, that didnt solve the health check issue.